zwave-js/zwave-js #1106: GE Enbrighten Z-Wave Plus Smart Dimmer not responding to ConfigurationCCInfoReport during NodeInfo interview

RoboPhred, 2020-11-08:
This is a reappearance of a previous issue: #765

Originally, I had a single switch not properly responding to interviews.  However, the problem is no longer intermittent and now reliably affects every switch of that type.  The specific device is a [GE Enbrighten Z-Wave Plus Smart Dimmer, 2nd generation](https://www.amazon.com/GE-Enbrighten-SimpleWire-SmartThings-46203/dp/B07RRD92T8/).

The interview of the switch proceeds normally, up until it reaches NodeInfo.  At that point, it tries and fails 5 times to interview.  Here is a log sample from the last try of one of the switches
```
15:28:08.891 CNTRLR   [Node 025] Beginning interview - last completed stage: NodeInfo
15:28:08.893 CNTRLR   [Node 025] ManufacturerSpecificCC: doing a partial interview...
15:28:08.903 CNTRLR   [Node 025] [~] [Manufacturer Specific] interviewComple [Endpoint 0] [internal]
                      te: true => true
15:28:08.909 CNTRLR   [Node 025] VersionCC: doing a partial interview...
15:28:08.911 CNTRLR   [Node 025] [~] [Version] interviewComplete: true => tr [Endpoint 0] [internal]
                      ue
15:28:08.912 CNTRLR   [Node 025] trying to load device config
15:28:08.914 CNTRLR   [Node 025] device config loaded
15:28:08.915 CNTRLR   [Node 025] ZWavePlusCC: doing a partial interview...
15:28:08.916 CNTRLR   [Node 025] [~] [Z-Wave Plus Info] interviewComplete: t [Endpoint 0] [internal]
                      rue => true
15:28:08.918 CNTRLR   [Node 025] MultilevelSwitchCC: doing a partial interview...
15:28:08.918 CNTRLR » [Node 025] requesting current switch state...
15:28:08.922 SERIAL » 0x010900131902260225ae51                                            (11 bytes)
15:28:08.932 DRIVER » [Node 025] [REQ] [SendData]
                      │ transmit options: 0x25
                      │ callback id:      174
                      └─[MultilevelSwitchCCGet]
15:28:08.935 SERIAL « [ACK]                                                                   (0x06)
15:28:08.942 SERIAL « 0x0104011301e8                                                       (6 bytes)
15:28:08.948 SERIAL » [ACK]                                                                   (0x06)
15:28:08.951 DRIVER « [RES] [SendData]
                        was sent: true
15:28:09.096 SERIAL « 0x01180013ae00001001d07f7f7f7f01010315000000010100008d              (26 bytes)
15:28:09.097 SERIAL » [ACK]                                                                   (0x06)
15:28:09.099 DRIVER « [REQ] [SendData]
                        callback id:     174
                        transmit status: OK
15:28:09.221 SERIAL « 0x010d00040019052603636300d0001f                                    (15 bytes)
15:28:09.223 CNTRLR   [Node 025] [~] [Multilevel Switch] targetValue: 99 => 99          [Endpoint 0]
15:28:09.224 CNTRLR   [Node 025] [~] [Multilevel Switch] duration: [Duration: 0seconds] [Endpoint 0]
                       => [Duration: 0seconds]
15:28:09.234 CNTRLR   [Node 025] [~] [Multilevel Switch] currentValue: 99 => 99         [Endpoint 0]
15:28:09.236 SERIAL » [ACK]                                                                   (0x06)
15:28:09.248 DRIVER « [Node 025] [REQ] [ApplicationCommand]
                      └─[MultilevelSwitchCCReport]
                          current value: 99
                          target value:  99
                          duration:      [Duration: 0seconds]
15:28:09.250 CNTRLR   [Node 025] [~] [Multilevel Switch] interviewComplete:  [Endpoint 0] [internal]
                      true => true
15:28:09.255 CNTRLR   [Node 025] ConfigurationCC: Loading configuration parameters from device confi
                      g
15:28:09.265 CNTRLR   [Node 025] [~] [Configuration] isParamInformationFromConfig: true = [internal]
                      > false
15:28:09.266 CNTRLR   [Node 025] [Configuration] 3: metadata updated
15:28:09.267 CNTRLR   [Node 025] [Configuration] 4: metadata updated
15:28:09.268 CNTRLR   [Node 025] [~] [Configuration] isParamInformationFromConfig: false  [internal]
                      => true
15:28:09.270 CNTRLR   [Node 025] ConfigurationCC: doing a complete interview...
15:28:09.281 CNTRLR » [Node 025] finding first configuration parameter...
15:28:09.284 SERIAL » 0x010b00131904700e000025af0e                                        (13 bytes)
15:28:09.285 DRIVER » [Node 025] [REQ] [SendData]
                      │ transmit options: 0x25
                      │ callback id:      175
                      └─[ConfigurationCCPropertiesGet]
                          parameter #: 0
15:28:09.301 SERIAL « [ACK]                                                                   (0x06)
15:28:09.308 SERIAL « 0x0104011301e8                                                       (6 bytes)
15:28:09.314 SERIAL » [ACK]                                                                   (0x06)
15:28:09.316 DRIVER « [RES] [SendData]
                        was sent: true
15:28:09.509 SERIAL « 0x01180013af00001401d07f7f7f7f010103150000000101000088              (26 bytes)
[... 90 lines cut ...]
                      │ transmit options: 0x25
                      │ callback id:      177
                      └─[ConfigurationCCInfoGet]
                          parameter #: 3
15:28:14.795 SERIAL « [ACK]                                                                   (0x06)
15:28:14.802 SERIAL « 0x0104011301e8                                                       (6 bytes)
15:28:14.803 SERIAL » [ACK]                                                                   (0x06)
15:28:14.805 DRIVER « [RES] [SendData]
                        was sent: true
15:28:14.960 SERIAL « 0x01180013b100001001d07f7f7f7f010103150000000101000092              (26 bytes)
15:28:14.961 SERIAL » [ACK]                                                                   (0x06)
15:28:14.963 DRIVER « [REQ] [SendData]
                        callback id:     177
                        transmit status: OK
15:28:15.119 SERIAL « 0x011d0004001915700d00030153657420746865204c4544206c696768d00019    (31 bytes)
15:28:15.121 SERIAL » [ACK]                                                                   (0x06)
15:28:15.123 DRIVER « [Node 025] [REQ] [ApplicationCommand]                                [partial]
                      └─[ConfigurationCCInfoReport]
                          parameter #:       3
                          info:              Set the LED ligh
                          reports to follow: 1
15:28:16.578 CNTRLR   [Node 025] Failed all interview attempts, giving up.
```

At one point, I was able to successfully talk to these switches as I had previously used this library to set up association groups.  However, now I can no longer interview or configure any of them.

RoboPhred, 2020-11-08:
It turns out I still have one of these devices active that does not exhibit the behavior.  An interesting difference is all switches that are failing their interviews have associations, mostly between eachother.  

This device that interviews successfully has no associations.  [Here is it's interview](https://pastebin.com/L4Ri4ZxE)

RoboPhred, 2020-11-09:
Few updates:

I have found another of these switches in the fail state, this time one without any associations set up, so that's out.

I also found one rarely used switch that never got included into my network.  I included it, and played around with it to confirm it was interviewing properly.  I then left it alone for several hours, and when I came back to it, the switch was now in the failure state / failing interviews.

I also contacted a friend who has a home full of these things on a SmartThings hub which has been established for over a year.  They report no issues at all with these switches, so either the SmartThings hub is not triggering this behavior, or more likely doesn't care about failures in the interview.  I am going to try and prod them to check if they can get / set the LED status configuration value, as that is the parameter that seems to be failing.  Unfortunately, they don't want me messing with their network so I cannot attach a controller in slave mode and attempt to interview it directly.

AlCalzone, 2020-11-09:
What you're seeing here is that the device responds to the request, but only with a partial message:
```
15:28:12.813 DRIVER « [Node 025] [REQ] [ApplicationCommand]                                [partial]
                      └─[ConfigurationCCInfoReport]
                          parameter #:       3
                          info:              Set the LED ligh
                          reports to follow: 1
15:28:14.268 CNTRLR   [Node 025] did not respond after 2/3 attempts. Scheduling next try in 500 ms.
```
The info text is not complete and the response indicates that another message with the rest follows - but then does not. Maybe the message got lost on the way - we cannot be sure.

Since the parameter description is non-critical, the driver should just ignore the query. I have this in place in several other CCs, but I think this is the first log of a device that actually supports these commands.

closed 2020-11-09 by AlCalzone

AlCalzone, 2020-11-14:
This is included in `v5.4.0`

PR #1108 (2020-11-09): Ignore node timeouts for name/info query in ConfigurationCC V3+
fixes: #1106 by ignoring incomplete name/info reports