zwave-js/zwave-js #1107: zwave-js@5 times out on Enbrighten Z-Wave Plus Smart Fan Control MultilevelSwitch set, zwave-js@2 does not.

RoboPhred, 2020-11-09:
I have several [GE Enbrighten Fan Controls](https://www.amazon.com/GE-Enbrighten-Required-SmartThings-14287/dp/B06XTKQTTV/).  These devices previously worked fine with zwave-js, but none of them will respond (both in actuation or over the zwave protocol) to Multilevel Switch set commands.

Here is a log of me changing the fan with zwave-js@2.15.7
```
22:24:53.042 DRIVER » [Node 031] [REQ] [SendData]                                        [P: Normal]
                      │ transmitOptions: 0x25
                      │ callbackId:      74
                      └─[MultilevelSwitchCCSet]
                          payload: 0x63
22:24:53.044 SERIAL » 0x010a00131f03260163254ad1                                          (12 bytes)
22:24:53.059 SERIAL « [ACK]                                                                   (0x06)
22:24:53.060 DRIVER   ACK received from controller for current transaction
22:24:53.061 SERIAL   The receive buffer is empty, waiting for the next chunk...
22:24:53.065 SERIAL « 0x01                                                                 (1 bytes)
22:24:53.066 SERIAL   [incomplete] Buffer := 0x01                                          (1 bytes)
22:24:53.067 SERIAL « 0x04011301e8                                                         (5 bytes)
22:24:53.069 SERIAL   Buffer := 0x0104011301e8                                             (6 bytes)
22:24:53.070 SERIAL » [ACK]                                                                   (0x06)
22:24:53.071 DRIVER « [RES] [SendData]                                                [confirmation]
                        wasSent: true
22:24:53.073 SERIAL   The receive buffer is empty, waiting for the next chunk...
22:24:53.109 SERIAL « 0x01                                                                 (1 bytes)
22:24:53.110 SERIAL   [incomplete] Buffer := 0x01                                          (1 bytes)
22:24:53.111 SERIAL « 0x1800134a00000401c77f7f7f7f000003290000000301000054                (25 bytes)
22:24:53.121 SERIAL   Buffer := 0x011800134a00000401c77f7f7f7f000003290000000301000054    (26 bytes)
22:24:53.122 SERIAL » [ACK]                                                                   (0x06)
22:24:53.123 DRIVER « [REQ] [SendData]                                                       [final]
                        callbackId:     74
                        transmitStatus: OK
22:24:53.124 DRIVER   ACK already received, resolving transaction
22:24:53.125 DRIVER   resuming send queue
22:24:53.126 SERIAL   The receive buffer is empty, waiting for the next chunk...
22:24:53.127 DRIVER » [Node 031] [REQ] [SendData]                                        [P: Normal]
                      │ transmitOptions: 0x25
                      │ callbackId:      75
                      └─[MultilevelSwitchCCGet]
22:24:53.139 SERIAL » 0x010900131f022602254bb2                                            (11 bytes)
22:24:53.141 DRIVER   workOffSendQueue > skipping because a transaction is pending
22:24:53.143 SERIAL « [ACK]                                                                   (0x06)
22:24:53.145 DRIVER   ACK received from controller for current transaction
22:24:53.157 SERIAL   The receive buffer is empty, waiting for the next chunk...
22:24:53.158 SERIAL « 0x0104011301e8                                                       (6 bytes)
22:24:53.159 SERIAL   Buffer := 0x0104011301e8                                             (6 bytes)
22:24:53.160 SERIAL » [ACK]                                                                   (0x06)
22:24:53.161 DRIVER « [RES] [SendData]                                                [confirmation]
                        wasSent: true
22:24:53.173 SERIAL   The receive buffer is empty, waiting for the next chunk...
22:24:53.193 SERIAL « 0x01                                                                 (1 bytes)
22:24:53.194 SERIAL   [incomplete] Buffer := 0x01                                          (1 bytes)
22:24:53.206 SERIAL « 0x1800134b00000501c77f7f7f7f000003290000000301000054                (25 bytes)
22:24:53.207 SERIAL   Buffer := 0x011800134b00000501c77f7f7f7f000003290000000301000054    (26 bytes)
22:24:53.208 SERIAL » [ACK]                                                                   (0x06)
22:24:53.209 DRIVER « [REQ] [SendData]                                                [confirmation]
                        callbackId:     75
                        transmitStatus: OK
22:24:53.210 DRIVER   ACK received from node for current transaction. RTT = 82.98 ms
22:24:53.211 SERIAL   The receive buffer is empty, waiting for the next chunk...
22:24:53.268 SERIAL « 0x01                                                                 (1 bytes)
22:24:53.269 SERIAL   [incomplete] Buffer := 0x01                                          (1 bytes)
22:24:53.270 SERIAL « 0x0b0004001f03260363c6006c                                          (12 bytes)
22:24:53.281 SERIAL   Buffer := 0x010b0004001f03260363c6006c                              (13 bytes)
22:24:53.282 CNTRLR   [Node 031] [~] [Multilevel Switch] currentValue: 0 => 99          [Endpoint 0]
22:24:53.283 SERIAL » [ACK]                                                                   (0x06)
22:24:53.284 DRIVER « [Node 031] [REQ] [ApplicationCommand]                                  [final]
                      └─[MultilevelSwitchCCReport]
                          payload: 0x63
22:24:53.287 DRIVER   ACK already received, resolving transaction
```
The fan switch responds appropriately to the commands.  zwave-js receives the appropriate response, and the fan starts correctly.

[... 61 lines cut ...]
ZWaveError: Timed out while waiting for a response from the node
    at Object.sendDataErrorToZWaveError (D:\dev\git\RoboPhred\wutwot\node_modules\zwave-js\src\lib\driver\StateMachineShared.ts:134:11)
    at D:\dev\git\RoboPhred\wutwot\node_modules\zwave-js\src\lib\driver\SendThreadMachine.ts:421:4
    at D:\dev\git\RoboPhred\wutwot\node_modules\xstate\lib\utils.js:409:33
    at Array.reduce (<anonymous>)
    at Object.updateContext (D:\dev\git\RoboPhred\wutwot\node_modules\xstate\lib\utils.js:399:25)
    at Object.resolveActions (D:\dev\git\RoboPhred\wutwot\node_modules\xstate\lib\actions.js:405:19)
    at StateNode.resolveTransition (D:\dev\git\RoboPhred\wutwot\node_modules\xstate\lib\StateNode.js:787:35)
    at StateNode.transition (D:\dev\git\RoboPhred\wutwot\node_modules\xstate\lib\StateNode.js:733:21)
    at D:\dev\git\RoboPhred\wutwot\node_modules\xstate\lib\interpreter.js:600:34
    at Object.exports.provide (D:\dev\git\RoboPhred\wutwot\node_modules\xstate\lib\serviceScope.js:11:18) {
  code: 17,
  context: 'Transaction: \n' +
    '    at Driver.sendMessage (D:\\dev\\git\\RoboPhred\\wutwot\\node_modules\\zwave-js\\src\\lib\\driver\\Driver.ts:1908:23)\n' +
    '    at Driver.sendCommand (D:\\dev\\git\\RoboPhred\\wutwot\\node_modules\\zwave-js\\src\\lib\\driver\\Driver.ts:2007:27)\n' +
    '    at Driver.sendSupervisedCommand (D:\\dev\\git\\RoboPhred\\wutwot\\node_modules\\zwave-js\\src\\lib\\driver\\Driver.ts:2044:28)\n' +
    '    at Driver.trySendCommandSupervised (D:\\dev\\git\\RoboPhred\\wutwot\\node_modules\\zwave-js\\src\\lib\\driver\\Driver.ts:2074:16)\n' +
    '    at MultilevelSwitchCCAPI.set (D:\\dev\\git\\RoboPhred\\wutwot\\node_modules\\zwave-js\\src\\lib\\commandclass\\MultilevelSwitchCC.ts:149:47)\n' +
    '    at Proxy.MultilevelSwitchCCAPI.<computed> (D:\\dev\\git\\RoboPhred\\wutwot\\node_modules\\zwave-js\\src\\lib\\commandclass\\MultilevelSwitchCC.ts:262:15)\n' +
    '    at ZWaveNode.setValue (D:\\dev\\git\\RoboPhred\\wutwot\\node_modules\\zwave-js\\src\\lib\\node\\Node.ts:626:14)\n' +
    '    at MultilevelSwitchMonitorImpl.<anonymous> (D:\\dev\\git\\RoboPhred\\wutwot\\packages\\wutwot-zwave\\src\\impl\\monitors\\MultilevelSwitchMonitorFactoryImpl.ts:69:28)\n' +
    '    at Generator.next (<anonymous>)\n' +
    '    at D:\\dev\\git\\RoboPhred\\wutwot\\packages\\wutwot-zwave\\lib\\impl\\monitors\\MultilevelSwitchMonitorFactoryImpl.js:14:71'
```
The command is never acknowledged by the switch, and the fan does not start.

AlCalzone, 2020-11-09:
Could you show me a log of an interview for that device? I'm curious if the NIF includes Supervision CC or if this CC got in the list differently somehow. If it is included in the NIF, we do need a blacklist.

RoboPhred, 2020-11-09:
Here is the full interview:

https://pastebin.com/9Q0xfV5F

Suprisingly, the Supervision CC isn't being reported from it.  I suppose the cert docs I found were for a later model:
```
10:16:19.189 CNTRLR « [Node 036] node info received
                      supported CCs:
                      · Z-Wave Plus Info
                      · Version
                      · Manufacturer Specific
                      · Device Reset Locally
                      · Association
                      · Association Group Information
                      · Powerlevel
                      · Multilevel Switch
                      · All Switch
                      · Configuration
                      · CRC-16 Encapsulation
                      · Scene Actuator Configuration
                      · Scene Activation
                      · Firmware Update Meta Data
                      controlled CCs:
10:16:19.195 CNTRLR   [Node 036] Interview stage completed: NodeInfo
```

I dont see any mention of supervision anywhere in the interview, so I am not sure where that CC is being applied from.

AlCalzone, 2020-11-09:
> I dont see any mention of supervision anywhere in the interview, so I am not sure where that CC is being applied from.

I don't either. Judging from the interview I would assume that Supervision is not used for the Multilevel Switch Set commands of Node 36.

RoboPhred, 2020-11-09:
The plot thickens!

It turns out, manually doing an interview on these nodes fixes them.  The second interview on any of these devices does NOT return Supervision as a supported CC, and after that second interview, I can control them just fine.

Here is a log showing this.  I first try to send a command, and watch it fail to use the Supervision CC.  Then I run a full interview, and wait for it to complete.  After the interview, I try to control it again, and this time it works.

https://pastebin.com/CsPZ4BeF

I am going to try and capture a log restricted to only this node, to see what the initial interview is doing.  I have seen an env var in the code to do this, but I need to track it down again...

RoboPhred, 2020-11-09:
I had to clear my cache to get the bug to occur.

Here is the **first** NIF reported, from a fresh startup
```
13:54:30.454 CNTRLR   [Node 034] The node is alive.
13:54:30.456 CNTRLR « [Node 034] node info received
                      supported CCs:
                      · Z-Wave Plus Info
                      · Multilevel Switch
                      · Association
                      · Association Group Information
                      · Version
                      · Manufacturer Specific
                      · Transport Service
                      · Device Reset Locally
                      · Powerlevel
                      · Central Scene
                      · Security 2
                      · Supervision
                      · Configuration
                      · Application Status
                      · Firmware Update Meta Data
                      controlled CCs:
```
Supervision is clearly there.

...And if I call refreshInfo, this comes back:
```
14:00:16.891 CNTRLR « [Node 034] node info received
                      supported CCs:
                      · Z-Wave Plus Info
                      · Version
                      · Manufacturer Specific
                      · Device Reset Locally
                      · Association
                      · Association Group Information
                      · Powerlevel
                      · Multilevel Switch
                      · All Switch
                      · Configuration
                      · CRC-16 Encapsulation
                      · Scene Actuator Configuration
                      · Scene Activation
                      · Firmware Update Meta Data
                      controlled CCs:
```

Really weird.

I have a log for a full interview of my network with the cache files deleted.  I was not able to figure out how to restrict the log to only one node unfortunately, so the log is too large for pastebin.  I had to resort to a zip file in google drive: https://drive.google.com/file/d/1q91R3H4VI-HIdCG-dZ2ME8_JUzTnVaTD/view?usp=sharing

The order of operations on this log are:
- Start a fresh no-cache instance and wait for all of the interviews to settle down
- Try to command node 34, watch it fail
- Call refreshInfo on node 34, wait for it to complete
- Try to command node 34, watch it succeed

AlCalzone, 2020-11-09:
Ok let me know. The interviews don't do anything differently, so I wonder where the CC is coming from

RoboPhred, 2020-11-09:
Sorry, I have a habit of posting first then editing my posts to include more info.  The previous post was updated to include the full logs, and has notes that more than one CC is disappearing, and a few that don't come across in the first NIF start to show up.

AlCalzone, 2020-11-09:
That is a completelY different set of Ccs 🤨

RoboPhred, 2020-11-09:
This seems to be a problem unique to zwave-js@5!

I still had my old library sitting around with zwave-js@2, and tried running it again to see what happens over there.  This is the **first** NIF I receive, when doing a full interview with no previous cache
```
14:41:52.257 CNTRLR « [Node 034] node info received
                      supported CCs:
                      · Z-Wave Plus Info
                      · Version
                      · Manufacturer Specific
                      · Device Reset Locally
                      · Association
                      · Association Group Information
                      · Powerlevel
                      · Multilevel Switch
                      · All Switch
                      · Configuration
                      · CRC-16 Encapsulation
                      · Scene Actuator Configuration
                      · Scene Activation
                      · Firmware Update Meta Data
```

Identical to the second requestInfo run on @5.  I think zwave-js@5 must be mixing up messages somewhere when all of the nodes are interviewing at once.

AlCalzone, 2020-11-09:
You are right - the NIF at 13:54 is the one of Node 033 but that seems to be matched to the request of Node 034. I think this might be another instance of a really sneaky bug I've been trying to hunt down where entirely wrong messages are being matched together.
I haven't been able to reproduce this in a test case, so every breadcrumb (like this one) might help.

RoboPhred, 2020-11-09:
I seem to have a reliable way of reproducing it with my network.  Is there any way I can help track down this bug?  I could shove in debug code into zwave-js to try and track what is going on.  Just let me know where to start investigating.

Interesting that it reliably effects every device of this specific type.  I wonder if it is happening to other devices too, but I just don't notice because the CCs all work out anyway.

AlCalzone, 2020-11-09:
I'll try to reproduce what is happening in your log with some end-to-end tests. Now that I have a specific set of messages that triggers this, it should be easier. From other reports I've seen, I'm pretty sure this bug also affects other devices.

AlCalzone, 2020-11-09:
Actually, you could do a test for me. Please create your driver instance with the following options:
```ts
const driver = new Driver("/your/serialport", {
  timeouts: { nodeAwake: Infinity },
  // ... your driver options
});
```
entirely remove this check:
```ts
// zwave-js/build/lib/driver/Driver.js, lines 136-139
    if (options.timeouts.nodeAwake < 1000 ||
        options.timeouts.nodeAwake > 30000) {
        throw new core_1.ZWaveError(`The Node awake timeout must be between 1000 and 30000 milliseconds!`, core_1.ZWaveErrorCodes.Driver_InvalidOptions);
    }
```
and try to reproduce the issue with this setup.

RoboPhred, 2020-11-09:
A timeout of Inifinity makes the Timeout object unhappy:
```
(node:220) TimeoutOverflowWarning: Infinity does not fit into a 32-bit signed integer.
Timeout duration was set to 1.
(Use `node --trace-warnings ...` to show where the warning was created)
(node:220) TimeoutOverflowWarning: Infinity does not fit into a 32-bit signed integer.
Timeout duration was set to 1.
(node:220) TimeoutOverflowWarning: Infinity does not fit into a 32-bit signed integer.
Timeout duration was set to 1.
(node:220) TimeoutOverflowWarning: Infinity does not fit into a 32-bit signed integer.
Timeout duration was set to 1.
(node:220) TimeoutOverflowWarning: Infinity does not fit into a 32-bit signed integer.
Timeout duration was set to 1.
```

I will keep running the test like this, then try to run it with the max 32 bit integer instead.

RoboPhred, 2020-11-09:
That seems to have been successful.  I now get the proper data on the first interview attempt:
```
16:26:29.378 CNTRLR « [Node 034] node info received
                      supported CCs:
                      · Z-Wave Plus Info
                      · Version
                      · Manufacturer Specific
                      · Device Reset Locally
                      · Association
                      · Association Group Information
                      · Powerlevel
                      · Multilevel Switch
                      · All Switch
                      · Configuration
                      · CRC-16 Encapsulation
                      · Scene Actuator Configuration
                      · Scene Activation
                      · Firmware Update Meta Data
```

The other two fan controllers seem to likewise have their proper CCs applied.

AlCalzone, 2020-11-10:
Ok then I think I know whats going on. When a node is marked as asleep because its awake timer has run out, the state machine responsible for sending seems to get confused.

AlCalzone, 2020-11-10:
This is the original source of the problem:

```
13:54:05.964 SERIAL » 0x010800130a01002503c9                                              (10 bytes)
13:54:05.966 DRIVER » [Node 010] [REQ] [SendData]
                      │ transmit options: 0x25
                      │ callback id:      3
                      └─[NoOperationCC]
[...]
13:54:05.973 SERIAL « [ACK]                                                                   (0x06)
13:54:05.981 SERIAL « 0x0104011301e8                                                       (6 bytes)
13:54:05.984 SERIAL » [ACK]                                                                   (0x06)
13:54:05.987 DRIVER « [RES] [SendData]
                        was sent: true
13:54:07.078 DRIVER   The awake timeout for node 10 has elapsed. Assuming it is asleep.
13:54:07.079 CNTRLR   [Node 010] The node is now asleep.
13:54:07.082 CNTRLR   [Node 010] ping failed: The node is asleep
[...]
13:54:10.087 SERIAL « 0x011800130301019b007f7f7f7f7f010107000000000204000012              (26 bytes)
13:54:10.089 SERIAL » [ACK]                                                                   (0x06)
13:54:10.092 DRIVER « [REQ] [SendData]
                        callback id:     3
                        transmit status: NoAck
13:54:10.096 SERIAL » 0x010800131101002504d5                                              (10 bytes)
13:54:10.098 DRIVER » [Node 017] [REQ] [SendData]
                      │ transmit options: 0x25
                      │ callback id:      4
                      └─[NoOperationCC]
13:54:10.101 CNTRLR   [Node 017] Node 17 did not respond after 1 attempts, it is presumed dead
13:54:10.102 CNTRLR   [Node 017] The node is dead.
```
Node 10 is assumed asleep before the stick calls back with the transmit report. Then Node 17 is pinged, which immediately fails because the previous callback is matched to the new transaction (despite the wrong callback id). From there it goes downhill.

AlCalzone, 2020-11-11:
This should now be fixed on master

closed 2020-11-11 by AlCalzone

AlCalzone, 2020-11-14:
This is included in `v5.4.0`

PR #1110 (2020-11-11): Do not mismatch messages when a node with a pending transaction is marked asleep
So, this is a sneaky one. The bug was an xstate transaction that would re-enter the `sending` state, thus re-triggering the entry action.

Also, if a node goes to sleep and its transaction is discarded while waiting for a response, we need to abort the ongoing `Send Data` API call to avoid overlapping transactions or waiting unnecessary long.

fixes: #1107