zwave-js/zwave-js #1114: Interview incomplete for battery device Sensative Strips Comfort

larstobi, 2020-11-12:
When using the  zwavejs2mqtt service, I can never get it to complete the interview with battery nodes. In this example, a Sensative Strips Comfort.

I woke up the device, then chose Refresh info, and this is what happened in the log:
```
  z2m:Zwave Node 22: notification: Tampering, invalid code  +10s
  z2m:Zwave Node 22 is now awake +55ms
  z2m:App Zwave api call: refreshInfo [ 22 ] +32s
  z2m:Zwave Success zwave api call refreshInfo  +177ms
  z2m:Zwave Node 22: metadata updated: 49-0-Air temperature +11s
  z2m:Zwave Node 22: value added: 49-0-Air temperature => 26.5 +1ms
  z2m:Zwave Node 22 is now asleep +6s
```

The UI still doesn't show the Manufacturer, Product or Productcode, nor are any Configuration parameters available.

AlCalzone, 2020-11-12:
I would need to see a logfile from zwave-js to know whats going on here. It **might** be that you are running into #1107 which I just fixed yesterday on master.

larstobi, 2020-11-14:
@AlCalzone 
I have built zwavejs2mqtt from the latest master of node-zwave-js and tested it without success.
I got you the log file in a gist: https://gist.github.com/larstobi/96ccd8a3ffd4b115f72def01a60be619

larstobi, 2020-11-14:
Node 22 is still a Sensative Strips Comfort.

larstobi, 2020-11-14:
```
  z2m:Zwave Node 22 is asleep +8s
  z2m:Zwave Node 22: metadata updated: 49-0-Illuminance +883ms
  z2m:Zwave Node 22: value updated: 49-0-Illuminance 1 => 25 +1ms
  z2m:Zwave Node 22: metadata updated: 49-0-Illuminance +1s
  z2m:Zwave Node 22: value updated: 49-0-Illuminance 25 => 1 +1ms
  z2m:Zwave Node 22 is asleep +2s
  z2m:Zwave Node 22: metadata updated: 49-0-Air temperature +6s
  z2m:Zwave Node 22: value updated: 49-0-Air temperature 25.9 => 24.9 +0ms
  z2m:Zwave Node 22: metadata updated: 49-0-Air temperature +45ms
  z2m:Zwave Node 22: value updated: 49-0-Air temperature 24.9 => 24.9 +0ms
  z2m:Zwave Node 22: metadata updated: 49-0-Air temperature +1s
  z2m:Zwave Node 22: value updated: 49-0-Air temperature 24.9 => 24.9 +1ms
  z2m:Zwave Node 22 is asleep +8s
  z2m:Zwave Node 22 is asleep +7s
  z2m:Zwave Node 22 is asleep +7s
  z2m:Zwave Node 22 is asleep +3s
  z2m:Zwave Node 22 is asleep +3s
  z2m:Zwave Node 22 is asleep +3s
  z2m:Zwave Node 22 is asleep +7s
  z2m:Zwave Node 22 is asleep +7s
  z2m:Zwave Node 22 is asleep +4s
  z2m:Zwave Node 22 is asleep +5s
```

AlCalzone, 2020-11-14:
Sorry that log doesn't really help, I only need the zwave-js part (without lines that start with `  z2m:Zwave`). AFAIK all you need to do is this:

>  * Go to Settings, Zwave section
>  * select log level (preferably DEBUG or VERBOSE if the files are getting too large)
>  * enable "log to file"

and send me the created file (should be called `zwave-<processid>.log`).

larstobi, 2020-11-14:
@AlCalzone Here you go:
https://gist.github.com/larstobi/a644872f2cfb2b96f5a0fb2c1a0d9e5d

AlCalzone, 2020-11-15:
It looks like you leave the driver running for a few minutes only. Some battery-powered devices take hours to wake up if you do not wake them up manually.
I do not see anything in the log that looks like the device ever woke up, so either double-check that your manual wakeup procedure is correct, or have some patience.

larstobi, 2020-11-15:
The driver was running for two hours and ten minutes. The OpenZwave driver is usually able to poll it within that time.

By the way, Node 052 and Node 053 are both Sensative Strips Guard, and they woke up during that time and reported Door State. They still didn't get past the ProtocolInfo step.

I'll try again with Node 022 and manually wake it up.

AlCalzone, 2020-11-15:
> The driver was running for two hours and ten minutes

```
22:02:37.387 DRIVER   ███████╗ ██╗    ██╗  █████╗  ██╗   ██╗ ███████╗             ██╗ ███████╗
                      ╚══███╔╝ ██║    ██║ ██╔══██╗ ██║   ██║ ██╔════╝             ██║ ██╔════╝
                        ███╔╝  ██║ █╗ ██║ ███████║ ██║   ██║ █████╗   █████╗      ██║ ███████╗
                       ███╔╝   ██║███╗██║ ██╔══██║ ╚██╗ ██╔╝ ██╔══╝   ╚════╝ ██   ██║ ╚════██║
                      ███████╗ ╚███╔███╔╝ ██║  ██║  ╚████╔╝  ███████╗        ╚█████╔╝ ███████║
                      ╚══════╝  ╚══╝╚══╝  ╚═╝  ╚═╝   ╚═══╝   ╚══════╝         ╚════╝  ╚══════╝
22:02:37.424 DRIVER   version 5.3.6
22:02:37.425 DRIVER   
22:02:37.425 DRIVER   starting driver...
```
I'm seeing this **fourteen times** in your log, which means that the driver wasn't running 2 hours non-stop.
It might be that you're affected by an issue I've fixed in version `5.4.0` - it is probably best if you update before doing any more tests.

> By the way, Node 052 and Node 053 are both Sensative Strips Guard, and they woke up during that time and reported Door State.

Yes, they sent messages but they did not wake up. When a device wakes up, it sends the controller a special message and actively listens for commands for a short time. Many battery-powered devices send notifications/unsolicited messages while still asleep.

larstobi, 2020-11-15:
Okay, I didn't catch that 14 times starting driver... I left the zwavejs2mqtt running for over two hours, in fact. Does that mean the driver crashed fourteen times?

In any case, I have updated to the lastest master and did a new run where I woke Node 022 several times, but it never came beyond ProtocolInfo. The Strips is awake for ten seconds before it sleeps.

larstobi, 2020-11-15:
It's `zwave-3.log` in the same gist:
https://gist.github.com/larstobi/a644872f2cfb2b96f5a0fb2c1a0d9e5d

It looks like it waits for a ping reply that it doesn't get, and then the Strips falls asleep.

```
18:59:21.512 DRIVER « [Node 022] [REQ] [ApplicationCommand]
18:59:21.583 DRIVER « [Node 022] [REQ] [ApplicationUpdateRequest]
18:59:21.584 CNTRLR « [Node 022] Received updated node info
18:59:21.585 CNTRLR   [Node 022] The node is now awake.
18:59:31.013 CNTRLR   [Node 022] Beginning interview - last completed stage: None
18:59:31.013 CNTRLR   [Node 022] new node, doing a full interview...
18:59:31.013 CNTRLR » [Node 022] querying protocol info...
18:59:36.957 CNTRLR « [Node 022] received response for protocol info:
18:59:36.983 CNTRLR   [Node 022] Interview stage completed: ProtocolInfo
18:59:36.983 CNTRLR » [Node 022] pinging the node...
18:59:46.960 CNTRLR   [Node 022] The node is now asleep.
18:59:46.962 CNTRLR   [Node 022] ping failed: The node is asleep
```

AlCalzone, 2020-11-15:
Thanks I'll check it out tomorrow. The logs look pretty messed up on mobile.

I'm guessing that either the ping isn't getting prioritized correctly or should even be left out completely since we know the node is awake.

AlCalzone, 2020-11-16:
Regarding the crashes: If you have a way to see the error message, it would be helpful.
It seems that the bug I mentioned is not completely fixed, but I need a log where it happens to help me reproduce it.

robertsLando, 2020-11-16:
Fixed in d8a26dab5fe5ded81d7ce64f252c1b04c7f073ec

robertsLando, 2020-11-16:
@larstobi Feel free to close this if the bug is solved :)

larstobi, 2020-11-16:
I'll rebuild and do another test run! :-)

AlCalzone, 2020-11-16:
There's still something fishy with the interview:
```
18:59:21.585 CNTRLR   [Node 022] The node is now awake.
18:59:31.013 CNTRLR   [Node 022] Beginning interview - last completed stage: None
```
From node awake until the first message is sent takes 10s - by that time the node is asleep again.

larstobi, 2020-11-16:
[image]

larstobi, 2020-11-16:
At least it was able to read Manufacturer and Product ID when I manually woke it up! :-)

AlCalzone, 2020-11-16:
@robertsLando I think how you display the interview stage might lead to some confusion. The value does not tell you which stage the interview is in, but rather which stage was completed last.

larstobi, 2020-11-16:
[image]

larstobi, 2020-11-16:
One wake-up later it got Complete.

larstobi, 2020-11-16:
Also only one occurence of the driver starting! 🎉

I added the latest debug log in https://gist.github.com/larstobi/a644872f2cfb2b96f5a0fb2c1a0d9e5d
as zwave-4.log 

I can see that Node 009 is having a hard time being Interviewed. It's a Heatit Z-TRM2 device. Maybe it deserves its own new github issue?

robertsLando, 2020-11-16:
> I think how you display the interview stage might lead to some confusion. The value does not tell you which stage the interview is in, but rather which stage was completed last

Isn't this good? that's the actual stage

robertsLando, 2020-11-16:
> Also only one occurence of the driver starting! tada

That was my fault, unfortunally I didn't know I could have an `undefined` deviceCOnfig in zwaveNode :confounded:

AlCalzone, 2020-11-16:
> Isn't this good? that's the actual stage

Well, if the last completed stage is `NodeInfo`, then the driver is currently interviewing the CommandClasses, **after which** the stage gets set to `CommandClasses`.

closed 2020-11-17 by larstobi

robertsLando, 2020-11-17:
@AlCalzone How do I get the current one so? there should be an event emitted from node like `interview update` ?

AlCalzone, 2020-11-17:
Not trivial, the property was not intended to do that.
You could do:
* If `Complete`, use that
* If not, add 1. If the result is `RestartFromCache`, add 1 more. Display the result

But no guarantees that this won't break some time in the future.

robertsLando, 2020-11-17:
> But no guarantees that this won't break some time in the future.

Humm need to think about that...