zwave-js/zwave-js #1115: [question] Adding device is not discovering which device is being used

crxporter, 2020-11-12:
I'm trying to add a zooz 4-in-1 sensor. It's adding fine but there are a few errors in the log. The device is communicating properly (motion shows in the log as expected) but the device name is not discovered.

I've included, then excluded, then included again the device. No change. I've also restarted the program.

Do I need to add to the device database like in openzwave times or is there something else going on?

Let me know if I can add any more helpful information. I could increase the log level if needed. 
Hardware:
-zooz controller
-zwavejs2mqtt pulled from github today, installed with npm
-running on raspberry pi 4

Logs when including the device:
```
  z2m:Zwave Secure inclusion started +14s
  z2m:Zwave Success zwave api call startInclusion true +1ms
  z2m:Zwave Inclusion stopped +2s
  z2m:Zwave Node 6: added +240ms
  z2m:Zwave Node added 6 +1ms
  z2m:Zwave Node 6 is awake +9ms
  z2m:Zwave Node 6: value added: 114-0-manufacturerId => 634 +466ms
  z2m:Zwave Node 6: value added: 114-0-productType => 8225 +1ms
  z2m:Zwave Node 6: value added: 114-0-productId => 8449 +1ms
  z2m:Zwave Node 6: value added: 134-0-libraryType => 3 +190ms
  z2m:Zwave Node 6: value added: 134-0-protocolVersion => 4.54 +3ms
  z2m:Zwave Node 6: value added: 134-0-firmwareVersions => [ '32.2' ] +1ms
16:10:57.380 CNTRLR   [Node 006] did not respond after 1/3 attempts. Scheduling next try in 500 ms.
  z2m:Zwave Node 6: value added: 94-0-zwavePlusVersion => 1 +4s
  z2m:Zwave Node 6: value added: 94-0-nodeType => 0 +1ms
  z2m:Zwave Node 6: value added: 94-0-roleType => 6 +1ms
  z2m:Zwave Node 6: value added: 94-0-installerIcon => 3079 +0ms
  z2m:Zwave Node 6: value added: 94-0-userIcon => 3079 +1ms
  z2m:Zwave Node 6: value added: 128-0-level => 100 +181ms
  z2m:Zwave Node 6: value added: 128-0-isLow => false +1ms
  z2m:Zwave Node 6: metadata updated: 132-0-wakeUpInterval +161ms
  z2m:Zwave Node 6: value added: 132-0-wakeUpInterval => 43200 +160ms
  z2m:Zwave Node 6: value added: 132-0-controllerNodeId => 255 +2ms
  z2m:Zwave Node 6: metadata updated: 49-0-Air temperature +611ms
  z2m:Zwave Node 6: value added: 49-0-Air temperature => 27.32 +1ms
  z2m:Zwave Node 6: metadata updated: 49-0-Illuminance +322ms
  z2m:Zwave Node 6: value added: 49-0-Illuminance => 100 +1ms
16:11:01.848 CNTRLR   [Node 006] did not respond after 1/3 attempts. Scheduling next try in 500 ms.
  z2m:Zwave Node 6: metadata updated: 49-0-Humidity +3s
  z2m:Zwave Node 6: value added: 49-0-Humidity => 49.22 +1ms
  z2m:Zwave Node 6: metadata updated: 112-0-1 +7ms
  z2m:Zwave Node 6: metadata updated: 112-0-2 +2ms
  z2m:Zwave Node 6: metadata updated: 112-0-3 +3ms
  z2m:Zwave Node 6: metadata updated: 112-0-4 +1ms
  z2m:Zwave Node 6: metadata updated: 112-0-5 +4ms
  z2m:Zwave Node 6: metadata updated: 112-0-6 +1ms
  z2m:Zwave Node 6: metadata updated: 112-0-7 +1ms
  z2m:Zwave Node 6: value added: 112-0-1 => 1 +162ms
  z2m:Zwave Node 6: value added: 112-0-2 => 10 +160ms
  z2m:Zwave Node 6: value added: 112-0-3 => 10 +165ms
  z2m:Zwave Node 6: value added: 112-0-4 => 10 +164ms
  z2m:Zwave Node 6: value added: 112-0-5 => 15 +167ms
  z2m:Zwave Node 6: value added: 112-0-6 => 3 +164ms
  z2m:Zwave Node 6: value added: 112-0-7 => 4 +163ms
16:11:05.290 DRIVER   Could not assemble partial CCs because the payload is invalid. Dropping them.
16:11:06.822 CNTRLR   [Node 006] did not respond after 1/3 attempts. Scheduling next try in 500 ms.
16:11:07.492 DRIVER   Could not assemble partial CCs because the payload is invalid. Dropping them.
16:11:09.025 CNTRLR   [Node 006] did not respond after 2/3 attempts. Scheduling next try in 500 ms.
16:11:09.693 DRIVER   Could not assemble partial CCs because the payload is invalid. Dropping them.
16:11:11.229 CNTRLR   [Node 006] Interview attempt 1/5 failed, retrying in 5000 ms...
  z2m:Zwave Node 6: value updated: 128-0-level 100 => 100 +13s
  z2m:Zwave Node 6: value updated: 128-0-isLow false => false +1ms
  z2m:Zwave Node 6: value updated: 132-0-wakeUpInterval 43200 => 43200 +170ms
  z2m:Zwave Node 6: value updated: 132-0-controllerNodeId 255 => 1 +1ms
  z2m:Zwave Node 6: metadata updated: 49-0-Air temperature +161ms
  z2m:Zwave Node 6: value updated: 49-0-Air temperature 27.32 => 27.32 +1ms
  z2m:Zwave Node 6: metadata updated: 49-0-Illuminance +160ms
  z2m:Zwave Node 6: value updated: 49-0-Illuminance 100 => 100 +1ms
  z2m:Zwave Node 6: metadata updated: 49-0-Humidity +161ms
  z2m:Zwave Node 6: value updated: 49-0-Humidity 49.22 => 49.22 +1ms
  z2m:Zwave Node 6: metadata updated: 112-0-1 +6ms
  z2m:Zwave Node 6: metadata updated: 112-0-2 +1ms
  z2m:Zwave Node 6: metadata updated: 112-0-3 +1ms
  z2m:Zwave Node 6: metadata updated: 112-0-4 +1ms
  z2m:Zwave Node 6: metadata updated: 112-0-5 +1ms
  z2m:Zwave Node 6: metadata updated: 112-0-6 +1ms
  z2m:Zwave Node 6: metadata updated: 112-0-7 +0ms
  z2m:Zwave Node 6: value updated: 112-0-1 1 => 1 +170ms
  z2m:Zwave Node 6: value updated: 112-0-2 10 => 10 +164ms
  z2m:Zwave Node 6: value updated: 112-0-3 10 => 10 +158ms
  z2m:Zwave Node 6: value updated: 112-0-4 10 => 10 +154ms
  z2m:Zwave Node 6: value updated: 112-0-5 15 => 15 +154ms
  z2m:Zwave Node 6: value updated: 112-0-6 3 => 3 +168ms
  z2m:Zwave Node 6: value updated: 112-0-7 4 => 4 +158ms
16:11:21.099 CNTRLR   [Node 006] did not respond after 1/3 attempts. Scheduling next try in 500 ms.
16:11:21.765 DRIVER   Could not assemble partial CCs because the payload is invalid. Dropping them.
16:11:23.296 CNTRLR   [Node 006] did not respond after 2/3 attempts. Scheduling next try in 500 ms.
16:11:24.037 DRIVER   Could not assemble partial CCs because the payload is invalid. Dropping them.
16:11:25.492 CNTRLR   [Node 006] Interview attempt 2/5 failed, retrying in 10000 ms...
  z2m:Zwave Node 6 is now asleep +16s
```

Control panel screenshot:
[image]

crxporter, 2020-11-13:
@AlCalzone thank you for looking into this - is there an easy way for me to install the new branch into the zwavejs2mqtt system? I'd like to continue my testing so I can get to my change to the JS world away from the open zwave world.

AlCalzone, 2020-11-13:
not an easy way, but what you can do:
1. clone this repo, check out the branch
2. `npx lerna bootstrap`, then `npx lerna exec -- npm link`
3. in the `zwavejs2mqtt` directory run `npm link zwave-js`.

robertsLando, 2020-11-13:
@AlCalzone Before `npx lerna bootstrap` needs also to run `npm install`

```
git clone https://github.com/zwave-js/node-zwave-js.git
cd node-zwave-js
# Build and bootstrap lerna
npm run prerelease 
npx lerna exec -- npm link

# Go to zwavejs2mqtt folder 
npm link zwave-js
```

AlCalzone, 2020-11-13:
`lerna bootstrap` does `npm install` internally. `npx` downloads a temporary copy of lerna and uses that. Or am I missing something here?

robertsLando, 2020-11-13:
> npx downloads a temporary copy of lerna and uses that. Or am I missing something here

Didn't know this, usually I always run npm install to run all deps

AlCalzone, 2020-11-13:
The problem is you need a global version of `lerna` to run the commands, have a local npm script or set up the PATH env manually to run a dependency. `npx` makes that easier, especially if you cannot be sure that someone has the necessary global deps.

robertsLando, 2020-11-13:
Yes I know that what I didn't know is that it automatically download a dep if it's not present, I thought it only checked the binaries on `node_modules/.bin` and run them. Nice to know  :)

AlCalzone, 2020-11-13:
https://medium.com/@maybekatz/introducing-npx-an-npm-package-runner-55f7d4bd282b mentions some more nice tricks (e.g. choosing node version).

robertsLando, 2020-11-13:
@AlCalzone When I see this posts dated 3 years ago I feel so bad, I'm always updated with the news but I always miss something, thanks for the link! :)

closed 2020-11-14 by AlCalzone

AlCalzone, 2020-11-14:
This is included in `v5.4.0`

crxporter, 2020-11-14:
Awesome. Thank you!

crxporter, 2020-11-16:
Confirming this is fixed with my zwavejs2mqtt pull today.

Thanks again!

PR #1116 (2020-11-14): fix(cc): ignore timeouts during AGI interview
fixes: #1115