zwave-js/zwave-js #1124: Support secure inclusion with Security S2

AlCalzone, 2020-11-15:


SuRGeoNix, 2021-03-09:
Hi guys, I've a Yale YDM 3168 door lock that accepts a z-wave module 2 (which I guess performs on secure s2 protocol?)
 I tried secure inclusion (from HA) but it fails so I guess that's the reason? I will drop you also the log if helps.

A partially workaround was to connect UZB1 to z-way secure inclusion with my lock (that worked) and did a proper interview. Then exported the network key and went back to my raspberry with HA and zwavejs2mqtt... it was kind of working, it was struggling as well... like it didn't have strong signal (I've used also an USB extension cable to avoid interference with USB3.0 ssd). Interviews were failing too (it managed to do a proper one but after very long time). Any direction will be appreciated.. Thanks!
[zwavejs_354.log](https://github.com/zwave-js/node-zwave-js/files/6109149/zwavejs_354.log)

AlCalzone, 2021-03-09:
> 13:50:36.311 CNTRLR   handling add node request (status = Failed)
13:50:36.312 CNTRLR     starting the inclusion failed

There's something weird with your stick. Doesn't look like it has to do with S2 support. For device-specific problems, please open a separate issue if you are sure that zwave-js does something wrong.

AlCalzone, 2021-08-13:
fixed by #1136

closed 2021-08-13 by AlCalzone