zwave-js/zwave-js #1126: Automate index file update in config PRs

AlCalzone, 2020-11-16:
Without it, the `lint:config` check won't pick up new files.

For example we could check if a device config file was changed but the index file was not. This action could comment on the PR with a command that updates the index file automatically in the PR.

robertsLando, 2020-11-17:
`npm run config -- -i` ?

AlCalzone, 2020-11-17:
yep, but I was thinking in the direction of:

> **Comment from Github Actions:**
> You forgot to update the config file index. Would you like me to do that? If so, comment "update index"

> **Comment from PR author:**
> update index

> **Github Actions** creates commit with updated index file
> (triggering a new build that lints the config file)

robertsLando, 2020-11-17:
Hehehehe that's a little bit tricky, what about simply push it when we detect that he forgot to do it?

AlCalzone, 2020-11-17:
Hm that could probably result in merge conflicts if the author keeps working on the branch. I'd rather make it explicit.

robertsLando, 2020-11-17:
@AlCalzone I have wrote a draft, lets discuss about this

PR #1127 (2020-11-17): feat(config): automate index update in PRs
Fixes #1126
PR #1324 (2021-01-09): refactor: remove device index file and generate it on demand
This PR is based on #1317 so it contains some more commits

closes: #1126
closed 2021-01-09 by AlCalzone