zwave-js/zwave-js #1128: Z-Wave driver not able to complete the interview with a Heatit Z-TRM2 device

larstobi, 2020-11-17:
**Describe the bug**
Node 009 is not able to complete the Interview. I can see that Node 009 is having a hard time being Interviewed. It's a Heatit Z-TRM2 device. 

**To Reproduce**
Steps to reproduce the behavior:
1. Start zwavejs2mqtt with a Heatit Z-TRM2 thermostat in the network
2. Observe the not completed interview. Last completed stage is NodeInfo.

**Device information**
Which device(s) is/are affected?   HeatIt Z-TRM2
What are the node IDs? `9`

**Logfile:** 
I added the debug log in https://gist.github.com/larstobi/a644872f2cfb2b96f5a0fb2c1a0d9e5d
as `zwave-4.log`

larstobi, 2020-11-17:
```
15:41:39.151 DRIVER « [Node 009] [REQ] [ApplicationCommand]
15:41:39.382 DRIVER « [Node 009] [REQ] [ApplicationCommand]
15:41:39.383 CNTRLR « [Node 009] received information for parameter #1:
15:41:39.383 CNTRLR » [Node 009] querying parameter #1 value...
15:41:39.386 DRIVER » [Node 009] [REQ] [SendData]
15:41:39.592 CNTRLR   [Node 009] [~] [Configuration] 1: 11 => 11
15:41:39.594 DRIVER « [Node 009] [REQ] [ApplicationCommand]
15:41:39.595 CNTRLR » [Node 009] querying parameter #2 information...
15:41:39.599 DRIVER » [Node 009] [REQ] [SendData]
15:41:39.805 DRIVER « [Node 009] [REQ] [ApplicationCommand]
15:41:39.811 DRIVER » [Node 009] [REQ] [SendData]
15:41:41.535 CNTRLR   [Node 009] did not respond after 1/3 attempts. Scheduling next try in 500 ms.
15:41:42.037 DRIVER » [Node 009] [REQ] [SendData]
15:41:43.800 CNTRLR   [Node 009] did not respond after 2/3 attempts. Scheduling next try in 500 ms.
15:41:44.302 DRIVER » [Node 009] [REQ] [SendData]
15:41:44.411 DRIVER » [Node 009] [REQ] [SendData]
15:41:49.388 DRIVER » [Node 009] [REQ] [SendData]
15:41:49.491 DRIVER » [Node 009] [REQ] [SendData]
15:41:52.152 CNTRLR   [Node 009] did not respond after 1/3 attempts. Scheduling next try in 500 ms.
15:41:52.657 DRIVER » [Node 009] [REQ] [SendData]
15:41:54.368 CNTRLR   [Node 009] did not respond after 2/3 attempts. Scheduling next try in 500 ms.
15:41:54.871 DRIVER » [Node 009] [REQ] [SendData]
15:42:09.472 DRIVER « [Node 009] [REQ] [ApplicationCommand]
15:42:09.473 CNTRLR « [Node 009] received information for parameter #2:
15:42:09.473 CNTRLR » [Node 009] querying parameter #2 value...
15:42:09.476 DRIVER » [Node 009] [REQ] [SendData]
15:42:09.682 CNTRLR   [Node 009] [~] [Configuration] 2: 0 => 0
15:42:09.683 DRIVER « [Node 009] [REQ] [ApplicationCommand]
15:42:09.684 CNTRLR » [Node 009] querying parameter #3 information...
15:42:09.689 DRIVER » [Node 009] [REQ] [SendData]
15:42:09.898 DRIVER « [Node 009] [REQ] [ApplicationCommand]
15:42:09.904 DRIVER » [Node 009] [REQ] [SendData]
15:42:11.633 CNTRLR   [Node 009] did not respond after 1/3 attempts. Scheduling next try in 500 ms.
15:42:12.134 DRIVER » [Node 009] [REQ] [SendData]
15:42:30.293 CNTRLR   [Node 009] did not respond after 2/3 attempts. Scheduling next try in 500 ms.
15:42:30.796 DRIVER » [Node 009] [REQ] [SendData]
15:42:32.461 DRIVER » [Node 009] [REQ] [SendData]
15:42:34.211 CNTRLR   [Node 009] did not respond after 1/3 attempts. Scheduling next try in 500 ms.
15:42:34.714 DRIVER » [Node 009] [REQ] [SendData]
15:42:36.500 CNTRLR   [Node 009] did not respond after 2/3 attempts. Scheduling next try in 500 ms.
15:42:37.004 DRIVER » [Node 009] [REQ] [SendData]
15:42:39.164 CNTRLR   [Node 009] Failed all interview attempts, giving up.
```

AlCalzone, 2020-11-17:
What is up with all these devices sending invalid `ConfigurationCCInfoReports`?
This is a duplicate of https://github.com/zwave-js/node-zwave-js/issues/1106

Please update to version 5.4.0.

larstobi, 2020-11-23:
I did use at least 5.4.0, or rather the current master. I have built again from the most recent master at this moment.

I added the logs as `zwave-5.log` in  https://gist.github.com/larstobi/a644872f2cfb2b96f5a0fb2c1a0d9e5d

AlCalzone, 2020-11-23:
Ok let me check - I might have looked at the wrong log. If you attach them directly to your comments instead of a gist, it would probably be easier :)

larstobi, 2020-11-23:
I thought the logs would be too big to past directly into here. There are more than 2000 lines of Node 009 in the log. However, I can try to pick out a selection.

```
14:26:01.568 CNTRLR   [Node 009] trying to load device config
14:26:01.577 CNTRLR   [Node 009] device config loaded
14:26:01.871 CNTRLR   [Node 009] Beginning interview - last completed stage: NodeInfo
14:26:01.871 CNTRLR   [Node 009] supports Security, but no network key was configured. Continuing in
14:26:01.878 CNTRLR   [Node 009] ManufacturerSpecificCC: doing a partial interview...
14:26:01.879 CNTRLR   [Node 009] [~] [Manufacturer Specific] interviewComple [Endpoint 0] [internal]
14:26:01.998 CNTRLR   [Node 009] VersionCC: doing a partial interview...
14:26:01.999 CNTRLR   [Node 009] [~] [Version] interviewComplete: true => tr [Endpoint 0] [internal]
14:26:01.999 CNTRLR   [Node 009] trying to load device config
14:26:02.027 CNTRLR   [Node 009] device config loaded
14:26:02.029 CNTRLR   [Node 009] ZWavePlusCC: doing a partial interview...
14:26:02.029 CNTRLR   [Node 009] [~] [Z-Wave Plus Info] interviewComplete: t [Endpoint 0] [internal]
14:26:02.029 CNTRLR » [Node 009] querying device endpoint information...
14:26:02.525 DRIVER » [Node 009] [REQ] [SendData]
14:26:02.810 CNTRLR   [Node 009] [+] [Multi Channel] commandClasses: undefin [Endpoint 0] [internal]
14:26:02.810 CNTRLR   [Node 009] [~] [Multi Channel] countIsDynamic: false = [Endpoint 0] [internal]
14:26:02.811 CNTRLR   [Node 009] [~] [Multi Channel] identicalCapabilities:  [Endpoint 0] [internal]
14:26:02.811 CNTRLR   [Node 009] [~] [Multi Channel] individualCount: 4 => 4 [Endpoint 0] [internal]
14:26:02.811 CNTRLR   [Node 009] [~] [Multi Channel] aggregatedCount: 0 => 0 [Endpoint 0] [internal]
14:26:02.812 DRIVER « [Node 009] [REQ] [ApplicationCommand]
14:26:02.818 CNTRLR   [Node 009] The node is alive.
14:26:02.819 CNTRLR « [Node 009] received response for device endpoints:
14:26:02.819 CNTRLR » [Node 009] querying all endpoints...
14:26:07.061 DRIVER » [Node 009] [REQ] [SendData]
14:26:07.357 DRIVER « [Node 009] [REQ] [ApplicationCommand]
14:26:07.362 CNTRLR « [Node 009] received endpoints: 1, 2, 3, 4
14:26:07.363 CNTRLR » [Node 009] querying capabilities for endpoint #1...
14:26:08.385 DRIVER » [Node 009] [REQ] [SendData]
14:26:08.682 CNTRLR   [Node 009] [~] [Multi Channel] commandClasses: 94,133, [Endpoint 1] [internal]
14:26:08.683 DRIVER « [Node 009] [REQ] [ApplicationCommand]
14:26:08.687 CNTRLR « [Node 009] received response for endpoint capabilities (#1):
14:26:08.688 CNTRLR » [Node 009] querying capabilities for endpoint #2...
14:26:09.638 DRIVER » [Node 009] [REQ] [SendData]
14:26:09.930 CNTRLR   [Node 009] [~] [Multi Channel] commandClasses: 94,133, [Endpoint 2] [internal]
14:26:09.931 DRIVER « [Node 009] [REQ] [ApplicationCommand]
14:26:09.936 CNTRLR « [Node 009] received response for endpoint capabilities (#2):
14:26:09.936 CNTRLR » [Node 009] querying capabilities for endpoint #3...
14:26:10.927 DRIVER » [Node 009] [REQ] [SendData]
14:26:11.220 CNTRLR   [Node 009] [~] [Multi Channel] commandClasses: 94,133, [Endpoint 3] [internal]
14:26:11.220 DRIVER « [Node 009] [REQ] [ApplicationCommand]
14:26:11.225 CNTRLR « [Node 009] received response for endpoint capabilities (#3):
14:26:11.225 CNTRLR » [Node 009] querying capabilities for endpoint #4...
14:26:12.182 DRIVER » [Node 009] [REQ] [SendData]
14:26:12.480 CNTRLR   [Node 009] [~] [Multi Channel] commandClasses: 94,133, [Endpoint 4] [internal]
14:26:12.481 DRIVER « [Node 009] [REQ] [ApplicationCommand]
14:26:12.485 CNTRLR « [Node 009] received response for endpoint capabilities (#4):
14:26:12.485 CNTRLR   [Node 009] [~] [Multi Channel] interviewComplete: true [Endpoint 0] [internal]
14:26:12.512 CNTRLR   [Node 009] ZWavePlusCC: doing a partial interview...              [Endpoint 1]
14:26:12.513 CNTRLR   [Node 009] [~] [Z-Wave Plus Info] interviewComplete: t [Endpoint 1] [internal]
14:26:12.513 CNTRLR   [Node 009] BinarySwitchCC: doing a partial interview...           [Endpoint 1]
14:26:12.513 CNTRLR » [Node 009] querying Binary Switch state...                        [Endpoint 1]
14:26:13.496 DRIVER » [Node 009] [REQ] [SendData]
14:26:13.793 CNTRLR   [Node 009] [~] [Binary Switch] currentValue: false => false       [Endpoint 1]
14:26:13.794 CNTRLR   [Node 009] [+] [Binary Switch] targetValue: undefined             [Endpoint 1]
14:26:13.795 DRIVER « [Node 009] [REQ] [ApplicationCommand]
14:26:13.799 CNTRLR « [Node 009] received Binary Switch state:                          [Endpoint 1]
14:26:13.799 CNTRLR   [Node 009] [~] [Binary Switch] interviewComplete: true [Endpoint 1] [internal]
14:26:13.829 CNTRLR   [Node 009] MeterCC: doing a partial interview...                  [Endpoint 1]
14:26:13.829 CNTRLR » [Node 009] querying meter value (type = Electric, scale = kWh)... [Endpoint 1]
14:26:14.764 DRIVER » [Node 009] [REQ] [SendData]
14:26:15.083 CNTRLR   [Node 009] [~] [Meter] value[65537]: 3707.3 => 3771.4             [Endpoint 1]
14:26:15.083 CNTRLR   [Node 009] [~] [Meter] deltaTime[65537]: 0 => 0                   [Endpoint 1]
14:26:15.085 DRIVER « [Node 009] [REQ] [ApplicationCommand]
14:26:15.088 CNTRLR » [Node 009] querying meter value (type = Electric, scale = W)...   [Endpoint 1]
14:26:21.912 DRIVER » [Node 009] [REQ] [SendData]
14:26:22.232 CNTRLR   [Node 009] [~] [Meter] value[66049]: 0.03 => 0.08                 [Endpoint 1]
[... cut ...]
14:30:12.111 CNTRLR   [Node 009] [~] [Binary Switch] targetValue: undefined => undefine [Endpoint 1]
14:30:12.112 DRIVER « [Node 009] [REQ] [ApplicationCommand]
14:30:12.115 CNTRLR « [Node 009] received Binary Switch state:                          [Endpoint 1]
14:30:12.115 CNTRLR   [Node 009] [~] [Binary Switch] interviewComplete: true [Endpoint 1] [internal]
14:30:12.116 CNTRLR   [Node 009] MeterCC: doing a partial interview...                  [Endpoint 1]
14:30:12.116 CNTRLR » [Node 009] querying meter value (type = Electric, scale = kWh)... [Endpoint 1]
```

robertsLando, 2020-11-23:
> I thought the logs would be too big to past directly into here. There are more than 2000 lines of Node 009 in the log. However, I can try to pick out a selection.

AlCalzone, 2020-11-23:
> I thought the logs would be too big to past directly into here

You can attach files too:
[image]

AlCalzone, 2020-11-23:
@larstobi Ok the problem are the borked InfoReports the device sends. This is 100% an issue in the firmware. As soon as the driver asks for the parameter description, the device goes tits up and sends 256 (i guess) invalid info reports without any content. This causes the following queries to fail too.

larstobi, 2020-11-23:
I see. Probably the certification process is a bit lacking... 

The device kinda works in OpenZwave, the only thing that doesn't work is setting the temperature via Home Assistant. It could be related to an issue which is confirmed present in Z-TRM3, but I couldn't say for certain. https://github.com/OpenZWave/open-zwave/pull/2458 

I also have bought a Z-TRM3 and will replace the Z-TRM2 with that one soon.

Is this the sort of thing that will require device hacks to get stuff working?

AlCalzone, 2020-11-23:
> Is this the sort of thing that will require device special handling to get stuff working?

Yup. I would have to add a compatibility flag that can be used in config files telling the driver not to use the V3 command to query configuration info.

> It could be related to an issue which is confirmed present in Z-TRM3

Might be. Make sure you install the custom firmware for that.

Shame that there is no good and good-looking thermostat for electrical floor heaters... My MCOHome thermostats frequently lock up and continue heating 😖

robertsLando, 2020-11-23:
> Shame that there is no good and good-looking thermostat for electrical floor heaters... My MCOHome thermostats frequently lock up and continue heating well past the setpoint

larstobi, 2020-11-23:
As I understood it there's that fixed precision number for setpoint that's causing the issue.

AlCalzone, 2020-11-23:
Yup but that is fixed in the "unofficial" firmware which HeatIt provide on their "official" website.

robertsLando, 2020-11-23:
@larstobi Also please do it using zjs2m UI I mean send the frimware update, so you can tell me if it's working.

If the device explodes it means it's not working :laughing:

larstobi, 2020-11-23:
I will try using zjs2m right now. I actually bought a cheap Windows laptop just for this purpose...but I guess I don't need it after all, haha...

AlCalzone, 2020-11-23:
> If the device explodes it means it's not working 😆

No guarantees 😬

robertsLando, 2020-11-23:
Actually I don't allow to specify the update target so it will update the chip by default

larstobi, 2020-11-23:
Okay. I have included the Z-TRM3 now, and it's actually completed the Interview, which is an improvement on the Z-TRM2. It's unknown, though. Now, trying to find the new firmware and will try the zw2m way first.

larstobi, 2020-11-23:
I can't imagine how all those commercial Z-Wave gateways handle all these device quirks. They should test devices better on different platforms!

robertsLando, 2020-11-23:
> I can't imagine how all those commercial Z-Wave gateways handle all these device quirks. They should test devices better on different platforms!

larstobi, 2020-11-23:
I have tried to use the setpoint feature of the unknown Z-TRM3 before firmware update.

I changed the setpoint value to 21.0 in the UI and clicked send. The corresponding entries from the logs are:

```
17:16:28.596 DRIVER » [Node 055] [REQ] [SendData]
17:16:28.769 DRIVER » [Node 055] [REQ] [SendData]
17:16:29.084 CNTRLR   [Node 055] [~] [Thermostat Setpoint] setpoint[1]: 18.5 => 18.5    [Endpoint 1]
17:16:29.084 CNTRLR   [Node 055] [~] [Thermostat Setpoint] setpointScale[1]: [Endpoint 1] [internal]
17:16:29.085 DRIVER « [Node 055] [REQ] [ApplicationCommand]
```

What I find a bit weird is that it tries to update with the current setpoint rather than the new one that I typed into the box.

AlCalzone, 2020-11-23:
> [Node 055] [~] [Thermostat Setpoint] setpoint[1]: 18.5 => 18.5    [Endpoint 1]

That is the value update which happens after the report from the device. If you increase the loglevel you will see how the correct setpoint is sent and the old one is reported.

larstobi, 2020-11-23:
Good tip about LOG_NODES!

Also, firmware update didn't pan out.

```
17:39:51.746 DRIVER « [Node 055] [REQ] [ApplicationCommand]
17:39:51.747 CNTRLR » [Node 055] Sending firmware fragment 6638 / 6643
17:39:51.751 DRIVER » [Node 055] [REQ] [SendData]
17:39:51.810 DRIVER « [Node 055] [REQ] [ApplicationCommand]
17:39:51.811 CNTRLR » [Node 055] Sending firmware fragment 6639 / 6643
17:39:51.815 DRIVER » [Node 055] [REQ] [SendData]
17:39:51.875 DRIVER « [Node 055] [REQ] [ApplicationCommand]
17:39:51.876 CNTRLR » [Node 055] Sending firmware fragment 6640 / 6643
17:39:51.879 DRIVER » [Node 055] [REQ] [SendData]
17:39:51.939 DRIVER « [Node 055] [REQ] [ApplicationCommand]
17:39:51.940 CNTRLR » [Node 055] Sending firmware fragment 6641 / 6643
17:39:51.943 DRIVER » [Node 055] [REQ] [SendData]
17:39:52.003 DRIVER « [Node 055] [REQ] [ApplicationCommand]
17:39:52.004 CNTRLR » [Node 055] Sending firmware fragment 6642 / 6643
17:39:52.007 DRIVER » [Node 055] [REQ] [SendData]
17:39:52.066 DRIVER « [Node 055] [REQ] [ApplicationCommand]
17:39:52.067 CNTRLR » [Node 055] Sending firmware fragment 6643 / 6643
17:39:52.071 DRIVER » [Node 055] [REQ] [SendData]
17:39:58.640 DRIVER « [Node 055] [REQ] [ApplicationCommand]
17:39:58.641 CNTRLR « [Node 055] Firmware update failed with status Error_InsufficientMemory
17:40:17.870 CNTRLR   [Node 055] [~] [Thermostat Mode] mode: 1 => 1                     [Endpoint 1]
17:40:17.871 DRIVER « [Node 055] [REQ] [ApplicationCommand]
17:40:17.898 CNTRLR   [Node 055] [~] [Thermostat Setpoint] setpoint[1]: 18.5 => 18.5    [Endpoint 1]
17:40:17.898 CNTRLR   [Node 055] [~] [Thermostat Setpoint] setpointScale[1]: [Endpoint 1] [internal]
17:40:17.899 DRIVER « [Node 055] [REQ] [ApplicationCommand]
17:40:17.916 CNTRLR   [Node 055] [~] [Thermostat Operating State] state: 0 => 0         [Endpoint 1]
```

AlCalzone, 2020-11-23:
> Error_InsufficientMemory

That is a report by the node, I'm assuming that is not the correct firmware or it has the wrong format (some need to be converted first). @robertsLando how do you handle the firmware files?

robertsLando, 2020-11-23:
I read the file as a binary buffer and I send it to the backend

larstobi, 2020-11-23:
The file I used is this one: https://drive.google.com/file/d/16RaxbnKDC9Mlqs2C6bUDNo9JDnm9cd75/view 

It has a `.otz` filename ending. Not sure if it needs converting...

larstobi, 2020-11-23:
Here is the content:
```
:020000040000FA
:1000000080000170D8348EB24B41FFE0FF22900394
:10001000C6E0FF121DD69003CB7B8A1F02184E1A32
:10002000341A39007FB8EF90181393901900730AAF
:1000300020364CE56EFF22EFF56E221C1FAA1CAF86
:10004000026EF91E1C1E21FF0258B9EE30E707C3ED
:10005000E49FFFE49EFE2219BC19C119C61F19CBEB
:1000600019D519E919EE19F3FF0260437581B075CE
[...]
:10706000796F90386E4009033B79CC22C28362BFAE
:107070000429F9EA3C22419C43F9032290339B62A4
:10708000172008009A20BB0B339822FF9020D9E0EC
:10709000FDEF5D2221F6A01A20D100FF3066404EA0
:1070A000042579E98F822050009D801101FF7433FF
:1070B0009800A82892013E85402B0722900664E0A4
:1070C000C3940A236000AF600F00E420442412033D
:1070D000229000254D7E450A023569E420160140C4
:1070E0003E401620070C29F0A3F022122488C30288
:0370F000238503F2
:00000001FF
```

AlCalzone, 2020-11-23:
Yeah that is an Intel HEX file. It first needs to be converted to binary data:
https://github.com/AlCalzone/ioBroker.zwave2/blob/ad27caa8ed4aadfd8cb973db37ee7d2346bf3bf8/src/main.ts#L1163
@robertsLando

robertsLando, 2020-11-24:
Ok let me work on it :)

robertsLando, 2020-11-24:
@larstobi Try to pull master, last commit should fix the firmware update

robertsLando, 2020-11-24:
P.S: @AlCalzone I think that the function `guessFirmwareFormat` should be included in zwavejs like `extractFirmware`

larstobi, 2020-11-24:
@robertsLando Okay, just tried, but got error:

 `Error while calling api beginFirmwareUpdate: Unable to extract firmware from file: Cannot read property 'otz' of undefined `

robertsLando, 2020-11-24:
I think that `FirmwareFileFormat[filename.slice(-3)]` is wrong, @AlCalzone how should I handle this?

larstobi, 2020-11-24:
The full filename of my file is: `Therm3_slave_enhanced_232_OTA_ZW050x_EU_SETPOINT_MOD_UNOFFICIAL.otz`

robertsLando, 2020-11-24:
Try now @larstobi , I have made a really stupid error when translating that function, that's not an enum

robertsLando, 2020-11-25:
@larstobi Is it working now?

larstobi, 2020-11-25:
I've just started zwavejs2mqtt with the latest master branch. Will report back in a jiffy!

larstobi, 2020-11-25:
New error:
`08:09:53.632 CNTRLR   [Node 055] Error while sending firmware fragment: data.slice is not a function`

AlCalzone, 2020-11-25:
Without context that looks like the data passed to the update method is not a buffer.

robertsLando, 2020-11-25:
It's strange because: https://github.com/zwave-js/zwavejs2mqtt/blob/master/lib/ZwaveClient.js#L1685

:confused:

AlCalzone, 2020-11-25:
Yup, that is wrong:
```
	public async beginFirmwareUpdate(
		data: Buffer,
		target: number = 0,
	): Promise<void> {
```

```
export function extractFirmware(
	data: Buffer,
	format: FirmwareFileFormat,
): Firmware {
// ^-- NOT Buffer!
```

robertsLando, 2020-11-25:
Totally missed that :confounded: @larstobi Try with the latest commit in master plaese, hope it would be the last one about this :crossed_fingers:

larstobi, 2020-11-25:
I'm having some slowness getting the Z-TRM3 to complete intervew before proceeding with FW upgrade. I found an error in the zwavejs log:
`14:07:29.603 CNTRLR   [Node 055] Error during node interview: Node 55 does not support the Command C`

robertsLando, 2020-11-25:
Is the log trimmed or it ends just with `C` ?

larstobi, 2020-11-25:
I think it ends with `C`

larstobi, 2020-11-25:
It eventually got Complete, and I have now started FW update. Currently "Sending firmware fragment 641 / 2362"

larstobi, 2020-11-25:
```
14:26:45.774 CNTRLR » [Node 055] Sending firmware fragment 2361 / 2362
14:26:45.778 DRIVER » [Node 055] [REQ] [SendData]
14:26:45.838 DRIVER « [Node 055] [REQ] [ApplicationCommand]
14:26:45.839 CNTRLR » [Node 055] Sending firmware fragment 2362 / 2362
14:26:45.842 DRIVER » [Node 055] [REQ] [SendData]
14:26:53.773 DRIVER « [Node 055] [REQ] [ApplicationCommand]
14:26:53.775 CNTRLR « [Node 055] Firmware update completed with status OK_RestartPending
14:26:53.775 CNTRLR   [Node 055] Firmware updated, scheduling interview in 30 ms...
14:26:53.843 CNTRLR   [Node 055] Beginning interview - last completed stage: None
14:26:53.843 CNTRLR   [Node 055] new node, doing a full interview...
14:26:53.843 CNTRLR » [Node 055] querying protocol info...
14:26:53.853 CNTRLR « [Node 055] received response for protocol info:
14:26:53.853 CNTRLR   [Node 055] Interview stage completed: ProtocolInfo
14:26:53.854 CNTRLR » [Node 055] pinging the node...
14:26:53.857 DRIVER » [Node 055] [REQ] [SendData]
14:27:01.568 CNTRLR   [Node 055] Node 55 did not respond after 1 attempts, it is presumed dead
14:27:01.572 CNTRLR   [Node 055] The node is now dead.
14:27:01.574 CNTRLR   [Node 055] ping failed: Failed to send the command after 1 attempts (Status No
14:27:01.574 CNTRLR » [Node 055] querying node info...
14:27:01.575 CNTRLR » [Node 055] pinging the node...
14:27:01.583 DRIVER » [Node 055] [REQ] [SendData]
14:27:08.933 CNTRLR   [Node 055] Node 55 did not respond after 1 attempts, it is presumed dead
14:27:08.934 CNTRLR   [Node 055] ping failed: Failed to send the command after 1 attempts (Status No
14:27:08.935 CNTRLR   [Node 055] Interview attempt (1/5) failed, node is dead.
14:27:43.169 CNTRLR   [Node 055] [+] [Thermostat Mode] mode: 1                          [Endpoint 1]
14:27:43.171 DRIVER « [Node 055] [REQ] [ApplicationCommand]
14:27:43.175 CNTRLR   [Node 055] The node is now alive.
14:27:43.176 CNTRLR   [Node 055] Beginning interview - last completed stage: ProtocolInfo
14:27:43.176 CNTRLR » [Node 055] pinging the node...
14:27:43.182 DRIVER » [Node 055] [REQ] [SendData]
14:27:43.195 CNTRLR   [Node 055] [Thermostat Setpoint] setpoint[1]: metadata updated    [Endpoint 1]
14:27:43.196 CNTRLR   [Node 055] [+] [Thermostat Setpoint] setpoint[1]: 18.5            [Endpoint 1]
14:27:43.197 CNTRLR   [Node 055] [+] [Thermostat Setpoint] setpointScale[1]: [Endpoint 1] [internal]
14:27:43.198 DRIVER « [Node 055] [REQ] [ApplicationCommand]
14:27:43.220 CNTRLR « [Node 055] ping successful
14:27:43.220 CNTRLR » [Node 055] querying node info...
14:27:43.225 DRIVER » [Node 055] [REQ] [RequestNodeInfo]
```

larstobi, 2020-11-25:
And setpoint actually works, with a caveat. If I try to set "23.5" it gets converted to "23", and then the thermostat shows "23.0". The thermostat supports setting half degrees C from the arrow buttons.

AlCalzone, 2020-11-25:
> If I try to set "23.5" it gets converted to "23", and then the thermostat shows "23.0"

Could you show me that part in a log? Just to make sure that it is not a bug on my end.

larstobi, 2020-11-25:
```

15:24:36.682 DRIVER » [Node 055] [REQ] [SendData]
15:24:36.717 DRIVER » [Node 055] [REQ] [SendData]
15:24:36.761 CNTRLR   [Node 055] [~] [Thermostat Setpoint] setpoint[1]: 23 => 23        [Endpoint 1]
15:24:36.762 CNTRLR   [Node 055] [~] [Thermostat Setpoint] setpointScale[1]: [Endpoint 1] [internal]
15:24:36.764 DRIVER « [Node 055] [REQ] [ApplicationCommand]
```

larstobi, 2020-11-25:
And the same zwavejs2mqtt log:
```

  z2m:App Zwave api call: writeValue [
  {
    nodeId: 55,
    commandClass: 67,
    endpoint: 1,
    property: 'setpoint',
    propertyKey: 1
  },
  23
] +50s
  z2m:Zwave Node 55: value updated: 67-1-setpoint-1 23 => 23 +26s
  z2m:Zwave Success zwave api call writeValue  +3ms
???

larstobi, 2020-11-25:
When I type 23.5 into the user parameter field called "undefined (55-67-1-setpoint-1)" and press the send arrow, it gets converted to an integer at once in the form field.

robertsLando, 2020-11-25:
> If I try to set "23.5" it gets converted to "23", and then the thermostat shows "23.0"

Are you using mqtt or the control panel?

robertsLando, 2020-11-25:
@larstobi Fixed in master

AlCalzone, 2020-11-25:
oh, so it was a bug in z2m?

AlCalzone, 2020-11-25:
> 15:24:36.682 DRIVER » [Node 055] [REQ] [SendData]
> 15:24:36.717 DRIVER » [Node 055] [REQ] [SendData]

For the reference: I usually need the log without filtered lines :) These two just tell me that some command was sent.

robertsLando, 2020-11-25:
> oh, so it was a bug in z2m?

Yep no worries it's already fixed

AlCalzone, 2020-11-25:
You might want to use `parseFloat` instead:
[image]

robertsLando, 2020-11-25:
> You might want to use parseFloat instead:

There is no advantages to use parseFloat in this situation as the input type for `numbers` is `number`, I think I have added that check because for some reason vue is sending me it as string, I should add to the model `.number` to parse it but I'm using the same input for text values: https://github.com/zwave-js/zwavejs2mqtt/blob/master/src/components/ValueId.vue#L13-L30

Cyberwizzard, 2020-11-28:
As author of https://github.com/OpenZWave/open-zwave/pull/2458 I'd like to point out that the whole goal of that pull request is to fix the Z-trm3 setpoint command *without* flashing firmware first.

If that PR would be included any Z-TRM3 would just work out of the box.

I'm not sure if that fix would also be applicable to the Z-TRM2 but since its relying on a compat flag in the device xml it should be easy to test for someone with that device.

[9 more replies]

PR #1341 (2021-01-11): feat(config): add compat option skipConfigurationInfoQuery
This PR adds a new compat option to skip the `ConfigurationCC` parameter info query which causes devices like the `Z-TRM2fx` to flood the network with garbage reports.

In addition, it adds a workaround for a bug **also** present in this device from hell, which causes the V3 query for the first parameter number to **also** send an invalid response.

fixes: #1309
fixes: #1128
closed 2021-01-11 by AlCalzone