zwave-js/zwave-js #1137: Better identify valueIds types

robertsLando, 2020-11-20:
In OZW I mostly used [this](https://github.com/OpenZWave/open-zwave/blob/master/config/Localization.xml) file to recognize valueIds types paired with [this](https://github.com/OpenZWave/open-zwave/blob/master/config/SensorMultiLevelCCTypes.xml) for sensors types and so on for meters alarms etc... 

I needed this informations because In zwave2mqtt I also support hass discovery, when i discover a new object in hass I need to set it's device_class and other parameters to discover it in the best way. 

So actually what I need is:

How do I retrieve the alarm sensor type from an alarm valueId ?
https://github.com/zwave-js/node-zwave-js/blob/master/packages/zwave-js/src/lib/commandclass/AlarmSensorCC.ts#L40

The same for sensorType number in sensor multilevel: https://github.com/zwave-js/node-zwave-js/blob/master/packages/zwave-js/src/lib/commandclass/MultilevelSensorCC.ts#L124

Same for metertype https://github.com/zwave-js/node-zwave-js/blob/master/packages/zwave-js/src/lib/commandclass/MeterCC.ts#L65

AlCalzone, 2020-11-20:
>  when i discover a new object in hass I need to set it's device_class and other parameters

Is there an overview for that?

robertsLando, 2020-11-20:
https://www.home-assistant.io/docs/mqtt/discovery/

Expecially for sensors: 

https://www.home-assistant.io/integrations/sensor.mqtt/

AlCalzone, 2020-11-21:
We discussed adding additional info to the ValueID, but on second thought I think this additional info should go on the metadata. You only need to figure out once per value ID what it represents.

This also results in less work that needs to be done for every value event.

robertsLando, 2020-11-21:
I'm ok for that! Let me know when you do it so I can go on on z2m with the hass discovery, also if I can help someway I submit a PR for this

AlCalzone, 2020-11-21:
If you could compile a list which value ids can benefit from this, it would help

robertsLando, 2020-11-21:
I wrote them in the first comment, do you need something more specific? 

I need the metertype and sensortype

AlCalzone, 2020-11-21:
Which properties? And there are probably more ccs?

robertsLando, 2020-11-21:
Actually what I need are the sensorType and the meterType, then I don't know if there are others, for the alarm I use the propertyKey to get the alarm type as you suggested.

AlCalzone, 2020-11-22:
> the alarm I use the propertyKey

That would be a good candidate too - like every ValueID where you need to infer the context somehow

robertsLando, 2020-11-22:
Maybe notification cc too?

robertsLando, 2020-11-23:
What I'm also struggling right now is to understand what will be the valueId of a device. Actually I have to check the code to identify the property/propertyKeys that a specific value will use

AlCalzone, 2020-11-23:
> the valueId of a device

What do you mean?

robertsLando, 2020-11-23:
I dunno if this could be a problem just for me or even for future developers that will create something that uses this driver but in ozw by using the localization.xml file I had a reference of all the value ids, the famous `index` property that now it's splitted in `property` and `propertyKey` here in zwave-js. In that Localization.xml file I hade a reference of all index of every cc so I know that index 0 of a sensro multilevel cc is always an air temperature and so on, in zwavejs to do the same I have to inspect the code to understand that the property is `Air temperature`

robertsLando, 2020-11-23:
Actually I'm converting my old devices database (a json database used for home assistant discovery of climate/fan devices) from oz to zwavejs:

Ozw: https://github.com/OpenZWave/Zwave2Mqtt/blob/master/hass/devices.js
Zwavejs: https://github.com/zwave-js/zwavejs2mqtt/blob/bdf5d4852bc8dcb53833b407ee8fdd5e903687d9/hass/devices.js

AlCalzone, 2020-11-23:
> In that Localization.xml file I had a reference of all index of every CC so I know that index 0 of a sensor multilevel is always an air temperature and so on, in zwavejs to do the same I have to inspect the code to understand that the property is `Air temperature`

So in the core this is an issue of documentation? We should really have an overview of the implemented CCs, their API methods and their values.

> index 0 of a sensor multilevel is always an air temperature

That would be solved by adding that info to the metadata, or wouldn't it? In other words: "Given valueID xyz, what does it represent?"

Looking at the js files you linked above, it seems like you really want the reverse lookup of this - in other words: "which value ID gives me the air temperature?"

robertsLando, 2020-11-23:
> That would be solved by adding that info to the metadata, or wouldn't it? In other words: "Given valueID xyz, what does it represent?"

Yes that will solve the problem, the second part of this issue it's kind of documentation issue, for developers not normal users

> Looking at the js files you linked above, it seems like you really want the reverse lookup of this - in other words: "which value ID gives me the air temperature?"

Exactly that! As we discussed some week ago I'm using this function: 

```js
function getValueID (v, withNode) {
  return `${withNode ? v.nodeId + '-' : ''}${v.commandClass}-${v.endpoint ||
    0}-${v.property}${v.propertyKey !== undefined ? '-' + v.propertyKey : ''}`
}
```

To get a unique string that identify a valueid in node values or, if it also have the node id, in the enitire zwave network

AlCalzone, 2020-11-23:
> > "which value ID gives me the air temperature?"
> 
> Exactly that!

Is it then enough if there is documentation to look that up? Or do you need a programmatical way to access this info? The latter could be done by inspecting the metadata of all known value IDs, but that shouldn't happen too often as it may become expensive.

robertsLando, 2020-11-23:
> is it then enough if there is documentation to look that up

Sure, I think that having a doc that shows all properties/propertikeys that you can see for each CC would be enought

robertsLando, 2020-11-23:
Maybe that doc could also be auto-generated? (would be much better and easy to maintain)

AlCalzone, 2020-11-23:
Absolutely!

robertsLando, 2020-11-23:
List of CC that needs a type:

AlarmSensor
BinarySensor
ClimateControlSchedule
DoorLock
Meter 
MultilevelSensor
MultilevelSwitch
ThermostatMode
ThermostatOperatingState
ThermostatSetback
ThermostatSetpoint
UserCodeCC
ZWavePlusCC

AlCalzone, 2020-11-23:
> ZWavePlusCC

Which property?

robertsLando, 2020-11-23:
Rule type and node type? both?

AlCalzone, 2020-11-23:
Uh are we still talking about the same thing? Those enums are exported. The **values** of these value IDs reference the enums

I thought this is about mapping valueIds where the cryptic values are part of the **Value ID** (like Meter type etc. for example)

robertsLando, 2020-11-23:
Yes but are them exposed in the propertyKey? Like for alarm?

AlCalzone, 2020-11-23:
No: https://github.com/zwave-js/node-zwave-js/blob/77f81a07d679d3e17342855ef39e4d09dc20ab27/packages/zwave-js/src/lib/commandclass/ZWavePlusCC.ts#L154-L206

robertsLando, 2020-11-23:
Oh ok sorry I missed that :) So remove it

robertsLando, 2020-11-23:
By double checking all command classes I think that MeterCC and SensorMultilevel CC are the only ones that would need this

PR #1141 (2020-11-24): Add CC-specific information to value metadata
fixes: #1137
by adding an optional `ccSpecific` field to the value metadata.

Checklist:

- [x] AlarmSensor (base type `AlarmSensorValueMetadata`)
- [x] BinarySensor (base type `BinarySensorValueMetadata`)
- [x] Indicator
- [x] Meter
- [x] MultilevelSensor (base type `MultilevelSensorValueMetadata`)
- [x] MultilevelSwitch (base type `MultilevelSwitchLevelChangeMetadata`)
- [x] Notification (notification type only for now)
- [x] ThermostatSetpoint (base type `ThermostatSetpointMetadata`) 

The other Thermostat classes don't use any specific type in  [... cut]
closed 2020-11-24 by AlCalzone