zwave-js/zwave-js #1144: [bug] TypeError: Cannot read property 'join' of undefined

ODINServ, 2020-11-24:
Hi,

I've ran into a crash which happens sometimes when the container is starting, or when I apply some configuration to the zwavejs project it self (not node configuration) which seem to trigger some restart.

I have now 14 nodes in the network
12 frg223
1 fgs222
1 fgs223

Still running the latest dev container.

Output podman logs

```
-23T20:01:38.616Z z2m:Zwave Node 14: value updated: 142-0-maxNodes-2 5 => 5
2020-11-23T20:01:38.616Z z2m:Zwave Node 14: value updated: 142-0-nodeIds-2 [] => []2020-11-23T20:01:38.616Z z2m:Zwave Node 14: value updated: 142-0-endpoints-2 [] => []
2020-11-23T20:01:41.176Z z2m:Zwave Node 14: value updated: 142-0-maxNodes-3 5 => 5
2020-11-23T20:01:41.177Z z2m:Zwave Node 14: value updated: 142-0-nodeIds-3 [] => []2020-11-23T20:01:41.177Z z2m:Zwave Node 14: value updated: 142-0-endpoints-3 [] => []
2020-11-23T20:01:41.473Z z2m:Zwave Node 17: value updated: 112-0-43 1 => 1
This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). The promise rejected with the reason:
TypeError: Cannot read property 'join' of undefined why
    at ZWaveNode.queryNeighborsInternal (/usr/src/app/node_modules/zwave-js/src/lib/node/Node.ts:1582:60)
    at ZWaveNode.queryNeighbors (/usr/src/app/node_modules/zwave-js/src/lib/node/Node.ts:1599:3)
    at tryInterviewStage (/usr/src/app/node_modules/zwave-js/src/lib/node/Node.ts:850:5)
    at ZWaveNode.interview (/usr/src/app/node_modules/zwave-js/src/lib/node/Node.ts:919:10)
    at Driver.interviewNode (/usr/src/app/node_modules/zwave-js/src/lib/driver/Driver.ts:812:10)
```

Managed to get some logging from the zwave-1 file.

```
20:01:41.429 DRIVER   send thread state: sending.waitForUpdate
20:01:41.431 DRIVER   send thread state: sending.waitForUpdate
20:01:41.436 DRIVER   send thread state: sending.waitForUpdate
20:01:41.475 DRIVER   send thread state: sending.execute
20:01:41.475 CNTRLR « [Node 017] parameter #43 has value: 1                        20:01:41.476 CNTRLR » [Node 017] querying parameter #44 value...
20:01:41.477 DRIVER   send thread state: sending.execute
20:01:41.488 DRIVER   send thread state: sending.execute
20:01:41.490 DRIVER   send thread state: sending.handshake.waitForCommandResult
20:01:41.491 DRIVER   send thread state: sending.handshake.waitForCommandResult    20:01:41.532 CNTRLR   [Node 014]   requesting the node neighbors failed: Cannot read property 'join'
                       of undefined
20:01:41.538 DRIVER   send thread state: sending.handshake.waitForCommandResult
20:01:41.539 DRIVER   send thread state: sending.handshake.waitForCommandResult    20:01:41.539 DRIVER   send thread state: sending.handshake.waitForHandshakeResponse20:01:41.540 DRIVER   send thread state: sending.handshake.waitForHandshakeResponse20:01:41.544 DRIVER   send thread state: sending.handshake.waitForHandshakeResponse20:01:41.698 DRIVER   send thread state: sending.handshake.waitForHandshakeResponse20:01:41.698 DRIVER   send thread state: sending.handshake.waitForHandshakeResponse20:01:41.700 DRIVER   Unexpected error during SerialAPI command: ZWaveError: Security CC requires a                                                                                         nonce to be sent!                                            20:01:41.700 DRIVER   send thread state: sending.execute                           20:01:41.701 CNTRLR   [Node 015] did not respond after 1/3 attempts. Scheduling next try in 500 ms.                                                                   20:01:41.701 DRIVER   send thread state: sending.retryWait
20:01:41.840 DRIVER   send thread state: sending.retryWait
20:01:41.841 DRIVER   send thread state: sending.retryWait
20:01:42.008 DRIVER   destroying driver instance...
```

I'm not asure why I'm not getting more detailed log info, I've set it to silly log level.

Thnx

AlCalzone, 2020-11-24:
I believe this captures the gist of the issue:

```log
08:49:37.926 DRIVER » [Node 017] [REQ] [SendData]
                      │ transmit options: 0x25
                      │ callback id:      254
                      └─[SecurityCCNonceGet]
08:49:37.932 SERIAL « [ACK]                                                                   (0x06)
08:49:37.933 SERIAL « 0x0104011301e8                                                       (6 bytes)
08:49:37.933 SERIAL » [ACK]                                                                   (0x06)
08:49:37.933 DRIVER « [RES] [SendData]
                        was sent: true
08:49:38.060 SERIAL « 0x0110000400110a9880f9db9676bf94c210d3                              (18 bytes)
08:49:38.060 SERIAL » [ACK]                                                                   (0x06)
08:49:38.061 DRIVER « [Node 017] [REQ] [ApplicationCommand]
                      └─[SecurityCCNonceReport]
                          nonce: 0xf9db9676bf94c210
08:49:38.099 SERIAL « 0x01070013fe00001104                                                 (9 bytes)
08:49:38.099 SERIAL » [ACK]                                                                   (0x06)
08:49:38.099 DRIVER « [REQ] [SendData]
                        callback id:     254
                        transmit status: OK
08:49:38.278 SERIAL « 0x0110000400110a9880acc91984b883c4d23d                              (18 bytes)
08:49:38.279 SERIAL » [ACK]                                                                   (0x06)
08:49:38.279 DRIVER « [Node 017] [REQ] [ApplicationCommand]
                      └─[SecurityCCNonceReport]
                          nonce: 0xacc91984b883c4d2
08:49:38.282 SERIAL » 0x011e00131117988179b4be8f8839c086bc17bf23ac111712bf9aaa0bdb25ffec  (32 bytes)
08:49:38.282 DRIVER » [Node 017] [REQ] [SendData]
                      │ transmit options: 0x25
                      │ callback id:      255
                      └─[SecurityCCCommandEncapsulation]
                        │ nonce id: 172
                        └─[ConfigurationCCGet]
                            parameter #: 43
08:49:38.285 SERIAL « [ACK]                                                                   (0x06)
08:49:38.289 SERIAL « 0x0104011301e8                                                       (6 bytes)
08:49:38.289 SERIAL » [ACK]                                                                   (0x06)
08:49:38.290 DRIVER « [RES] [SendData]
                        was sent: true
08:49:38.318 SERIAL « 0x01080004001102984038                                              (10 bytes)
08:49:38.319 SERIAL » [ACK]                                                                   (0x06)
08:49:38.319 DRIVER « [Node 017] [REQ] [ApplicationCommand]
                      └─[SecurityCCNonceGet]
08:49:38.354 SERIAL « 0x01070013ff00000713                                                 (9 bytes)
08:49:38.354 SERIAL » [ACK]                                                                   (0x06)
08:49:38.355 DRIVER « [REQ] [SendData]
                        callback id:     255
                        transmit status: OK
08:49:38.357 SERIAL » 0x01110013110a98809469f90b9b026e60050162                            (19 bytes)
08:49:38.358 DRIVER » [Node 017] [REQ] [SendData]
                      │ transmit options: 0x05
                      │ callback id:      1
                      └─[SecurityCCNonceReport]
                          nonce: 0x9469f90b9b026e60
08:49:38.360 SERIAL « [ACK]                                                                   (0x06)
08:49:38.363 SERIAL « 0x0104011301e8                                                       (6 bytes)
08:49:38.363 SERIAL » [ACK]                                                                   (0x06)
08:49:38.364 DRIVER « [RES] [SendData]
                        was sent: true
08:49:38.425 SERIAL « 0x011f00040011199881a24c2b056068e25fd8f261e4c638940cab028a4ff2d2424 (33 bytes)
                      7
08:49:38.425 SERIAL » [ACK]                                                                   (0x06)
08:49:38.432 DRIVER « [Node 017] [REQ] [ApplicationCommand]
                      └─[SecurityCCCommandEncapsulation]
                        │ sequenced: false
                        └─[ConfigurationCCReport]
                            parameter #: 43
                            value size:  1
                            value:       1
08:49:38.470 SERIAL « 0x010700130100000be1                                                 (9 bytes)
08:49:38.470 SERIAL » [ACK]                                                                   (0x06)
08:49:38.471 DRIVER « [REQ] [SendData]
                        callback id:     1
                        transmit status: OK
08:49:38.473 SERIAL » 0x010700800f00000077                                                 (9 bytes)
08:49:38.474 DRIVER » [Node 015] [REQ] [GetRoutingInfo]
                        remove non-repeaters: false
                        remove bad links:     false
08:49:38.525 CNTRLR   [Node 015]   requesting the node neighbors failed: Cannot read property 'join'
                       of undefined
```

The SendData callback for the SecurityCCNonceReport we sent earlier comes in **after** the transaction was already resolved.
This somehow causes it to be matched to the following unrelated request. Let's see if I can reproduce this in a test.

AlCalzone, 2020-11-24:
The test works! (actually it fails but that's what I wanted)

PR #1145 (2020-12-02): fix: do not mismatch SendData requests that are received out of sequence
fixes: #1144
closed 2020-12-02 by AlCalzone