zwave-js/zwave-js #1155: Add `genre` field to valueIDs

robertsLando, 2020-11-26:
Actually OZW assigns a Genre of all valueids:

ValueGenre_Basic | The 'level' as controlled by basic commands. Usually duplicated by another command class.
-- | --
ValueGenre_User | Basic values an ordinary user would be interested in.
ValueGenre_Config | Device-specific configuration parameters. These cannot be automatically discovered via Z-Wave, and are usually described in the user manual instead.
ValueGenre_System | Values of significance only to users who understand the Z-Wave protocol

http://www.openzwave.com/dev/classOpenZWave_1_1ValueID.html#af0360f2b3ecbb70dc9ab89ca74f7f305

AlCalzone, 2020-11-26:
I understand why you need this, but I decided not to do copy everything from OZW when I started (mainly because some things didn't make sense in my opinion). 

* Basic: `zwave-js` hides the Basic CC when it makes sense. If it is not hidden, it represents a value the user is interested in
* System: `zwave-js` has `internal` value IDs to hide them from users. IMO you shouldn't mess with those value IDs, since the driver handles them for you
* Config: Isn't that just `CommandClasses.Configuration`?
* User: Everything else

If that is not enough, maybe I need to understand the use case better. Are there system value IDs that don't need to be hidden? Which kind of value IDs that are currently "User" shouldn't be?

robertsLando, 2020-11-26:
> Are there system value IDs that don't need to be hidden? Which kind of value IDs that are currently "User" shouldn't be?

Essentially this is the question. I just would like to find the best way to expose valueids in user interface

I'm ok with all your considerations, so essentially we could consider everything that is not a configuraiton value a User/Application value?

AlCalzone, 2020-11-26:
> Essentially application would be like user and management like configuration?

Basically, yes. Although there will be some exceptions to the rule. 

Like `Door Lock CC`, which has the ability to configure timeouts and such. That doesn't fit the description of the **Configuration genre** above, but one could argue that it is configuration rather than basic values. On the other hand, as a user I would like values to be grouped if they belong together.

robertsLando, 2020-11-26:
> On the other hand, as a user I would like values to be grouped if they belong together.

In poor words we should simply identify CC that are Application (user) and CC that are 'Management' (system in ozw). The configuration CC is for cc 112 and that's already ok. 

So what about add a property to metadata like `category` and it could be `Application|Management|Network|Transport` based on: https://www.silabs.com/documents/login/miscellaneous/SDS13548-List-of-defined-Z-Wave-Command-Classes.xlsx

AlCalzone, 2020-11-26:
You can just `import { applicationCCs } from "@zwave-js/core` and check if a ValueID's `commandClass` is included in that array (Application) or not (Management).
Transport Encapsulation CCs are low-level and have no exposed value IDs
Network CCs are not used in zwave-js, since they are for Z/IP

robertsLando, 2020-11-26:
Nice! Will use this so :)

closed 2020-11-26 by robertsLando