zwave-js/zwave-js #1157: Support Indicator CC V4

AlCalzone, 2020-11-26:
Indicator CC V4 adds two new commands:
* Indicator Description Get Command
* Indicator Description Report Command

https://github.com/zwave-js/node-zwave-js/blob/master/specs/SDS13782%20Z-Wave%20Management%20Command%20Class%20Specification.pdf

**TODOs:**
- [ ] Implement the new commands
- [ ] Implement a `getDescription` method on `IndicatorCCAPI`
- [ ] update `@implementedVersion` decorator

Maverick1872, 2021-03-10:
Updated link to documentation can be found [here](https://github.com/zwave-js/specs/blob/0c0a0f9be2ff58330784c1888a4df07c4ac33804/SDS13782%20Z-Wave%20Management%20Command%20Class%20Specification.pdf).

Reedham20, 2021-11-17:
Hey, I am new to open source and found this issue can I get any status updates on it. Any help is Highly appreciated

PR #5395 (2023-02-01): feat(cc): support Indicator CC v4
fixes: #1157
closed 2023-02-01 by AlCalzone