zwave-js/zwave-js #1158: Support Wake Up CC V3 and Supervision CC V2

AlCalzone, 2020-11-26:
This combination adds support for requesting a node to wake up.
https://github.com/zwave-js/node-zwave-js/blob/master/specs/SDS13782%20Z-Wave%20Management%20Command%20Class%20Specification.pdf

Implementation is more complicated and might involve the driver and message flow

Ikcelaks, 2021-09-13:
#3332 will provide the support for this at the CC level, but we need to make some decisions on when the `Driver` chooses to request a Wake Up from supporting nodes.

I see three basic options:

1. Request a wake up any time there are messages in the transaction queue that would be ran if the node were awake.
2. Same as 1, but check for a tighter subset of more important messages.
3. Require the client to specify that it wants the node to wake up early (and still do the checks from option 1).

AlCalzone, 2021-09-13:
I think I'd go with an entry in `SendCommandOptions` (so basically 3.). If a command with that option should be sent to a supporting node, also queue a wake up request. If that one has priority `Handshake`, it **should** automatically be queued in front of all other commands to that node.

This allows applications to use the feature on demand and makes it possible to bake the wakeup into some commands in the driver.

Ikcelaks, 2021-09-13:
I'm not sure what you mean by "queue a wake up request".

When we want a node to wake up, we have to wait until they send us a `SupervisionCCGet` encapsulated command. When/if the node does that, we get to sneak the wake up request into the `SupervisionCCReport` that we send back to them.

Here's how I envision implementing option 3 (using `SendCommandOptions`):

- Add `requestWakeUp?: boolean` to `SendCommandOptions`
- Add a field (`nodesToRequestWakeUp: Set<number>`) to the `Driver` to keep a list of nodes that be asked to wake up if we get a `SupervisionCCGet` request for them
- When the `Driver` receives a call to `sendCommand` with `options.requestWakeUp` set to `true`, if verifies that the node supports Wake Up On Demand and adds the NodeId to the `nodesToRequestWakeUp` Set.
- When receiving a `SupervisionCCGet` command, check if the sending node is in `nodesToRequestWakeup`. If so, set `requestWakeUp = true` in the `SupervisionCCReport` that we return.
- When we set a node to `Awake`, remove it from `nodesToRequestWakeUp`.

I don't think anything special at all needs to be added to the message queue.

AlCalzone, 2021-09-14:
I misread the specs yesterday and thought we could send a sleeping node a request to wake up on our turns. 

Your proposal sounds good, I'd probably put the property `requestWakeUp` on the node instance instead of keeping track in the driver.

Ikcelaks, 2021-09-14:
I updated the #3332 to be a full implementation of this feature. Let me know if you like the approach, then I'll get to testing it on real hardware.

AlCalzone, 2021-09-16:
PR reviewed. I still prefer the idea of a `requestWakeUp` property on the node instance.

PR #3332 (2021-09-19): feat(cc): support Wake Up On Demand
This PR is the first step in resolving #1158.

This adds support for Wake Up On Demand.

`SupervisionCC` support is upgraded to `V2`.
`WakeUpCC` support is upgraded to `V3`.
The `Driver` and `Node` classes are updated to set the Wake Up On Demand bit when sending `SupervisionCCReport` replies to sleeping nodes that have important messages pending in the queue.

A Wake Up On Demand for a node is requested by setting `options.requestWakeUpOnDemand = true` in the `SendCommandOptions` when calling `driver.sendCommand`.

The code is currently untested, because I' [... cut]
closed 2021-09-19 by AlCalzone