zwave-js/zwave-js #1163: Support polling

AlCalzone, 2020-11-27:
> Silicon's Zware/ZIP libraries poll the main device properties every 30 seconds. Perhaps having an option for enabling polling for a specific device would be even better, since you'd only enable it for devices, that would need this, to avoid unneeded traffic.

I think this should configurable very granularly:
* Which nodes?
* For each node: Which CC?
* For each node: Which endpoints? Default: 0
* For each node: how often? Maybe even as a CRON, so we can spread out the requests

robertsLando, 2020-11-27:
AFAIK, about the "HOW OFTEN", you should set a pollInterval greather then the number of nodes in seconds. If we have 10 nodes then the pollInterval should be 10 or more seconds. Is there anything about this in specs?

petrovcicklemen, 2020-11-27:
There is this:
https://www.silabs.com/documents/login/miscellaneous/SDS11846-Z-Wave-Plus-Role-Type-Specification.pdf

From section 3.7 onwards.

robertsLando, 2020-11-27:
> The following requirements apply to the normal case where a polling request is successful.
 CommandTime MUST be measured by the application
 The application MUST wait PollTime before polling any other node
 PollTime SHOULD be 10 seconds + CommandTime or more
 PollTime MUST NOT be less than 1 second + CommandTime

petrovcicklemen, 2020-11-27:
There's also this:
https://www.silabs.com/community/wireless/z-wave/forum.topic.html/zware_hcapi_defaultdevicepolling-Ikwq

> When using ZWare service (or any application built on top of ZWare HCAPI) I've noticed that Z-Wave devices are polled every 30 seconds (active get status).

> Z-wares polling behavior is compliant with the Z-wave Plus polling requirements that sets a maximum rate for polling. This means that as more device are included in the network, the polling of each devices will be longer interval to prevent network congestion. For Z-wave plus v.2 devices that are required to send true status lifeline reporting Z-ware will not poll at all, as it can be certain that the devices will report any state change automatically.
> 
> The are pros and cons for polling, and Z-ware has chosen an approach, where it can keep the UI as updated as possible within the confines of the Z-wave spec. Reasons for less polling would typically be concerns about network congestion, but if fulfilling the Z-wave spec requirements, as Z-ware does, it should not be a problem.

OnFreund, 2021-01-12:
FWIW this is the only thing preventing me from switching over from OZW. I have a couple of [old devices](https://smarterhome.sk/en/zwave-wireless-thermostats/secure-sir-321-countdown-timer-285.html) that do not report state changes and need to be polled.

AlCalzone, 2021-01-26:
https://github.com/zwave-js/node-zwave-js/pull/1479 adds the basic support for polling. I'm hesitant to do polling automatically and if something happens, it has to be separated from the driver. On top of that, polling should be used very sparingly, so for now:

If you need regular polling, handle it yourself as part of an automation. The tools are there.

closed 2021-01-26 by AlCalzone