zwave-js/zwave-js #1165: Please add 2gig device configs

chilicheech, 2020-11-27:
For reference the configs from open-zwave for the 2gig devices is here: https://github.com/OpenZWave/open-zwave/tree/master/config/2gig

AlCalzone, 2020-11-28:
After a short glance at the config files over there, it does look like these devices could benefit from our partial config params.
```
        Bits 0 - 7 -> HVAC Setup: Normal (0x01) or Heat Pump (0x02)
        Bits 8 - 11 -> Number of Auxiliary Stages (Heat Pump) / Number of Heat Stages (Normal)
        Bits 12 - 15 -> Aux Setup: Gas (0x01) or Electric (0x02)
        Bits 16 - 23 -> Number of Heat Pump Stages
        Bits 24 - 31 -> Number of Cool Stages
```
Needs a good look at the manuals though.

robertsLando, 2020-11-30:
@chilicheech If we send you a partial config would you test and edit it?

chilicheech, 2020-12-01:
oh yeah. I'm willing to test and edit it.. let me know what to do :)

AlCalzone, 2020-12-01:
Hm I noticed something weird. The Z-Wave specs list 2GIG under the manufacturer ID 0x009b. OZW has them under 0x0098, which should be "Radio Thermostat Company of America (RTC)". Our config files also have them under 0x0098.

Maybe that's why you think there aren't any config files? Which manufacturer id, product id and product type do your devices have?

chilicheech, 2020-12-02:
Ah, makes sense.. Pardon my noobness. I should've looked for the manufacturer ID instead of the name. I believe 2gig and RTC had some thermostats in common and hence the confusion. I do see the config for my thermostat as it's showing up as 0x0098.

Cheers,

closed 2020-12-02 by chilicheech