zwave-js/zwave-js #1172: [bug] possible bug on doorbell 6

billiaz, 2020-11-30:
**Describe the bug**
I have a device of aeotec Siren 6/doorbell 6. when the button is pressed. Devices seems to be sending twice some messages

**Reproduce**
Press Doorbell 6 button, with alert enabled

**Device information**
Aeotec Doorbell 6 ZW164 - Node 52

[logzwave.tar.gz](https://github.com/zwave-js/node-zwave-js/files/5618341/logzwave.tar.gz)

AlCalzone, 2020-11-30:
Just took a short look. I cannot identify what you mean by sending messages twice. I see two interview processes and then nothing. Also, the loglevel seems to be less than "debug" - might be worth changing that to debug communication issues.

billiaz, 2020-11-30:
I changed it to silly under zwavejs2mqtt and restarted driver

In a specific moment i sent 2 changes

Change defaultVolume of 121/3 from 2 to 3 %

time of change i think is 20:49:36.834

earlier i changed volume from 2% to 2% (here I see no repeat on messages).

same happens when i get a siren state notification. I cannot log that now but if needed i can
new log attached
[zwave-1.log.gz](https://github.com/zwave-js/node-zwave-js/files/5618692/zwave-1.log.gz)

AlCalzone, 2020-11-30:
After the `SoundSwitchCCConfigurationSet` the driver requests an update with `SoundSwitchCCConfigurationGet` to which the device responds with `SoundSwitchCCConfigurationReport`. This is expected.
Also it seems like it sends two unsolicited `SoundSwitchCCConfigurationReport`s, one to the endpoint 0 and one to the endpoint 1.

The latter happens because you have two multi channel associations configured:
```
20:48:06.265 DRIVER « [Node 052] [REQ] [ApplicationCommand]
                      └─[MultiChannelAssociationCCReport]
                          group id:           1
                          maximum # of nodes: 5
                          node ids:
                          endpoints:          1:1, 1:0
```
Only `1:0` (node 1, root endpoint) should exist. The other is probably a misconfiguration on your part.

This brings you down to 2 reports instead of 3. We could think about adding a flag to disable the get-after-set for some node/cc combinations (where we know that the device will send an update on its own), which would bring it down to 1 report.

billiaz, 2020-12-01:
I have no clue how this device got two multichannel. I reset it a month ago. 
The strange part is, both are in lifeline, can it be the device "registered" twice the main controller and the reason of having double messages?

AlCalzone, 2020-12-01:
> can it be the device "registered" twice the main controller

Devices don't create associations themselves. During the interview, the driver checks the associations and adds the 1:0 association if it is missing. The 1:1 was either added manually or another driver did that. zwave-js doesn't delete those associations because it cannot know if it was added on purpose.

aretakisv, 2020-12-01:
I suspect the culprit. Is there a way to enforse zwave-js to accept one association and delete it? this way I do not have to go through the reset of the device.
Other wise I suspect Exclution, reset (just in case), inclusion is the only way

AlCalzone, 2020-12-01:
Like we discussed in slack, zwave-js has methods to manage (add/remove) associations. It should also be possible through the zwavejs2mqtt UI.

aretakisv, 2020-12-01:
deleting association: 
```

10:47:38.927 DRIVER   Send queue:                                                       (6 messages)
                      · [REQ] SendData (MultiChannelAssociationCCRemove) [Node 52, awake]
                      · [REQ] SendData (BatteryCCGet) [Node 35, asleep]
                      · [REQ] SendData (BatteryCCGet) [Node 42, asleep]
                      · [REQ] SendData (BatteryCCGet) [Node 59, asleep]
                      · [REQ] SendData (BatteryCCGet) [Node 60, asleep]
                      · [REQ] SendData (BatteryCCGet) [Node 78, asleep]
10:47:38.931 SERIAL » 0x010d001334068e040100010125e895                                    (15 bytes)
10:47:38.931 DRIVER » [Node 052] [REQ] [SendData]
                      │ transmit options: 0x25
                      │ callback id:      232
                      └─[MultiChannelAssociationCCRemove]
                          group id:  1
                          node ids:
                          endpoints: 1:1
10:47:38.931 DRIVER   send thread state: sending.execute
10:47:38.933 SERIAL « [ACK]                                                                   (0x06)
10:47:38.934 DRIVER   send thread state: sending.execute
10:47:38.940 SERIAL « 0x0104011301e8                                                       (6 bytes)
10:47:38.940 SERIAL » [ACK]                                                                   (0x06)
10:47:38.941 DRIVER « [RES] [SendData]
                        was sent: true
10:47:38.942 DRIVER   send thread state: sending.execute
10:47:38.956 SERIAL « 0x01070013e800000201                                                 (9 bytes)
10:47:38.957 SERIAL » [ACK]                                                                   (0x06)
10:47:38.957 DRIVER « [REQ] [SendData]
                        callback id:     232
                        transmit status: OK
10:47:38.961 DRIVER   send thread state: sending.execute
10:47:38.961 DRIVER   Send queue:                                                       (5 messages)
                      · [REQ] SendData (BatteryCCGet) [Node 35, asleep]
                      · [REQ] SendData (BatteryCCGet) [Node 42, asleep]
                      · [REQ] SendData (BatteryCCGet) [Node 59, asleep]
                      · [REQ] SendData (BatteryCCGet) [Node 60, asleep]
                      · [REQ] SendData (BatteryCCGet) [Node 78, asleep]
10:47:38.961 DRIVER   send thread state: idle
10:47:38.963 DRIVER   Send queue:                                                       (6 messages)
                      · [REQ] SendData (MultiChannelAssociationCCGet) [Node 52, awake]
                      · [REQ] SendData (BatteryCCGet) [Node 35, asleep]
                      · [REQ] SendData (BatteryCCGet) [Node 42, asleep]
                      · [REQ] SendData (BatteryCCGet) [Node 59, asleep]
                      · [REQ] SendData (BatteryCCGet) [Node 60, asleep]
                      · [REQ] SendData (BatteryCCGet) [Node 78, asleep]
10:47:38.966 SERIAL » 0x010a001334038e020125e990                                          (12 bytes)
10:47:38.967 DRIVER » [Node 052] [REQ] [SendData]
                      │ transmit options: 0x25
                      │ callback id:      233
                      └─[MultiChannelAssociationCCGet]
                          group id: 1
10:47:38.967 DRIVER   send thread state: sending.execute
10:47:38.968 SERIAL « [ACK]                                                                   (0x06)
10:47:38.969 DRIVER   send thread state: sending.execute
10:47:38.975 SERIAL « 0x0104011301e8                                                       (6 bytes)
10:47:38.975 SERIAL » [ACK]                                                                   (0x06)
10:47:38.976 DRIVER « [RES] [SendData]
                        was sent: true
10:47:38.978 DRIVER   send thread state: sending.execute
10:47:38.991 SERIAL « 0x01070013e900000200                                                 (9 bytes)
10:47:38.991 SERIAL » [ACK]                                                                   (0x06)
10:47:38.992 DRIVER « [REQ] [SendData]
                        callback id:     233
                        transmit status: OK
10:47:38.994 DRIVER   send thread state: sending.execute
10:47:38.995 DRIVER   send thread state: sending.waitForUpdate
10:47:39.007 SERIAL « 0x0110000400340a8e0301050000010101005d                              (18 bytes)
10:47:39.008 SERIAL » [ACK]                                                                   (0x06)
10:47:39.010 CNTRLR   [Node 052] [~] [Multi Channel Association] maxNodes[1]: 5 => 5
[... 59 lines cut ...]
                      · [REQ] SendData (BatteryCCGet) [Node 60, asleep]
                      · [REQ] SendData (BatteryCCGet) [Node 78, asleep]
10:47:42.552 DRIVER   send thread state: idle
```
and then the log from zj2m
```
zwavejs2mqtt-6bc86c4689-hlgv8 z2m 2020-12-01T10:47:28.549Z z2m:Zwave Node 64: metadata updated: 49-0-Power
zwavejs2mqtt-6bc86c4689-hlgv8 z2m 2020-12-01T10:47:28.550Z z2m:Zwave Node 64: value updated: 49-0-Power 79.1 => 84.6
zwavejs2mqtt-6bc86c4689-hlgv8 z2m 2020-12-01T10:47:38.924Z z2m:App Zwave api call: removeAssociations [ 52, 1, [ { nodeId: 1, endpoint: 1 } ] ]
zwavejs2mqtt-6bc86c4689-hlgv8 z2m 2020-12-01T10:47:38.924Z z2m:Zwave Assocaitions: Removing associations from Node 52 Group 1: [ { nodeId: 1, endpoint: 1 } ]
[... cut ...]
zwavejs2mqtt-6bc86c4689-hlgv8 z2m 2020-12-01T10:47:39.930Z z2m:Zwave Success zwave api call getAssociations [ { nodeId: 1, endpoint: 1 }, { nodeId: 1, endpoint: 0 } ]
zwavejs2mqtt-6bc86c4689-hlgv8 z2m 2020-12-01T10:47:42.548Z z2m:Zwave Node 64: metadata updated: 49-0-Power
zwavejs2mqtt-6bc86c4689-hlgv8 z2m 2020-12-01T10:47:42.549Z z2m:Zwave Node 64: value updated: 49-0-Power 80.1 => 84.8
zwavejs2mqtt-6bc86c4689-hlgv8 z2m 2020-12-01T10:48:09.236Z z2m:Zwave Node 63: value updated: 50-2-value-66049 0 => 0
```

Node still 52

AlCalzone, 2020-12-01:
Ahh, the error message is a bug in the code.

But nonetheless, the removeAssociation call does not do anything on the device. Maybe you need to try and factory-reset it after all - although I can't guarantee that this extra association won't come back. Might be a firmware bug.

aretakisv, 2020-12-01:
if this comes back, I will open a bug to AEOTEC as it is a serious one . Will keep you posted

PR #1203 (2020-12-13): fix: don't complain about missing MCA support in the fallback branch
In #1094 we added support to remove associations that are duplicated between Multichannel and normal Associations. This caused errors when the fallback branch was executed while trying to remove endpoint associations. The fallback logic wasn't really sound, since the thrown error used to imply that there's no MCA support - which is no longer true after #1094.

Now we only execute the fallback when there's a node association to be removed. The error is only thrown when trying to remove endpoint associations and if there is rea [... cut]
closed 2020-12-13 by AlCalzone