zwave-js/zwave-js #1180: Turn ZWave+ CC's roleType and nodeType into internal Value IDs and expose them through the node instance

AlCalzone, 2020-12-01:
as discussed in #1179

robertsLando, 2020-12-02:
@AlCalzone What about also `name` and `location` ?

robertsLando, 2020-12-02:
In zwavejs2mqtt I have a custom management of them as I had many and many problems with ozw nodes names and locations. I think I will keep my custom implementation but could help someone else (this means they will become internal too)

AlCalzone, 2020-12-02:
That would be the `Node Naming And Location CC` properties. I'm fine with reusing those properties for our purpose.

Setting them on the node (if the CC is supported) should still happen through the corresponding API.

robertsLando, 2020-12-03:
node type and role type are specific to endpoints, sot it should be added to node endpoints

AlCalzone, 2020-12-03:
Good point. The node instance automatically inherits the one for endpoint 0.

PR #1205 (2020-12-14): refactor: expose Z-Wave+ properties through endpoint/node instances
fixes: #1180
closed 2020-12-14 by AlCalzone