zwave-js/zwave-js #1181: Thermostat Air Temp not being reported or updated

chilicheech, 2020-12-02:
**Describe the bug**
My 2gig CT32 thermostat will report the air temperature during the initial interview, but will not report/update any temperature changes after that.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Device information**
Which device(s) is/are affected?  
Thermostat: 2gig CT32
What are the node IDs?
`31-49-1-Air temperature`

**Logfile:** 
[zwave-1.log](https://github.com/zwave-js/node-zwave-js/files/5626191/zwave-1.log)

chilicheech, 2020-12-02:
So looks like it is actually being reported:
```
12:55:57.401 CNTRLR   [Node 031] Interview stage completed: Neighbors
12:55:57.401 CNTRLR   [Node 031] Interview completed
13:01:11.753 CNTRLR   [Node 031] [Multilevel Sensor] Air temperature: metadata updated  [Endpoint 0]
13:01:11.757 CNTRLR   [Node 031] [~] [Multilevel Sensor] Air temperature: 65 => 64.5    [Endpoint 0]
13:01:11.768 DRIVER « [Node 031] [REQ] [ApplicationCommand]
13:09:11.762 CNTRLR   [Node 031] [Multilevel Sensor] Air temperature: metadata updated  [Endpoint 0]
13:09:11.763 CNTRLR   [Node 031] [~] [Multilevel Sensor] Air temperature: 64.5 => 65    [Endpoint 0]
13:09:11.767 DRIVER « [Node 031] [REQ] [ApplicationCommand]
13:24:11.803 CNTRLR   [Node 031] [Multilevel Sensor] Air temperature: metadata updated  [Endpoint 0]
13:24:11.805 CNTRLR   [Node 031] [~] [Multilevel Sensor] Air temperature: 65 => 64.5    [Endpoint 0]
13:24:11.809 DRIVER « [Node 031] [REQ] [ApplicationCommand]
13:39:48.872 CNTRLR   [Node 031] [Multilevel Sensor] Air temperature: metadata updated  [Endpoint 0]
13:39:48.874 CNTRLR   [Node 031] [~] [Multilevel Sensor] Air temperature: 64.5 => 65    [Endpoint 0]
13:39:48.879 DRIVER « [Node 031] [REQ] [ApplicationCommand]
14:02:24.789 CNTRLR   [Node 031] [Multilevel Sensor] Air temperature: metadata updated  [Endpoint 0]
14:02:24.791 CNTRLR   [Node 031] [~] [Multilevel Sensor] Air temperature: 65 => 64.5    [Endpoint 0]
14:02:24.796 DRIVER « [Node 031] [REQ] [ApplicationCommand]
14:02:24.818 CNTRLR   [Node 031] [Multilevel Sensor] Humidity: metadata updated         [Endpoint 0]
14:02:24.820 CNTRLR   [Node 031] [~] [Multilevel Sensor] Humidity: 33 => 31             [Endpoint 0]
14:02:24.825 DRIVER « [Node 031] [REQ] [ApplicationCommand]
14:08:11.743 CNTRLR   [Node 031] [Multilevel Sensor] Air temperature: metadata updated  [Endpoint 0]
14:08:11.745 CNTRLR   [Node 031] [~] [Multilevel Sensor] Air temperature: 64.5 => 65    [Endpoint 0]
14:08:11.748 DRIVER « [Node 031] [REQ] [ApplicationCommand]
14:39:36.273 CNTRLR   [Node 031] [Multilevel Sensor] Air temperature: metadata updated  [Endpoint 0]
14:39:36.275 CNTRLR   [Node 031] [~] [Multilevel Sensor] Air temperature: 65 => 65.5    [Endpoint 0]
14:39:36.280 DRIVER « [Node 031] [REQ] [ApplicationCommand]
14:39:36.296 CNTRLR   [Node 031] [Multilevel Sensor] Humidity: metadata updated         [Endpoint 0]
14:39:36.298 CNTRLR   [Node 031] [~] [Multilevel Sensor] Humidity: 31 => 32             [Endpoint 0]
14:39:36.325 DRIVER « [Node 031] [REQ] [ApplicationCommand]
15:16:11.761 CNTRLR   [Node 031] [Multilevel Sensor] Air temperature: metadata updated  [Endpoint 0]
15:16:11.763 CNTRLR   [Node 031] [~] [Multilevel Sensor] Air temperature: 65.5 => 66    [Endpoint 0]
15:16:11.769 DRIVER « [Node 031] [REQ] [ApplicationCommand]
15:55:35.810 CNTRLR   [Node 031] [Multilevel Sensor] Air temperature: metadata updated  [Endpoint 0]
15:55:35.812 CNTRLR   [Node 031] [~] [Multilevel Sensor] Air temperature: 66 => 66.5    [Endpoint 0]
15:55:35.817 DRIVER « [Node 031] [REQ] [ApplicationCommand]
15:55:35.831 CNTRLR   [Node 031] [Multilevel Sensor] Humidity: metadata updated         [Endpoint 0]
15:55:35.833 CNTRLR   [Node 031] [~] [Multilevel Sensor] Humidity: 32 => 33             [Endpoint 0]
15:55:35.838 DRIVER « [Node 031] [REQ] [ApplicationCommand]
17:36:11.809 CNTRLR   [Node 031] [Multilevel Sensor] Air temperature: metadata updated  [Endpoint 0]
17:36:11.811 CNTRLR   [Node 031] [~] [Multilevel Sensor] Air temperature: 66.5 => 67    [Endpoint 0]
17:36:11.816 DRIVER « [Node 031] [REQ] [ApplicationCommand]
17:36:11.832 CNTRLR   [Node 031] [Multilevel Sensor] Humidity: metadata updated         [Endpoint 0]
17:36:11.834 CNTRLR   [Node 031] [~] [Multilevel Sensor] Humidity: 33 => 34             [Endpoint 0]
17:36:11.838 DRIVER « [Node 031] [REQ] [ApplicationCommand]
17:53:51.880 CNTRLR   [Node 031] [Multilevel Sensor] Humidity: metadata updated         [Endpoint 0]
17:53:51.883 CNTRLR   [Node 031] [~] [Multilevel Sensor] Humidity: 34 => 35             [Endpoint 0]
17:53:51.888 DRIVER « [Node 031] [REQ] [ApplicationCommand]
19:09:49.007 CNTRLR   [Node 031] [Multilevel Sensor] Air temperature: metadata updated  [Endpoint 0]
19:09:49.009 CNTRLR   [Node 031] [~] [Multilevel Sensor] Air temperature: 67 => 67.5    [Endpoint 0]
```
but it's not updating in mqtt with zjs2m. mqtt has this:
```
{
  "id": "31-49-1-Air temperature",
  "nodeId": 31,
  "commandClass": 49,
  "commandClassName": "Multilevel Sensor",
  "endpoint": 1,
  "property": "Air temperature",
  "propertyName": "Air temperature",
  "type": "number",
  "readable": true,
  "writeable": false,
  "label": "Air temperature",
  "ccSpecific": {
    "sensorType": 1,
    "scale": 1
  },
  "genre": "user",
  "unit": "°F",
  "list": false,
  "value": 65,
  "lastUpdate": 1606856126192
}
```
lastUpdate is Tuesday, December 1, 2020 12:55:26.192 PM GMT-08:00 in my timezone. The weird thing is that endpoint 0 is updating the air temp instead of endpoint 1. further up in the log i see this:
```
12:55:26.181 CNTRLR   [Node 031] [Multilevel Sensor] Air temperature: metadata updated  [Endpoint 1]
12:55:26.191 CNTRLR   [Node 031] [~] [Multilevel Sensor] Air temperature: 65 => 65      [Endpoint 1]
```
I recently modified the thermostat config in zjs2m to read from endpoint 1 instead of endpoint 0 because endpoint 0 wasn't showing up in mqtt.. instead endpoints 1 and 2 do show up.

[image]

CC'ing @robertsLando just in case he has any ideas.

robertsLando, 2020-12-02:
@chilicheech Are you using latest version from master?

robertsLando, 2020-12-02:
> but it's not updating in mqtt with zjs2m. mqtt has this:

Could you also show me the zjs2mqtt log?

AlCalzone, 2020-12-02:
`zwave-js` has functionality to map root endpoint reports to endpoint 1, which isn't working in this case.

Seems like my logic (is Multi Channel Association CC V3+ supported?) does not consider the case that Multi Channel Association CC is not supported at all. 🙈

AlCalzone, 2020-12-02:
@chilicheech can you try https://github.com/zwave-js/node-zwave-js/pull/1182 ?

chilicheech, 2020-12-02:
@AlCalzone I just tried it.. It is working. The thermostat is now reporting both the air temperature and the humidity. Thanks for the fix 😄

PR #1182 (2020-12-02): fix: also map reports from the root endpoint to endpoint 1 if MCA is not supported at all
fixes: #1181
closed 2020-12-02 by AlCalzone