zwave-js/zwave-js #1184: Not all values are present when node is ready

robertsLando, 2020-12-03:
After some digging into this problem I discovered that when the `node ready` event is triggered if I call `zwaveNode.getDefinedValueIDs()` that doesn't return all valueids, expecially seems that for this specific device it is missing the valueIds for endpoints > 0: 

```
2020-12-03T08:39:19.676Z z2m:Zwave Node 3 ready: ThermoFloor - Heatit Z-TRM3 (Floor thermostat)
2020-12-03T08:39:19.733Z z2m:Zwave Node 3: metadata updated: 49-2-Air temperature
2020-12-03T08:39:19.734Z z2m:Zwave Node 3: value updated: 49-2-Air temperature 24.8 => 24.8
2020-12-03T08:39:20.053Z z2m:Zwave Node 3: metadata updated: 49-3-Air temperature
2020-12-03T08:39:20.054Z z2m:Zwave Node 3: value updated: 49-3-Air temperature 0 => 0
2020-12-03T08:39:21.676Z z2m:Zwave Node 3: metadata updated: 49-4-Air temperature
2020-12-03T08:39:21.676Z z2m:Zwave Node 3: value updated: 49-4-Air temperature 19 => 19
```

Ref: https://github.com/zwave-js/zwavejs2mqtt/issues/38#issuecomment-737732308

AlCalzone, 2020-12-03:
> for this specific device it is missing the valueIds for endpoints > 0

I do see values for the individual endpoints. Not every endpoint has the same functionality. Which values are expected that aren't present here?

robertsLando, 2020-12-03:
Those that comes immediatly after the node ready:

```2020-12-03T08:39:19.733Z z2m:Zwave Node 3: metadata updated: 49-2-Air temperature
2020-12-03T08:39:19.734Z z2m:Zwave Node 3: value updated: 49-2-Air temperature 24.8 => 24.8
2020-12-03T08:39:20.053Z z2m:Zwave Node 3: metadata updated: 49-3-Air temperature
2020-12-03T08:39:20.054Z z2m:Zwave Node 3: value updated: 49-3-Air temperature 0 => 0
2020-12-03T08:39:21.676Z z2m:Zwave Node 3: metadata updated: 49-4-Air temperature
2020-12-03T08:39:21.676Z z2m:Zwave Node 3: value updated: 49-4-Air temperature 19 => 19
```

AlCalzone, 2020-12-03:
Since it is a value updated event, not a value added event, they should already be there. Are they not returned with `getDefinedValueIDs`?

robertsLando, 2020-12-03:
Nope, them are not returned or you would see them also in value added. That's the strange thing

AlCalzone, 2020-12-03:
Can you ask for the cache files please, so I can try to reproduce this in a test?

roflmao, 2020-12-03:
[zwavejs-cache.zip](https://github.com/zwave-js/node-zwave-js/files/5635888/zwavejs-cache.zip)

robertsLando, 2020-12-03:
Just for reference

https://github.com/zwave-js/zwavejs2mqtt/blob/master/lib/ZwaveClient.js#L284

Here is where I add all values when I receive the node ready event, and I would expect to have all the valueIds at that point

AlCalzone, 2020-12-03:
According to the cached information, Node 3 doesn't support Multilevel Sensor CC, therefore `getDefinedValueIDs` does not return those values.
@roflmao please try re-interviewing it to see if that was a glitch. If it still doesn't work afterwards we need to mark it as supported through the configuration file.

roflmao, 2020-12-03:
@AlCalzone restart or refresh node?

AlCalzone, 2020-12-03:
Refresh node, that resets the information we know about the node, including the supported CCs.

roflmao, 2020-12-03:
[zwavejs-cache.zip](https://github.com/zwave-js/node-zwave-js/files/5637379/zwavejs-cache.zip)

robertsLando, 2020-12-03:
@AlCalzone Seems only node 2 and 7 are reporting 0x31 as supported CC. So we should add a compact flag for this? @roflmao Could you add the hex identifier of this device? You can find it in control panle when you select a node near the device id

roflmao, 2020-12-03:
Device ID: 411-515-3 (0x019b-0x0203-0x0003)

robertsLando, 2020-12-03:
https://github.com/zwave-js/node-zwave-js/blob/master/packages/config/config/devices/0x019b/z-trm3.json

AlCalzone, 2020-12-03:
Seems like OZW specifically adds the CC and its values through the config files. Wonder if the device will respond to requests though.

robertsLando, 2020-12-04:
> Wonder if the device will respond to requests though.

Based on my little experience in issues here I say that it may work :) It's like the stopLevelChange that was reported as unsupported and then just by sending it without checks works :man_shrugging:

AlCalzone, 2020-12-04:
>  It's like the stopLevelChange that was reported as unsupported

That was different. Sending it with Supervision was unsupported. The command itself is supported or the zwave-js method would throw immediately.

AlCalzone, 2020-12-04:
@roflmao could you try the `compat-add-cc` branch and see if this resolves your issue?

roflmao, 2020-12-06:
I still get the same in the logs and now the device isnt identified:

```
 2020-12-06T20:24:56.019Z z2m:Gateway Unable to discover climate device, there is no valid temperature/thermostat mode valueIds                                                                                    
 2020-12-06T20:24:56.021Z z2m:Zwave Node 3 ready: Unknown manufacturer 411 - Unknown product 515 (3)    
```
[zwavejs-cache.zip](https://github.com/zwave-js/node-zwave-js/files/5649488/zwavejs-cache.zip)

AlCalzone, 2020-12-06:
Hmm, seems it doesn't find the device config file now. I'll investigate.

roflmao, 2020-12-06:
@AlCalzone same goes for master

AlCalzone, 2020-12-06:
Hmm, I can't reproduce the behavior in tests where the file is loaded correctly. Only the changes were not applied because I forgot a line, but that shouldn't cause the lookup to fail.

Are you sure that the dependencies are included correctly?

roflmao, 2020-12-06:
node_modules:

[node_modules.find.log](https://github.com/zwave-js/node-zwave-js/files/5649580/node_modules.find.log)

Or test with docker image: roflmao/zwavejs2mqtt:compat-add-cc

roflmao, 2020-12-06:
[npm.ls.log](https://github.com/zwave-js/node-zwave-js/files/5649586/npm.ls.log)

Maybe this helps (npm ls --json output)

AlCalzone, 2020-12-07:
The libraries seem ok, although I'm not sure how the symlinks in the zwave-js workspace are displayed there.

Since it works on my machine (and I can't test whats going on without your device), I guess you'll need to add some logging:
`@zwave-js/config/config/build/Devices.js`:
```diff
// lines 113-114:
-        if (!(await fs_extra_1.pathExists(filePath)))
-            return;
+        if (!(await fs_extra_1.pathExists(filePath))) {
+            Logger_1.default.config.print(`${filePath} does not exist`, "error");
+            return;
+        }

// before line 111
+    if (!indexEntry) Logger_1.default.config.print(`did not find device config for ${manufacturerId}, ${productType}, ${productId}, ${firmwareVersion}`, "error");
```

roflmao, 2020-12-07:
Ok, just located an error with the latest Dockerfile.contrib dependency-wise.

Here it is with correct zwave-js branch ```compat-add-cc``` and dependencies:

[zwavejs-cache.zip](https://github.com/zwave-js/node-zwave-js/files/5652650/zwavejs-cache.zip)

robertsLando, 2020-12-07:
@roflmao Could you also give a check to z2m master now? I have submitted a fix that may fix some discovery issues

roflmao, 2020-12-07:
@robertsLando 
[zwavejs-log-bundle.zip](https://github.com/zwave-js/node-zwave-js/files/5653430/zwavejs-log-bundle.zip)

robertsLando, 2020-12-07:
@roflmao Could you also tell me something? :) Better ? Worse? same? new errors?

roflmao, 2020-12-07:
@robertsLando Haha, sorry! Its the same it looks to me. First unable to climate, then climate discovered.

robertsLando, 2020-12-07:
Ok I see the error is still there but it's a device compatibility error as that CC isn't reported as supported. Once fixed you should not see that error on console :)

AlCalzone, 2020-12-07:
That's what the mentioned branch should fix. I'll take a look at the logs later to see if I can recognize a problem.

AlCalzone, 2020-12-07:
@roflmao it looks a bit better:
```
16:00:39.700 CNTRLR   [Node 003] trying to load device config
```
but the additional CCs still don't show up in the network cache. Is your test based on the latest commit of the branch?

AlCalzone, 2020-12-07:
So, I tested locally and modified a config file to add a non-supported CC to the list. Here's my MultiSensor 6 cache file after the interview:
```
...
                "0x51": {
                    "name": "Move To Position Window Covering",
                    "endpoints": {
                        "0": {
                            "isSupported": true,
                            "isControlled": false,
                            "secure": false,
                            "version": 0
                        }
                    }
                },
...
```

roflmao, 2020-12-07:
Ok my sha tracking has been very much needed so i hacked together something:

```
root@bender:/usr/src/app# cat package.json  | grep \"version
  "version": "1.0.0-alpha.0+b9a6315",
root@bender:/usr/src/app# cd node_modules/zwave-js/
root@bender:/usr/src/app/node_modules/zwave-js# cat package.json  | grep \"version
  "version": "5.5.0-fe071207+791f5736",
```

Confirming:
```
# cat node-zwave-js.version-sha version-sha 
fe071207
b9a6315
```

Logs: 
[zwavejs-log-bundle.zip](https://github.com/zwave-js/node-zwave-js/files/5655342/zwavejs-log-bundle.zip)

AlCalzone, 2020-12-07:
Hmmm. I still don't see changes to the network cache file. I guess there's no way that you can step through `packages\zwave-js\src\lib\node\Node.ts` (lines 1539-1549) with a debugger?

roflmao, 2020-12-07:
Sure, just tell me how :)

AlCalzone, 2020-12-08:
The docker part makes it harder - maybe @robertsLando has some insight?
I usually open a repo in VSCode, build it and execute the main entry point in a Javascript Debug Terminal or start a dedicated task.

roflmao, 2020-12-08:
@AlCalzone Im up for anything, debug statements, remote shell etc.

mpowell90, 2020-12-09:
Not sure if this is related, I also have an Aeotec Multisensor 6, just started testing with it today.
Parameters 111, 112, 113 (set report group time intervals) aren't included in the getDefinedValueIDs array.
However, after digging through the product spec, I was able to set these parameter values successfully.

robertsLando, 2020-12-09:
@roflmao https://code.visualstudio.com/docs/containers/debug-node

AlCalzone, 2020-12-09:
> Parameters 111, 112 and 113 (set report group time intervals) aren't included in the getDefinedValueIDs array

They are defined in the config file, so my guess is that reading them from the device failed. However I can't say for sure without logs.

AlCalzone, 2020-12-13:
@roflmao did you have some success with the debugger yet?

roflmao, 2020-12-13:
@AlCalzone Sorry but ive been busy with dead nodes in my network so need to get it up and running first. Also im not so familiar with vscode but ill try when i get the time

PR #1192 (2020-12-20): feat(compat): add additional CCs defined in config files
fixes: #1184
closed 2020-12-20 by AlCalzone