zwave-js/zwave-js #1189: Notifications from Vision Security ZD2105 not being parsed

chilicheech, 2020-12-04:
**Version**

Build/Run method
- [x] Docker
- [ ] PKG
- [ ] Manually built (git clone - npm install - npm run build ) 

zwavejs2mqtt version: 1.2.3

**Describe the bug**
I have Vision Security Recessed Door Window Sensor (ZD2105) that has command class 113 with `26-113-0-Access Control-Door state` and `26-113-0-Home Security-Cover status`. The mqtt payload for both of these have no `value` field. Here's an example:
```
{
  "id": "26-113-0-Access Control-Door state",
  "nodeId": 26,
  "commandClass": 113,
  "commandClassName": "Notification",
  "endpoint": 0,
  "property": "Access Control",
  "propertyName": "Access Control",
  "propertyKey": "Door state",
  "type": "number",
  "readable": true,
  "writeable": false,
  "label": "Door state",
  "genre": "system",
  "min": 0,
  "max": 255,
  "list": true,
  "states": [
    {
      "text": "Window/door is open",
      "value": 22
    },
    {
      "text": "Window/door is closed",
      "value": 23
    }
  ],
  "lastUpdate": 1606856089699
}
```
Without the `value` it won't report when the sensor is triggered.

**Expected behavior**
The payload should have a `value` key/field that reports a value from the `states` list field. So in this case the value of the `value` field would be either `0`, `22`, or `23`.

AlCalzone, 2020-12-04:
Ok the problem is the following:
Notification CC V2+ require the legacy V1 alarm bytes to be zero. This device doesn't do that. Should be easy to fix.

AlCalzone, 2020-12-04:
@chilicheech try again :)

closed 2020-12-04 by AlCalzone

chilicheech, 2020-12-04:
looks like it's reporting correctly now :)
```
15:43:44.549 DRIVER « [Node 026] [REQ] [ApplicationCommand]
                      └─[BasicCCSet]
                          target value: 255
15:43:44.564 CNTRLR   [Node 026] treating BasicCC Set as a report
15:43:44.567 CNTRLR   [Node 026] [~] [Basic] currentValue: 0 => 255                     [Endpoint 0]
15:43:44.649 SERIAL « 0x01100004001a0a710506ff00ff0616000099                              (18 bytes)
15:43:44.654 SERIAL » [ACK]                                                                   (0x06)
15:43:44.660 DRIVER « [Node 026] [REQ] [ApplicationCommand]
                      └─[NotificationCCReport]
                          notification type:   Access Control
                          notification status: 255
                          notification state:  Window/door is open
15:43:44.673 CNTRLR   [Node 026] [+] [Notification] Access Control[Door state]: 22      [Endpoint 0]
15:43:52.252 SERIAL « 0x01090004001a03200100ca                                            (11 bytes)
15:43:52.255 SERIAL » [ACK]                                                                   (0x06)
15:43:52.260 DRIVER « [Node 026] [REQ] [ApplicationCommand]
                      └─[BasicCCSet]
                          target value: 0
15:43:52.268 CNTRLR   [Node 026] treating BasicCC Set as a report
15:43:52.270 CNTRLR   [Node 026] [~] [Basic] currentValue: 255 => 0                     [Endpoint 0]
15:43:52.361 SERIAL « 0x01100004001a0a7105060000ff0617000067                              (18 bytes)
15:43:52.365 SERIAL » [ACK]                                                                   (0x06)
15:43:52.370 DRIVER « [Node 026] [REQ] [ApplicationCommand]
                      └─[NotificationCCReport]
                          notification type:   Access Control
                          notification status: 255
                          notification state:  Window/door is closed
15:43:52.380 CNTRLR   [Node 026] [~] [Notification] Access Control[Door state]: 22 => 2 [Endpoint 0]
```
also, the mqtt payload now has a value key:
```
{
  "id": "26-113-0-Access Control-Door state",
  "nodeId": 26,
  "commandClass": 113,
  "commandClassName": "Notification",
  "endpoint": 0,
  "property": "Access Control",
  "propertyName": "Access Control",
  "propertyKey": "Door state",
  "type": "number",
  "readable": true,
  "writeable": false,
  "label": "Door state",
  "genre": "system",
  "min": 0,
  "max": 255,
  "list": true,
  "states": [
    {
      "text": "Window/door is open",
      "value": 22
    },
    {
      "text": "Window/door is closed",
      "value": 23
    }
  ],
  "value": 23,
  "lastUpdate": 1607125432381
}
```
So, things are looking good from the zjs side. On the zjs2m side the dashboard only shows the currentValue and targetValue and it doesn't show the Notification CC values. Also the hass discovery isn't creating a sensor for the door. I'll open an issue on zjs2m for that.

Thanks so much for your help :)

robertsLando, 2020-12-07:
> On the zjs2m side the dashboard only shows the currentValue and targetValue and it doesn't show the Notification CC values. Also the hass discovery isn't creating a sensor for the door. I'll open an issue on zjs2m for that.

Did you opened the issue?

Cyberwizzard, 2021-01-25:
I seem to have the same issue with a Fibaro Z-Wave Window / Door Sensor 2. ZwaveJS correctly seems to report the state in the debug logs when the sensor state changes, but HomeAssistant does not seem to receive this info.

What is the ticket for the zjs2m project?

robertsLando, 2021-01-26:
@Cyberwizzard I answered you on the other issue

PR #1191 (2020-12-04): fix(cc): remove check if V1 alarm bytes are zero
fixes: #1189