zwave-js/zwave-js #1208: [bug] Aeotec Z-Stick Gen5+ has no configuration options

katiuskt, 2020-12-14:
**Version**

Build/Run method
- [X] Docker
- [ ] PKG
- [ ] Manually built (git clone - npm install - npm run build ) 

zwavejs2mqtt version: I'm not sure how to get this. I'm using zwavejs/zwavejs2mqtt:dev pulled on 12th Dec

**Describe the bug**
Newest Aeotec Z-Stick Gen5+ (https://aeotec.com/z-wave-usb-stick/) has no configuration options at zwavejs2mqtt GUI
[image]

While they are available using OpenZwave stack:
[image]

**Device information**
Which device(s) is/are affected?  Aeotec Z-Stick Gen5+
What are the node IDs? 1

**Logfile:** 
Despite I have selected "Log Level=Debug and Log to file is enabled, I'm not able to find the log file but I attach <hex-number>.json, <hex-number>.values.jsonl files: [json_files.zip](https://github.com/zwave-js/zwavejs2mqtt/files/5684373/json_files.zip)

**Expected behavior**
Device information/settins are expected to be shown into zwavejs2mqtt GUI both for information sake and for tunning it too.

katiuskt, 2020-12-17:
Just in case it helps: ZWave2MQTT seems to be able to manage the stick options:
[image]

I'm not sure it will be relevant because I think ZWave2MQTT relies on OZW too but I'm still learning about all the implementations (ZWay, OpenZWave, ZWave2MQTT, ZWaveJS, ...) so just let the information in case it could be helpful somehow.

robertsLando, 2020-12-18:
@katiuskt Could you try #1032 ?

robertsLando, 2020-12-18:
@katiuskt Yep because zwave-js doesn't interview controllers, check updates on #1032

AlCalzone, 2020-12-20:
@katiuskt could you try https://github.com/zwave-js/node-zwave-js/pull/1230 please?

katiuskt, 2020-12-21:
@AlCalzone , sure, glad to help but I'm afraid I'm newbie in these github/development stuff so I'm afraid I need some help from your side: I'm using zwavejs2mqtt docker images (dev ones).

I've updated it to its most recent version (apparently it has 7 hours) and I can't see any change.

Taking a look at https://github.com/zwave-js/node-zwave-js/pull/1230 I see it has not been merged so I guess I isn't available on any docker image because it needs to be tested or whatever and, I guess at that point, I'm the guy to check it but... I have no idea of how to check if it works unless it is already included into a docker image :-(

But I can learn :-) Apparently I need to put some lines into the "packages/config/config/devices/0x0086/zw090.json", isn't it? I'll keep trying to figure out how to do that in my current docker image but if you guys can give me a clue about the process I'll appreciate.

(Sorry not to have enough skills to fully understand everything related with this development world in general/zwavejs in particular)

AlCalzone, 2020-12-21:
I think there's some way in the zwavejs2mqtt docker image to install a specific branch. @robertsLando can you explain how?

robertsLando, 2020-12-21:
Check instructions in docker folder: https://github.com/zwave-js/zwavejs2mqtt/tree/master/docker#building-a-container-using-dockerfilecontrib

katiuskt, 2020-12-22:
Thank you @robertsLando , I'll try (disclaimer: no previous expertise on that)

AlCalzone, 2020-12-23:
I've investigated a bit more. It seems that the config params are only useful for secondary controllers. They cannot be used to configure the stick itself.

https://github.com/OpenZWave/open-zwave/issues/2102
> You can't set params on your Controller. (if you have a secondary controller, you could setup config params on it).

https://github.com/zwave-js/node-zwave-js/discussions/1241#discussioncomment-238256
> The test was performed on some old code of mine, that allows setting configuration values for ZWave Nodes (.Net, C#)
it takes the Node ID, Config Param, Config Value, and the values size.
packs it up - setting the checksum of course and sending to the controller.
> the same snippet of code, sets the config values for Non-Controller devices without a problem.

However they can be set through the Serial API, but that will require some more work:
https://github.com/zwave-js/node-zwave-js/issues/1245

closed 2020-12-23 by AlCalzone