zwave-js/zwave-js #1218: [bug] AEOTEC Range Extender 7 is not identified.

cinadr, 2020-12-17:
**Version**

Build/Run method
- [ ] Docker
- [ ] PKG 
- [x] Manually built (git clone - npm install - npm run build )  - running on Windows Server 2019 core

zwavejs2mqtt version: latest alpha

**Describe the bug**
former ZWave2MQTT identified my zwave repeater. This version says unknown manufacturer 881, under Product states number 4, and product code is Unkown Product 189
Tried to Heal and request Info from the node several times. No success.

**Expected behavior**
Correctly identify product as an AEON Labs product [Range Extender 7](https://aeotec.com/z-wave-repeater/).

AlCalzone, 2020-12-17:
Yup, we have no config file for that one.

robertsLando, 2020-12-18:
@cinadr there is an open pr for this

billiaz, 2020-12-18:
I pushed a pull for this and an update on fibaro

AlCalzone, 2020-12-18:
closed in #1220

closed 2020-12-18 by AlCalzone