zwave-js/zwave-js #1219: Door lock fails interview

smitt04, 2020-12-17:
**Describe the bug**
Interviewing fails for a schlage door lock. This is in regards to #1214 

**Device information**
Which device(s) is/are affected?  
What are the node IDs?

NODE ID 12

**Logfile:**
[zwave-4161.log](https://github.com/zwave-js/node-zwave-js/files/5712615/zwave-4161.log)

smitt04, 2020-12-17:
@AlCalzone This is in reference to the zwave2mqtt issue we have been discussing

AlCalzone, 2020-12-21:
Okay it seems this device is actually just a bit slow to respond.

AlCalzone, 2020-12-21:
So this would be an argument to do https://github.com/zwave-js/node-zwave-js/issues/997

AlCalzone, 2020-12-21:
As a short-term workaround, increasing `timeouts.response` could work if there's an option to do that in zjs2mqtt.
@robertsLando ?

robertsLando, 2020-12-21:
Actually there isn't an option for that but I can add it

AlCalzone, 2020-12-21:
Generally, people shouldn't need to play around with it, so maybe hide it in expert options or somewhere ;)

smitt04, 2020-12-21:
@AlCalzone I tried it with the maximum allowed `5000ms` in `timeouts.response` but seemed to make no difference

AlCalzone, 2020-12-21:
@robertsLando is that already implemented in zjs2mqtt?
@smitt04 share a log please :)

smitt04, 2020-12-21:
I modified my own instance of the library. I only saw one place the driver was instantiated and if I put in a larger number then 5000 the logs complained and the driver never connected, so I assumed it was working. I will get you a new log in a little bit

AlCalzone, 2020-12-21:
Uh... `timeouts.response` was not correct 😅
`timeouts.report` is the one you need to modify. That goes up to 40000, but you should get by with 5000 to 10000.
The longer this timeout is, the longer nothing happens when a node does not respond.

smitt04, 2020-12-21:
[zwave-11479.log](https://github.com/zwave-js/node-zwave-js/files/5726380/zwave-11479.log)

@AlCalzone I tried 5000, then 10000 and neither one made a difference on my end. Attached is the 10000ms logs

AlCalzone, 2020-12-21:
It did make a difference. You just didn't notice because the interview still doesn't go though.
The device doesn't respond to `DoorLockCCOperationGet` at all.

AlCalzone, 2020-12-21:
PR #1237 should fix that. Can you try that branch to make sure that we found all the quirks?

smitt04, 2020-12-22:
@AlCalzone Updated log file
[zwave-12257.log](https://github.com/zwave-js/node-zwave-js/files/5727341/zwave-12257.log)

AlCalzone, 2020-12-22:
Ugh... now it doesn't respond to the user code query...

AlCalzone, 2020-12-22:
Try again please :)

smitt04, 2020-12-22:
@AlCalzone attached is the latest log

I was also looking for more info on what the door lock supports and found these resources

classes: https://products.z-wavealliance.org/products/3130/classes
configuration: [24352403.pdf](https://github.com/zwave-js/node-zwave-js/files/5730900/24352403.pdf)
log: [zwave-13787.log](https://github.com/zwave-js/node-zwave-js/files/5730924/zwave-13787.log)
more info: http://manuals-backend.z-wave.info/make.php?lang=en&sku=BE469ZP&cert=ZC10-18106281

AlCalzone, 2020-12-22:
I don't get it... now another command fails that was supported earlier. What's especially weird is how the communication goes:
```
--> NonceGet (to send encrypted command to node)
<-- NonceReport (from node)
--> encrypted command
<-- NonceGet (to send encrypted response)
--> NonceReport (from zwave-js to node)
... nothing
```
If the node does not support the encrypted command, I would expect the last two commands **not** to be exchanged.
So this smells more like the node's responses get lost (instead of it not responding like I thought earlier).

AlCalzone, 2020-12-22:
BTW the same happened in your earlier logs - I just didn't notice.

smitt04, 2020-12-22:
I wonder if part of the problem is that the lock is a zwave plus and all the other devices except for one is zwave classic. I was planning on upgrading them in the new year. If you think it would help I could try removing all other devices and try the lock by itself

AlCalzone, 2020-12-22:
No idea if it helps, but it is certainly worth a try.

smitt04, 2020-12-22:
excluded all devices, did a factory reset on the controller and lock and only included the lock

[zwave-13916.log](https://github.com/zwave-js/node-zwave-js/files/5731288/zwave-13916.log)

AlCalzone, 2020-12-22:
That thing doesn't respond to **anything**
(╯°□°）╯︵ ┻━┻

Could you try it out with zwave2mqtt (with OZW) and make a log so I can check if there's a difference?

smitt04, 2020-12-22:
Here is the OZW logs

[OZW_Log.log](https://github.com/zwave-js/node-zwave-js/files/5731447/OZW_Log.log)

smitt04, 2020-12-22:
I haven't tried doing everything i want to do with the lock, but I tried smart things a couple of weeks ago and it seemed to work, not sure if it completed a full interview or not. If all else fails I can see if i can get it fully working on ST, i just didn't want it connected to the cloud

AlCalzone, 2020-12-23:
Ok the OZW log is also full of these (with all commands):
```
2020-12-22 18:24:29.210 Error, Node002, ERROR: Dropping command, expected response not received after 1 attempt(s). Command: "UserNumberCmd_Get (Node=2): 0x01, 0x09, 0x00, 0x13, 0x02, 0x02, 0x63, 0x04, 0x25, 0x15, 0xb2"
```

Just a thought: your device supports `Antitheft CC V2`. Maybe that was activated in ST, so now it can't be used in another network? From the specs:
> The Anti-theft Command Class is used to disable all command classes related to the actual application
functionality of a device if is it excluded and later included in another network.
> The device MUST NOT respond to application commands while in the protected state. A device in
protected state MUST NOT leave its protected state if it is re-included into its home network. **Another
Anti-Theft Set command** MUST be used to either disable locking or to clear the protected state.

smitt04, 2020-12-23:
hmm, maybe. How do we send the Anti-Theft Set command? Also this device was originally included in HA with another zwave controller directly mounted to the HA instance.

AlCalzone, 2020-12-23:
It is not implemented yet :(
How are you using this library? Do you have a way to execute your own code?

smitt04, 2020-12-23:
I am currently using it with zwavejs2mqtt. I can write my own script if I need to. or if there is another driver i can execute it with i can try that.

AlCalzone, 2020-12-23:
I'll try to make you a little script later.

smitt04, 2020-12-23:
awesome, thanks

AlCalzone, 2020-12-23:
So, this is how you can query the anti-theft state of the device (to see if it is even used):

```ts
require("reflect-metadata");
import { CommandClasses } from "@zwave-js/core";
import { Driver } from "zwave-js";
import { CommandClass } from "zwave-js/build/lib/commandclass/CommandClass";

const driver = new Driver("/path/to/serial", {
	networkKey: Buffer.from("your-network-key", "hex"),
})
	.on("error", console.error)
	.once("driver ready", async () => {
		// wait a few seconds
		await require("alcalzone-shared/async").wait(5000);

		// Query the current status
		const antitheftGet = new CommandClass(driver, {
			nodeId: 2,
			payload: Buffer.from([
				CommandClasses["Anti-theft"],
				0x02, // GET
			]),
		});
		await driver.sendCommand(antitheftGet);

		// wait a few seconds
		await require("alcalzone-shared/async").wait(5000);
	});
void driver.start();
```

Put that in a js file, make sure that `zwave-js` is installed as a dependency. You might have to replace the imports with require statements.

After running this, you should hopefully get a response that includes the anti-theft status. I'll need to see that to decode it.

smitt04, 2020-12-23:
This is what i get
```
10:47:17.521 DRIVER   ███████╗ ██╗    ██╗  █████╗  ██╗   ██╗ ███████╗             ██╗ ███████╗
                      ╚══███╔╝ ██║    ██║ ██╔══██╗ ██║   ██║ ██╔════╝             ██║ ██╔════╝
                        ███╔╝  ██║ █╗ ██║ ███████║ ██║   ██║ █████╗   █████╗      ██║ ███████╗
                       ███╔╝   ██║███╗██║ ██╔══██║ ╚██╗ ██╔╝ ██╔══╝   ╚════╝ ██   ██║ ╚════██║
                      ███████╗ ╚███╔███╔╝ ██║  ██║  ╚████╔╝  ███████╗        ╚█████╔╝ ███████║
                      ╚══════╝  ╚══╝╚══╝  ╚═╝  ╚═╝   ╚═══╝   ╚══════╝         ╚════╝  ╚══════╝
10:47:17.535 DRIVER   version 5.6.1
10:47:17.537 DRIVER
10:47:17.538 DRIVER   starting driver...
10:47:17.579 DRIVER   opening serial port /dev/ttyUSB0
10:47:17.612 DRIVER   serial port opened
10:47:17.617 SERIAL » [NAK]                                                                   (0x15)
10:47:19.129 DRIVER   loading configuration...
10:47:20.140 DRIVER   beginning interview...
10:47:20.143 DRIVER   added request handler for AddNodeToNetwork (0x4a)...
                      1 registered
10:47:20.145 DRIVER   added request handler for RemoveNodeFromNetwork (0x4b)...
                      1 registered
10:47:20.146 DRIVER   added request handler for ReplaceFailedNode (0x63)...
                      1 registered
10:47:20.150 CNTRLR   beginning interview...
10:47:20.151 CNTRLR   querying version info...
10:47:20.604 SERIAL » 0x01030015e9                                                         (5 bytes)
10:47:20.609 DRIVER » [REQ] [GetControllerVersion]
10:47:20.628 SERIAL « [ACK]                                                                   (0x06)
10:47:20.635 SERIAL « 0x011001155a2d5761766520342e3035000197                              (18 bytes)
10:47:20.638 SERIAL » [ACK]                                                                   (0x06)
10:47:20.642 DRIVER « [RES] [GetControllerVersion]
                        payload: 0x5a2d5761766520342e30350001
10:47:20.661 CNTRLR   received version info:
                        controller type: Static Controller
                        library version: Z-Wave 4.05
10:47:20.662 CNTRLR   querying controller IDs...
10:47:20.675 SERIAL » 0x01030020dc                                                         (5 bytes)
10:47:20.676 DRIVER » [REQ] [GetControllerId]
10:47:20.681 SERIAL « [ACK]                                                                   (0x06)
10:47:20.685 SERIAL « 0x01080120d4998752014f                                              (10 bytes)
10:47:20.687 SERIAL » [ACK]                                                                   (0x06)
10:47:20.689 DRIVER « [RES] [GetControllerId]
                        payload: 0xd499875201
10:47:20.700 CNTRLR   received controller IDs:
                        home ID:     0xd4998752
                        own node ID: 1
10:47:20.701 CNTRLR   querying controller capabilities...
10:47:20.713 SERIAL » 0x01030005f9                                                         (5 bytes)
10:47:20.714 DRIVER » [REQ] [GetControllerCapabilities]
10:47:20.718 SERIAL « [ACK]                                                                   (0x06)
10:47:20.723 SERIAL « 0x0104010508f7                                                       (6 bytes)
10:47:20.724 SERIAL » [ACK]                                                                   (0x06)
10:47:20.726 DRIVER « [RES] [GetControllerCapabilities]
                        payload: 0x08
10:47:20.737 CNTRLR   received controller capabilities:
                        controller role:     primary
                        is in other network: false
                        is SIS present:      false
                        was real primary:    true
                        is a SUC:            false
10:47:20.738 CNTRLR   querying API capabilities...
10:47:20.755 SERIAL » 0x01030007fb                                                         (5 bytes)
10:47:20.757 DRIVER » [REQ] [GetSerialApiCapabilities]
10:47:20.765 SERIAL « [ACK]                                                                   (0x06)
10:47:20.768 SERIAL « 0x012b01070420000000010001fe81ff88cf1f0000fb9f7da067008080008086000 (45 bytes)
                      000e87300000e000060000003
10:47:20.771 SERIAL » [ACK]                                                                   (0x06)
10:47:20.774 DRIVER « [RES] [GetSerialApiCapabilities]
                        payload: 0x0420000000010001fe81ff88cf1f0000fb9f7da067008080008086000000e8730
                      0000e0000600000
10:47:20.784 CNTRLR   received API capabilities:
[... 237 lines cut ...]
10:47:24.535 CNTRLR   [Node 002] [~] [Manufacturer Specific] productId: 1129 => 1129    [Endpoint 0]
10:47:24.537 DRIVER « [Node 002] [REQ] [ApplicationCommand]
                      └─[SecurityCCCommandEncapsulation]
                        │ sequenced: false
                        └─[ManufacturerSpecificCCReport]
                            manufacturer id: 0x3b
                            product type:    0x01
                            product id:      0x0469
10:47:25.029 SERIAL » 0x010900130202984025041c                                            (11 bytes)
10:47:25.031 DRIVER » [Node 002] [REQ] [SendData]
                      │ transmit options: 0x25
                      │ callback id:      4
                      └─[SecurityCCNonceGet]
10:47:25.036 SERIAL « [ACK]                                                                   (0x06)
[... cut ...]
This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). The promise rejected with the reason:
ZWaveError: No command class defined for CommandClass!
    at getCommandClass (/home/ksmithson/node-zwave-js/packages/zwave-js/src/lib/commandclass/CommandClass.ts:1033:9)
    at new CommandClass (/home/ksmithson/node-zwave-js/packages/zwave-js/src/lib/commandclass/CommandClass.ts:92:15)
    at Driver.<anonymous> (/home/ksmithson/node-zwave-js/test.js:15:24)
10:47:26.489 DRIVER   destroying driver instance...
```

AlCalzone, 2020-12-23:
Oops, I mixed something up. Try

```ts
require("reflect-metadata");
import { CommandClasses } from "@zwave-js/core";
import { Driver } from "zwave-js";
import { Message } from "zwave-js/build/lib/message/Message";

const driver = new Driver("/path/to/serial", {
	networkKey: Buffer.from("your-network-key", "hex"),
})
	.on("error", console.error)
	.once("driver ready", async () => {
		// wait a few seconds
		await require("alcalzone-shared/async").wait(5000);

		// Query the current status
		const antitheftGet = new Message(driver, {
			nodeId: 2,
			payload: Buffer.from([
				CommandClasses["Anti-theft"],
				0x02, // GET
			]),
		});
		await driver.sendMessage(antitheftGet, {
			supportCheck: false,
		});

		// wait a few seconds
		await require("alcalzone-shared/async").wait(5000);
	});
void driver.start();
```

smitt04, 2020-12-23:
Latest logs

```
15:20:06.708 DRIVER   ███████╗ ██╗    ██╗  █████╗  ██╗   ██╗ ███████╗             ██╗ ███████╗
                      ╚══███╔╝ ██║    ██║ ██╔══██╗ ██║   ██║ ██╔════╝             ██║ ██╔════╝
                        ███╔╝  ██║ █╗ ██║ ███████║ ██║   ██║ █████╗   █████╗      ██║ ███████╗
                       ███╔╝   ██║███╗██║ ██╔══██║ ╚██╗ ██╔╝ ██╔══╝   ╚════╝ ██   ██║ ╚════██║
                      ███████╗ ╚███╔███╔╝ ██║  ██║  ╚████╔╝  ███████╗        ╚█████╔╝ ███████║
                      ╚══════╝  ╚══╝╚══╝  ╚═╝  ╚═╝   ╚═══╝   ╚══════╝         ╚════╝  ╚══════╝
15:20:06.723 DRIVER   version 5.6.1
15:20:06.724 DRIVER
15:20:06.725 DRIVER   starting driver...
15:20:06.766 DRIVER   opening serial port /dev/ttyUSB0
15:20:06.802 DRIVER   serial port opened
15:20:06.807 SERIAL » [NAK]                                                                   (0x15)
15:20:08.319 DRIVER   loading configuration...
15:20:09.366 DRIVER   beginning interview...
15:20:09.369 DRIVER   added request handler for AddNodeToNetwork (0x4a)...
                      1 registered
15:20:09.370 DRIVER   added request handler for RemoveNodeFromNetwork (0x4b)...
                      1 registered
15:20:09.371 DRIVER   added request handler for ReplaceFailedNode (0x63)...
                      1 registered
15:20:09.376 CNTRLR   beginning interview...
15:20:09.377 CNTRLR   querying version info...
15:20:09.861 SERIAL » 0x01030015e9                                                         (5 bytes)
15:20:09.868 DRIVER » [REQ] [GetControllerVersion]
15:20:09.884 SERIAL « [ACK]                                                                   (0x06)
15:20:09.895 SERIAL « 0x011001155a2d5761766520342e3035000197                              (18 bytes)
15:20:09.899 SERIAL » [ACK]                                                                   (0x06)
15:20:09.906 DRIVER « [RES] [GetControllerVersion]
                        payload: 0x5a2d5761766520342e30350001
15:20:09.934 CNTRLR   received version info:
                        controller type: Static Controller
                        library version: Z-Wave 4.05
15:20:09.936 CNTRLR   querying controller IDs...
15:20:09.950 SERIAL » 0x01030020dc                                                         (5 bytes)
15:20:09.952 DRIVER » [REQ] [GetControllerId]
15:20:09.958 SERIAL « [ACK]                                                                   (0x06)
15:20:09.963 SERIAL « 0x01080120d4998752014f                                              (10 bytes)
15:20:09.966 SERIAL » [ACK]                                                                   (0x06)
15:20:09.976 DRIVER « [RES] [GetControllerId]
                        payload: 0xd499875201
15:20:09.988 CNTRLR   received controller IDs:
                        home ID:     0xd4998752
                        own node ID: 1
15:20:09.990 CNTRLR   querying controller capabilities...
15:20:10.004 SERIAL » 0x01030005f9                                                         (5 bytes)
15:20:10.006 DRIVER » [REQ] [GetControllerCapabilities]
15:20:10.012 SERIAL « [ACK]                                                                   (0x06)
15:20:10.017 SERIAL « 0x0104010508f7                                                       (6 bytes)
15:20:10.019 SERIAL » [ACK]                                                                   (0x06)
15:20:10.022 DRIVER « [RES] [GetControllerCapabilities]
                        payload: 0x08
15:20:10.034 CNTRLR   received controller capabilities:
                        controller role:     primary
                        is in other network: false
                        is SIS present:      false
                        was real primary:    true
                        is a SUC:            false
15:20:10.037 CNTRLR   querying API capabilities...
15:20:10.050 SERIAL » 0x01030007fb                                                         (5 bytes)
15:20:10.052 DRIVER » [REQ] [GetSerialApiCapabilities]
15:20:10.061 SERIAL « [ACK]                                                                   (0x06)
15:20:10.065 SERIAL « 0x012b01070420000000010001fe81ff88cf1f0000fb9f7da067008080008086000 (45 bytes)
                      000e87300000e000060000003
15:20:10.069 SERIAL » [ACK]                                                                   (0x06)
15:20:10.073 DRIVER « [RES] [GetSerialApiCapabilities]
                        payload: 0x0420000000010001fe81ff88cf1f0000fb9f7da067008080008086000000e8730
                      0000e0000600000
[... 258 lines cut ...]
15:20:15.806 SERIAL » [ACK]                                                                   (0x06)
15:20:15.809 DRIVER « [Node 004] [REQ] [ApplicationCommand]
                      └─[VersionCCReport]
                          library type:      Routing Slave
                          protocol version:  3.67
                          firmware versions: 3.37
15:20:15.883 SERIAL » 0x010900130202984025031b                                            (11 bytes)
15:20:15.885 DRIVER » [Node 002] [REQ] [SendData]
                      │ transmit options: 0x25
                      │ callback id:      3
                      └─[SecurityCCNonceGet]
15:20:15.887 CNTRLR   [Node 004] The node is alive.
15:20:15.889 CNTRLR « [Node 004] received response for node versions:
                        library type:      Routing Slave (0x06)
                        protocol version:  3.67
                        firmware versions: 3.37
[... cut ...]
15:20:15.891 CNTRLR » [Node 004]   querying the CC version for Multilevel Switch...
15:20:15.899 SERIAL « [ACK]                                                                   (0x06)
15:20:15.902 SERIAL « 0x0104011301e8                                                       (6 bytes)
15:20:15.904 SERIAL » [ACK]                                                                   (0x06)
15:20:15.906 DRIVER « [RES] [SendData]
                        was sent: true
15:20:15.932 DRIVER   destroying driver instance...
```

smitt04, 2020-12-24:
I was able to get by the missing message type by adding it, but i am now stuck at function type.

```
This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). The promise rejected with the reason:
ZWaveError: A message must have a given or predefined function type
    at new Message (/home/ksmithson/node-zwave-js/packages/zwave-js/src/lib/message/Message.ts:114:11)
    at Driver.<anonymous> (/home/ksmithson/node-zwave-js/test.js:16:24)
```

AlCalzone, 2020-12-24:
Uhh sorry, should have gone to bed earlier. That should have been a SendDataRequest not a Message. Only on mobile for the next few days, but you might be able to figure the syntax out by looking at test/run.ts

smitt04, 2020-12-24:
I actually couldn't find an example of `SendDataRequest` but I tried something like

```
const cc = new CommandClass(driver, {
			nodeId: 2,

			ccCommand: CommandClasses["Anti-theft"],
			payload: Buffer.from([
				CommandClasses["Anti-theft"],
				0x02, // GET
			]),
		});
		const req = new SendDataRequest(driver, {
			command: cc,
		});
```
but it looks like there isn't a way to send unsupported commands?

AlCalzone, 2020-12-24:
I'll try to write you something tomorrow, it should absolutely be possible.

smitt04, 2020-12-27:
@AlCalzone Just wondering if you have had time to write something up? I know it's the holidays so no rush

AlCalzone, 2020-12-28:
Hey, just got around to it. Turns out it is not as easy as I thought. Try this please:
```ts
const msg = new Message(driver, {
	functionType: FunctionType.SendData,
	type: MessageType.Request,
	payload: Buffer.from("16025d0225fe", "hex"),
	//                    ^^ your node ID as hex
});
await driver.sendMessage(msg, {
	priority: MessagePriority.Normal,
});
```

You'll get a warning and and (hopefully) an unsolicited response, but we can ignore that for now.

smitt04, 2020-12-28:
Here is the latest logs
```
17:12:44.042 DRIVER   ███████╗ ██╗    ██╗  █████╗  ██╗   ██╗ ███████╗             ██╗ ███████╗
                      ╚══███╔╝ ██║    ██║ ██╔══██╗ ██║   ██║ ██╔════╝             ██║ ██╔════╝
                        ███╔╝  ██║ █╗ ██║ ███████║ ██║   ██║ █████╗   █████╗      ██║ ███████╗
                       ███╔╝   ██║███╗██║ ██╔══██║ ╚██╗ ██╔╝ ██╔══╝   ╚════╝ ██   ██║ ╚════██║
                      ███████╗ ╚███╔███╔╝ ██║  ██║  ╚████╔╝  ███████╗        ╚█████╔╝ ███████║
                      ╚══════╝  ╚══╝╚══╝  ╚═╝  ╚═╝   ╚═══╝   ╚══════╝         ╚════╝  ╚══════╝
17:12:44.069 DRIVER   version 5.6.1
17:12:44.071 DRIVER
17:12:44.073 DRIVER   starting driver...
17:12:44.136 DRIVER   opening serial port /dev/ttyUSB0
17:12:44.215 DRIVER   serial port opened
17:12:44.223 SERIAL » [NAK]                                                                   (0x15)
17:12:45.738 DRIVER   loading configuration...
17:12:47.119 DRIVER   beginning interview...
17:12:47.123 DRIVER   added request handler for AddNodeToNetwork (0x4a)...
                      1 registered
17:12:47.125 DRIVER   added request handler for RemoveNodeFromNetwork (0x4b)...
                      1 registered
17:12:47.127 DRIVER   added request handler for ReplaceFailedNode (0x63)...
                      1 registered
17:12:47.134 CNTRLR   beginning interview...
17:12:47.136 CNTRLR   querying version info...
17:12:47.824 SERIAL » 0x01030015e9                                                         (5 bytes)
17:12:47.830 DRIVER » [REQ] [GetControllerVersion]
17:12:47.845 SERIAL « [ACK]                                                                   (0x06)
17:12:47.857 SERIAL « 0x011001155a2d5761766520342e3035000197                              (18 bytes)
17:12:47.861 SERIAL » [ACK]                                                                   (0x06)
17:12:47.867 DRIVER « [RES] [GetControllerVersion]
                        payload: 0x5a2d5761766520342e30350001
17:12:47.893 CNTRLR   received version info:
                        controller type: Static Controller
                        library version: Z-Wave 4.05
17:12:47.894 CNTRLR   querying controller IDs...
17:12:47.912 SERIAL » 0x01030020dc                                                         (5 bytes)
17:12:47.914 DRIVER » [REQ] [GetControllerId]
17:12:47.921 SERIAL « [ACK]                                                                   (0x06)
17:12:47.926 SERIAL « 0x01080120d4998752014f                                              (10 bytes)
17:12:47.930 SERIAL » [ACK]                                                                   (0x06)
17:12:47.940 DRIVER « [RES] [GetControllerId]
                        payload: 0xd499875201
17:12:47.956 CNTRLR   received controller IDs:
                        home ID:     0xd4998752
                        own node ID: 1
17:12:47.958 CNTRLR   querying controller capabilities...
17:12:47.977 SERIAL » 0x01030005f9                                                         (5 bytes)
17:12:47.979 DRIVER » [REQ] [GetControllerCapabilities]
17:12:47.985 SERIAL « [ACK]                                                                   (0x06)
17:12:47.990 SERIAL « 0x0104010508f7                                                       (6 bytes)
17:12:47.993 SERIAL » [ACK]                                                                   (0x06)
17:12:47.996 DRIVER « [RES] [GetControllerCapabilities]
                        payload: 0x08
17:12:48.011 CNTRLR   received controller capabilities:
                        controller role:     primary
                        is in other network: false
                        is SIS present:      false
                        was real primary:    true
                        is a SUC:            false
17:12:48.012 CNTRLR   querying API capabilities...
17:12:48.030 SERIAL » 0x01030007fb                                                         (5 bytes)
17:12:48.032 DRIVER » [REQ] [GetSerialApiCapabilities]
17:12:48.041 SERIAL « [ACK]                                                                   (0x06)
17:12:48.046 SERIAL « 0x012b01070420000000010001fe81ff88cf1f0000fb9f7da067008080008086000 (45 bytes)
                      000e87300000e000060000003
17:12:48.050 SERIAL » [ACK]                                                                   (0x06)
17:12:48.053 DRIVER « [RES] [GetSerialApiCapabilities]
                        payload: 0x0420000000010001fe81ff88cf1f0000fb9f7da067008080008086000000e8730
                      0000e0000600000
17:12:48.068 CNTRLR   received API capabilities:
[... 1158 lines cut ...]
17:15:02.664 DRIVER « [RES] [SendData]
                        was sent: true
17:15:02.676 SERIAL « 0x010500131100f8                                                     (7 bytes)
17:15:02.678 SERIAL » [ACK]                                                                   (0x06)
17:15:02.680 DRIVER « [REQ] [SendData]
                        callback id:     17
                        transmit status: OK
17:15:04.297 CNTRLR   [Node 002] did not respond after 2/3 attempts. Scheduling next try in 500 ms.
17:15:04.813 SERIAL » 0x01090013020263042511b6                                            (11 bytes)
17:15:04.816 DRIVER » [Node 002] [REQ] [SendData]
                      │ transmit options: 0x25
                      │ callback id:      17
                      └─[UserCodeCCUsersNumberGet]
[... cut ...]
17:15:04.835 SERIAL » [ACK]                                                                   (0x06)
17:15:04.840 DRIVER « [RES] [SendData]
                        was sent: true
17:15:06.101 SERIAL « 0x010500131100f8                                                     (7 bytes)
17:15:06.103 SERIAL » [ACK]                                                                   (0x06)
17:15:06.106 DRIVER « [REQ] [SendData]
                        callback id:     17
                        transmit status: OK
17:15:07.723 CNTRLR   [Node 002] Interview attempt 2/5 failed, retrying in 10000 ms...
```

AlCalzone, 2020-12-28:
You didn't replace the hex 16 with your node id didn't you?

smitt04, 2020-12-28:
My bad, the node id is now 2, so here is the updated logs

```
21:52:42.385 DRIVER   ███████╗ ██╗    ██╗  █████╗  ██╗   ██╗ ███████╗             ██╗ ███████╗
                      ╚══███╔╝ ██║    ██║ ██╔══██╗ ██║   ██║ ██╔════╝             ██║ ██╔════╝
                        ███╔╝  ██║ █╗ ██║ ███████║ ██║   ██║ █████╗   █████╗      ██║ ███████╗
                       ███╔╝   ██║███╗██║ ██╔══██║ ╚██╗ ██╔╝ ██╔══╝   ╚════╝ ██   ██║ ╚════██║
                      ███████╗ ╚███╔███╔╝ ██║  ██║  ╚████╔╝  ███████╗        ╚█████╔╝ ███████║
                      ╚══════╝  ╚══╝╚══╝  ╚═╝  ╚═╝   ╚═══╝   ╚══════╝         ╚════╝  ╚══════╝
21:52:42.401 DRIVER   version 5.6.1
21:52:42.403 DRIVER
21:52:42.404 DRIVER   starting driver...
21:52:42.451 DRIVER   opening serial port /dev/ttyUSB0
21:52:42.489 DRIVER   serial port opened
21:52:42.494 SERIAL » [NAK]                                                                   (0x15)
21:52:44.007 DRIVER   loading configuration...
21:52:45.216 DRIVER   beginning interview...
21:52:45.219 DRIVER   added request handler for AddNodeToNetwork (0x4a)...
                      1 registered
21:52:45.221 DRIVER   added request handler for RemoveNodeFromNetwork (0x4b)...
                      1 registered
21:52:45.222 DRIVER   added request handler for ReplaceFailedNode (0x63)...
                      1 registered
21:52:45.227 CNTRLR   beginning interview...
21:52:45.228 CNTRLR   querying version info...
21:52:45.779 SERIAL » 0x01030015e9                                                         (5 bytes)
21:52:45.784 DRIVER » [REQ] [GetControllerVersion]
21:52:45.796 SERIAL « [ACK]                                                                   (0x06)
21:52:45.805 SERIAL « 0x011001155a2d5761766520342e3035000197                              (18 bytes)
21:52:45.808 SERIAL » [ACK]                                                                   (0x06)
21:52:45.813 DRIVER « [RES] [GetControllerVersion]
                        payload: 0x5a2d5761766520342e30350001
21:52:45.833 CNTRLR   received version info:
                        controller type: Static Controller
                        library version: Z-Wave 4.05
21:52:45.834 CNTRLR   querying controller IDs...
21:52:45.849 SERIAL » 0x01030020dc                                                         (5 bytes)
21:52:45.850 DRIVER » [REQ] [GetControllerId]
21:52:45.855 SERIAL « [ACK]                                                                   (0x06)
21:52:45.860 SERIAL « 0x01080120d4998752014f                                              (10 bytes)
21:52:45.862 SERIAL » [ACK]                                                                   (0x06)
21:52:45.874 DRIVER « [RES] [GetControllerId]
                        payload: 0xd499875201
21:52:45.887 CNTRLR   received controller IDs:
                        home ID:     0xd4998752
                        own node ID: 1
21:52:45.888 CNTRLR   querying controller capabilities...
21:52:45.902 SERIAL » 0x01030005f9                                                         (5 bytes)
21:52:45.904 DRIVER » [REQ] [GetControllerCapabilities]
21:52:45.909 SERIAL « [ACK]                                                                   (0x06)
21:52:45.913 SERIAL « 0x0104010508f7                                                       (6 bytes)
21:52:45.915 SERIAL » [ACK]                                                                   (0x06)
21:52:45.918 DRIVER « [RES] [GetControllerCapabilities]
                        payload: 0x08
21:52:45.930 CNTRLR   received controller capabilities:
                        controller role:     primary
                        is in other network: false
                        is SIS present:      false
                        was real primary:    true
                        is a SUC:            false
21:52:45.931 CNTRLR   querying API capabilities...
21:52:45.945 SERIAL » 0x01030007fb                                                         (5 bytes)
21:52:45.946 DRIVER » [REQ] [GetSerialApiCapabilities]
21:52:45.954 SERIAL « [ACK]                                                                   (0x06)
21:52:45.959 SERIAL « 0x012b01070420000000010001fe81ff88cf1f0000fb9f7da067008080008086000 (45 bytes)
                      000e87300000e000060000003
21:52:45.962 SERIAL » [ACK]                                                                   (0x06)
21:52:45.965 DRIVER « [RES] [GetSerialApiCapabilities]
                        payload: 0x0420000000010001fe81ff88cf1f0000fb9f7da067008080008086000000e8730
                      0000e0000600000
[... 1192 lines cut ...]
21:54:16.665 DRIVER « [RES] [SendData]
                        was sent: true
21:54:17.916 SERIAL « 0x010500131500fc                                                     (7 bytes)
21:54:17.918 SERIAL » [ACK]                                                                   (0x06)
21:54:17.921 DRIVER « [REQ] [SendData]
                        callback id:     21
                        transmit status: OK
21:54:19.542 CNTRLR   [Node 002] did not respond after 2/3 attempts. Scheduling next try in 500 ms.
21:54:20.090 SERIAL » 0x01090013020263042515b2                                            (11 bytes)
21:54:20.092 DRIVER » [Node 002] [REQ] [SendData]
                      │ transmit options: 0x25
                      │ callback id:      21
                      └─[UserCodeCCUsersNumberGet]
21:54:20.099 SERIAL « [ACK]                                                                   (0x06)
[... cut ...]
21:54:20.109 SERIAL » [ACK]                                                                   (0x06)
21:54:20.113 DRIVER « [RES] [SendData]
                        was sent: true
21:54:20.122 SERIAL « 0x010500131500fc                                                     (7 bytes)
21:54:20.124 SERIAL » [ACK]                                                                   (0x06)
21:54:20.126 DRIVER « [REQ] [SendData]
                        callback id:     21
                        transmit status: OK
21:54:21.741 CNTRLR   [Node 002] Interview attempt 3/5 failed, retrying in 15000 ms...
```

AlCalzone, 2020-12-28:
Hmm, doesn't look like that worked since we now got overlapping transactions :(
I'll try to implement #1253 as soon as possible, then we can try again.

AlCalzone, 2021-01-10:
Sorry that I didn't reply earlier. #1253 was merged, so you should be able to do this and have the driver handle the message as usual without the problems we faced above:
```
const cc = new CommandClass(driver, {
    nodeId: 2,
    ccId: CommandClasses["Anti-theft"], // or 0x5d
    ccCommand: 0x02,
});
await driver.sendCommand(cc);
```

AlCalzone, 2021-02-03:
Hey @smitt04 did you get around to test this yet?

closed 2021-02-14 by zwave-js-assistant[bot]