zwave-js/zwave-js #1221: [feat] Add missing Neo Coolcam devices

jghaanstra, 2020-12-18:
**Is your feature request related to a problem? Please describe.**
I'd like to see some missing Neo Coolcam devices to the database.

* NEO Coolcam plug V2 NAS-WR02ZE with ID 600-4135-512 (0x0258-0x1027-0x0200)
* NEO Coolcam plug of unknown type with ID 1050-8-512 (0x041a-0x0008-0x0200)

robertsLando, 2020-12-18:
@AlCalzone @billiaz

robertsLando, 2020-12-18:
@jghaanstra Could you paste here the devices manual?

jghaanstra, 2020-12-18:
I dont think there is a manual specifically for the V2 but it's similar to the V1 and the unidentified model. See the link below.

https://fccid.io/Z52NAS-WR01Z/User-Manual/User-manual-2948691

AlCalzone, 2020-12-20:
@robertsLando what about your PR?

robertsLando, 2020-12-21:
@AlCalzone I closed my pr because the result had no sense, I used our import method but seems it didn't imported it correctly, check it out in the pr to understand

AlCalzone, 2020-12-21:
The first one is here: https://github.com/OpenZWave/open-zwave/blob/b0afd4c60b38065fdbc57835cb3eb6035afcc00f/config/manufacturer_specific.xml#L1584
The second one doesn't exist in OZW either - so we'll need a new config file (or add it to the existing ones if applicable).

AlCalzone, 2021-02-03:
@blhoward2 any chance those are on your TODO list?

blhoward2, 2021-02-04:
There are some neo coolcams but it is a large number of files so finding a specific id within they is difficult. I didn't see these, but they could be there.

ookonbekend, 2021-02-16:
Hello,

I use 4 NEO Coolcam Power Plugs V2 
With Device ID: **600-4135-512 (0x0258-0x1027-0x0200)**
and can confirm the following link contains the right information for this model (at least much better).
https://github.com/mruiter/com.neo/files/5674626/NEO.Coolcam.NAS-WR02ZE.v3.pdf

I think the following file should be renamed to **wr01z.json**
https://github.com/zwave-js/node-zwave-js/blob/master/packages/config/config/devices/0x0258/nas-wr01ze.json
and adjusted like this one
https://github.com/OpenZWave/open-zwave/blob/master/config/shenzen_neo/nas-wr01z.xml

And the current one **wr01ze.json** should be like
https://github.com/OpenZWave/open-zwave/blob/master/config/shenzen_neo/nas-wr01ze.xml

When I check the values (see below) from the log after a Re-interview, you can see these are not possible with the current JSON. 

I can toggle the light by changing the value **16-112-0-3** to 0 or1 

INFO ZWAVE: Node 16: value added 16-112-0-1 => 1
INFO ZWAVE: Node 16: value added 16-112-0-2 => 1
INFO ZWAVE: Node 16: value added 16-112-0-3 => 1
INFO ZWAVE: Node 16: value added 16-112-0-4 => 0
INFO ZWAVE: Node 16: value added 16-112-0-5 => 120
INFO ZWAVE: Node 16: value added 16-112-0-6 => 1
INFO ZWAVE: Node 16: value added 16-112-0-7 => 300
INFO ZWAVE: Node 16: value added 16-112-0-8 => 16
INFO ZWAVE: Node 16: value added 16-112-0-9 => 50
INFO ZWAVE: Node 16: value added 16-112-0-10 => 13

Hopefully this information helps to solve this issue!
Many thanks for your support!

porzino, 2021-03-06:
Hi.

The new versions have different parameters order.
I have add a comment about this problem and the current manual on this topic: https://github.com/zwave-js/node-zwave-js/issues/1997.

AlCalzone, 2021-03-09:
Information from another issue:
[image]
device ID: 0x0258:0x0200:0x1027
firmware: 2.23

AlCalzone, 2021-05-24:
Is this still an issue?

jghaanstra, 2021-05-25:
Fixed

closed 2021-05-25 by jghaanstra

ookonbekend, 2021-05-25:
> Is this still an issue?

Issue solved, thx!