zwave-js/zwave-js #1223: [bug] Driver logs are not pushed on main STDOUT/STDERR

billiaz, 2020-12-18:
We face an issue with the zwave-js Driver logs. When running the command in our Shell, we can see all logs from zwavejs2mqtt and zwave-js driver in console.
Once we run a Docker container *OR* push stdout/stderr logs to a file, Driver logs missing

Testing this:
Console output:
```
❯ node bin/www
 ______                       _     ___                  _   _
|___  /                      (_)   |__ \                | | | |
   / /_      ____ ___   _____ _ ___   ) |_ __ ___   __ _| |_| |_
  / /\ \ /\ / / _` \ \ / / _ \ / __| / /| '_ ` _ \ / _` | __| __|
 / /__\ V  V / (_| |\ V /  __/ \__ \/ /_| | | | | | (_| | |_| |_
/_____|\_/\_/ \__,_| \_/ \___| |___/____|_| |_| |_|\__, |\__|\__|
                            _/ |                      | |
                           |__/                       |_|
2020-12-17 19:49:14.791 WARN STORE: scenes.json not found
2020-12-17 19:49:14.793 WARN STORE: nodes.json not found
2020-12-17 19:49:15.172 INFO APP: Version: 1.0.0-alpha.0.023fccf
2020-12-17 19:49:15.173 INFO APP: Application path:/home/billias/ongit/zwavejs2mqtt
2020-12-17 19:49:15.218 INFO MQTT: Connecting to mqtt://mqtt.data:1883
2020-12-17 19:49:15.255 INFO ZWAVE: Connecting to /dev/ttyACM0
19:49:15.257 DRIVER   ███████╗ ██╗    ██╗  █████╗  ██╗   ██╗ ███████╗             ██╗ ███████╗
                      ╚══███╔╝ ██║    ██║ ██╔══██╗ ██║   ██║ ██╔════╝             ██║ ██╔════╝
                        ███╔╝  ██║ █╗ ██║ ███████║ ██║   ██║ █████╗   █████╗      ██║ ███████╗
                       ███╔╝   ██║███╗██║ ██╔══██║ ╚██╗ ██╔╝ ██╔══╝   ╚════╝ ██   ██║ ╚════██║
                      ███████╗ ╚███╔███╔╝ ██║  ██║  ╚████╔╝  ███████╗        ╚█████╔╝ ███████║
                      ╚══════╝  ╚══╝╚══╝  ╚═╝  ╚═╝   ╚═══╝   ╚══════╝         ╚════╝  ╚══════╝
19:49:15.258 DRIVER   version 5.5.0
19:49:15.258 DRIVER
19:49:15.258 DRIVER   starting driver...
19:49:15.267 DRIVER   Send queue: (empty)                                               (0 messages)
19:49:15.267 DRIVER   send thread state: idle
19:49:15.268 DRIVER   opening serial port /dev/ttyACM0
2020-12-17 19:49:15.273 INFO APP: Listening on port 8091
2020-12-17 19:49:15.274 INFO MQTT: getaddrinfo ENOTFOUND mqtt.data mqtt.data:1883
2020-12-17 19:49:15.276 INFO MQTT: MQTT client offline
2020-12-17 19:49:15.277 INFO MQTT: MQTT client closed
19:49:15.277 DRIVER   serial port errored: Error: No such file or directory, cannot open /dev/ttyACM
                      0
19:49:15.278 DRIVER   destroying driver instance...
19:49:15.279 DRIVER   destroying driver instance...
2020-12-17 19:49:15.279 INFO ZWAVE: Driver: Error: No such file or directory, cannot open /dev/ttyACM0
2020-12-17 19:49:15.280 WARN ZWAVE: Retry connection in 3 seconds...
^C2020-12-17 19:49:16.041 INFO APP: Closing clients...
2020-12-17 19:49:16.042 INFO GATEWAY: Closing Gateway...
19:49:16.043 DRIVER   destroying driver instance...
```

Pushing STDOUT to a file (like docker does)
```
❯ node bin/www 1>a 2>b
^C%                                                                                                                                                                   

 ❯ cat a
 ______                       _     ___                  _   _
|___  /                      (_)   |__ \                | | | |
   / /_      ____ ___   _____ _ ___   ) |_ __ ___   __ _| |_| |_
  / /\ \ /\ / / _` \ \ / / _ \ / __| / /| '_ ` _ \ / _` | __| __|
 / /__\ V  V / (_| |\ V /  __/ \__ \/ /_| | | | | | (_| | |_| |_
/_____|\_/\_/ \__,_| \_/ \___| |___/____|_| |_| |_|\__, |\__|\__|
                            _/ |                      | |
                           |__/                       |_|
2020-12-17 19:48:37.324 WARN STORE: scenes.json not found
2020-12-17 19:48:37.325 WARN STORE: nodes.json not found
2020-12-17 19:48:37.694 INFO APP: Version: 1.0.0-alpha.0.023fccf
2020-12-17 19:48:37.695 INFO APP: Application path:/home/billias/ongit/zwavejs2mqtt
2020-12-17 19:48:37.735 INFO MQTT: Connecting to mqtt://mqtt.data:1883
2020-12-17 19:48:37.769 INFO ZWAVE: Connecting to /dev/ttyACM0
2020-12-17 19:48:37.780 INFO APP: Listening on port 8091
2020-12-17 19:48:37.781 INFO MQTT: getaddrinfo ENOTFOUND mqtt.data mqtt.data:1883
2020-12-17 19:48:37.783 INFO MQTT: MQTT client offline
2020-12-17 19:48:37.783 INFO MQTT: MQTT client closed
2020-12-17 19:48:37.784 INFO ZWAVE: Driver: Error: No such file or directory, cannot open /dev/ttyACM0
2020-12-17 19:48:37.784 WARN ZWAVE: Retry connection in 3 seconds...
2020-12-17 19:48:38.392 INFO APP: Closing clients...
2020-12-17 19:48:38.394 INFO GATEWAY: Closing Gateway...
```

```
 ❯ cat b
```
b file is STDERR and is empty

AlCalzone, 2020-12-18:
Could you please try one of the following and see if it changes things?

1) Open `packages\zwave-js\src\lib\log\Driver.ts`  
2a) in lines 33, 34, 39 replace `"driver"` with a different key  
2b) in line 27, change the loglevel from `"verbose"` to `"debug"` or `"info"`.  
3) Compile and link the packages, as described [here](https://zwave-js.github.io/node-zwave-js/#/development/installing-from-github?id=installing-from-github)

billiaz, 2020-12-18:
@AlCalzone 
the change is not resolving the problem:
Change:
```
 import type { Message, ResponseRole } from "../message/Message";

 export const DRIVER_LABEL = "DRIVER";
-const DRIVER_LOGLEVEL = "verbose";
+const DRIVER_LOGLEVEL = "debug";
 const SENDQUEUE_LOGLEVEL = "debug";

 let _logger: ZWaveLogger | undefined;
 function getLogger(): ZWaveLogger {
 	if (!_logger) {
-		if (!winston.loggers.has("driver")) {
-			winston.loggers.add("driver", {
+		if (!winston.loggers.has("river")) {
+			winston.loggers.add("river", {
 				transports: createLogTransports(),
 				format: createLoggerFormat(DRIVER_LABEL),
 			});
 		}
-		_logger = (winston.loggers.get("driver") as unknown) as ZWaveLogger;
+		_logger = (winston.loggers.get("river") as unknown) as ZWaveLogger;
 	}
 	return _logger;
 }
 ```
 Logs still the same as before

AlCalzone, 2020-12-18:
@robertsLando I can't reproduce this without zjs2mqtt. Running zwave-js alone locally produces the expected behavior:
* env variable LOGTOFILE set -> all logging completely goes to file
* env variable not set, no redirects -> all logging completely goes to console
* env variable not set, redirect stdout to file -> no logging, transports silenced because of no TTY

This seems to have to do with the way the dashboard intercepts the zwave-js logs.

AlCalzone, 2020-12-18:
What if we add the ability to pass a custom winston transport to zwave-js, which just gets the raw, uncolored logs from zwave-js?

robertsLando, 2020-12-18:
> What if we add the ability to pass a custom winston transport to zwave-js, which just gets the raw, uncolored logs from zwave-js? That way you could make sure to put the logs where they belong.

This is already possible with the changes on 6.0.0 branch

robertsLando, 2020-12-18:
> This seems to have to do with the way the dashboard intercepts the zwave-js logs.

You mean this? https://github.com/zwave-js/zwavejs2mqtt/blob/master/app.js#L73-L84

AlCalzone, 2020-12-20:
@billiaz @robertsLando I'm not even sure anymore that this is a bug. Not logging to console if the stdout is not a TTY was a concious decision because the huge amount of logging requires quite a bit of performance.

When 6.0.0 hits, you will be able to define the log file name yourself and just tail that file.

AlCalzone, 2020-12-20:
If you agree, I'm going to close this.

billiaz, 2020-12-21:
Would it give us also possibility to push logs without tty? 
If thsi is easy to add as option, might worth! more people move to containers, and the load on cpu (for more power systems like rPi4) can be helpful during debugging

File can be worst than stdout, due to more load on system. Ofcourse you could assume we can redirect file to stdout. but still hacky way.

robertsLando, 2020-12-21:
I agree with @billiaz about the option

PR #1233 (2020-12-21): feat: add option to force console logging when not a TTY
This PR adds the option `forceConsole` to the logger options, which causes zwave-js to also log to the stdout if it is not connected to a TTY. Setting `logToFile` still has preference though.

fixes: #1223
PR #1167 (2020-12-23): Prepare Release v6.0.0
This draft PR is to collect the work that goes into version 6.

fixes #1062
fixes: #1135
fixes: #1223
closed 2020-12-23 by AlCalzone