zwave-js/zwave-js #1226: Updating devices from source 'oh' does not work due to URL changes

ahochsteger, 2020-12-19:
**Describe the bug**
The OpenHAB source for Z-Wave devices from <https://www.cd-jackson.com/index.php/zwave/zwave-device-database/> has changed to <https://opensmarthouse.org/zwavedatabase/> causing the device update task to fail.

**To Reproduce**
Steps to reproduce the behavior:
1. `yarn run config -D -s oh`

**Logfile:**
```bash
$ yarn run config -D -s oh
yarn run v1.22.10
$ ts-node maintenance/importConfig.ts -D -s oh
Fetching manufacturers...
./node_modules/axios/lib/core/createError.js:16
  var error = new Error(message);
              ^
Error: Request failed with status code 404
    at createError (./node_modules/axios/lib/core/createError.js:16:15)
    at settle (./node_modules/axios/lib/core/settle.js:17:12)
    at IncomingMessage.handleStreamEnd (./node_modules/axios/lib/adapters/http.js:244:11)
    at IncomingMessage.emit (node:events:388:22)
    at IncomingMessage.EventEmitter.emit (node:domain:470:12)
    at endReadableNT (node:internal/streams/readable:1295:12)
    at processTicksAndRejections (node:internal/process/task_queues:80:21)
error Command failed with exit code 1.
info Visit https://yarnpkg.com/en/docs/cli/run for documentation about this command.
```

**Solution Infos**
This URL returns a list of devices in JSON format:
<https://opensmarthouse.org/dmxConnect/api/zwavedatabase/device/list.php?filter=&manufacturer=-1&limit=2000>

This URL returns the details of a device in JSON format:
<https://opensmarthouse.org/dmxConnect/api/zwavedatabase/device/read.php?device_id=256>

More URLs can be observed using the network tab of the browser developer tools.
NOTE: This website aggressively caches requests in the LocalStorage, so reloading a page may not trigger any requests. Clearing the LocalStorage of the site opensparthouse.org solves this.

AlCalzone, 2020-12-20:
Accepting PRs if anyone wants to chime in.

ahochsteger, 2020-12-20:
@AlCalzone, maybe I'll give it a try.

Can you give me answers to the following questions, so I'll go into the right direction?

* Should the device config be downloaded and stored as is or transformed into a standard format used by zwave-js? I assume a standard format is used.
* Are existing device config files replaced (thus loosing previous changes) or not touched anymore (thus loosing the possibility to get updates)?
* The same questions apply to the use of multiple sources for device configs. What's the typical approach you've been using to mass-update device config files?

AlCalzone, 2020-12-20:
> I assume a standard format is used.

AlCalzone, 2020-12-20:
IMO, we should still update the import script so we can choose to import individual configs.

robertsLando, 2020-12-21:
> IMO, we should still update the import script so we can choose to import individual configs.

I can do this

AlCalzone, 2020-12-23:
@robertsLando This issue is about the OpenHAB import though ;)

robertsLando, 2020-12-23:
@AlCalzone doesn't openhub import already allow this? That's why I did it for ozw :laughing:

AlCalzone, 2020-12-23:
uhhh yup, nevermind

PR #1246 (2020-12-23): fix: update import script to handle new OpenHAB database
fixes: #1226 

/cc @ahochsteger
closed 2020-12-23 by AlCalzone