zwave-js/zwave-js #1228: [bug] Unable to add inovelli switch to zwavejs2mqtt

praveen12bnitt, 2020-12-20:
**Version** Version: 1.0.0-alpha.1

Build/Run method
- [ ] Docker

zwavejs2mqtt version: 1.0.0-alpha.1

**Describe the bug**
Unable to add inovelli switch to zwavejs2mqtt

**To Reproduce**
Set the controller to inclusion mode and add the zwave device. The device show up in the UI, but it does not show the manufacture, product or other info.

AlCalzone, 2020-12-20:
The device does not seem to behave correctly. After the secure inclusion, it tells us that it supports these CCs without encryption:
```
16:44:46.161 DRIVER « [Node 002] [REQ] [ApplicationUpdateRequest]
                        payload: 0x8402090410015e6c55989f22
16:44:46.164 CNTRLR « [Node 002] node info received
                      supported CCs:
                      · Z-Wave Plus Info
                      · Supervision
                      · Transport Service
                      · Security
                      · Security 2
                      · Application Status
                      controlled CCs:
```
and these with encryption:
```
16:44:46.335 CNTRLR « [Node 002] received secure commands
                      supported CCs:
                      · Configuration
                      · Association
                      · Association Group Information
                      · Version
                      · Binary Switch
                      · Manufacturer Specific <== !!!
                      · Device Reset Locally
                      · Central Scene
                      · Powerlevel
                      · Protection
                      · Firmware Update Meta Data
                      controlled CCs:
```
The driver then requests the device IDs using (encrypted) `Manufacturer Specific CC`, but the device does not respond. Since it usually responds within a couple of milliseconds, it is unlikely that increasing the timeout will do anything.

I suggest you try to include it non-securely.

praveen12bnitt, 2020-12-20:
Ok. Will do that shortly.

praveen12bnitt, 2020-12-20:
I tried to add it in in-secure mode and still the same behavior.
[zwave-2.log](https://github.com/zwave-js/node-zwave-js/files/5721174/zwave-2.log)

AlCalzone, 2020-12-20:
Ok it does not respond to that message at all... Did you get that thing to work with zwave2mqtt/OZW? If so, could you make a log with that? No need to re-include, just start that version.

AlCalzone, 2020-12-20:
@robertsLando is there already a possibility to change the driver timeouts through zjs2mqtt, especially the `timeouts.report` one?

praveen12bnitt, 2020-12-20:
> Ok it does not respond to that message at all... Did you get that thing to work with zwave2mqtt/OZW? If so, could you make a log with that? No need to re-include, just start that version.

Sure. Will do that shortly.

praveen12bnitt, 2020-12-20:
here you go.

[OZW_Log.txt](https://github.com/zwave-js/node-zwave-js/files/5721449/OZW_Log.txt)
Strangely, zwave2mqtt also does not show up the device correctly.  With zwave2mqtt, atleast the UI shows it as a " Binary Power Switch" and allows me to toggle the switch on and off.

[image]

praveen12bnitt, 2020-12-20:
here is what i did to get them all added to zwavejs2mqtt.

* brought up zwave2mqtt and increased the timeout to 35000 ms.
* Then added all the inovelli switches one by one. I had a total of 5. All of them got added successfully. I am not sure if its because of the timeout, but my gut feeling is it does have a impact. The reason is, when the timeout is set at 5000ms , even zwave2mqtt failed to recognize the manufacture. After changing the timeout to 35000 ms, it worked. 
* Then i shutdown zwave2mqtt and started up zwavejs2mqtt and now all of them show up fine in zwavejs2mqtt :)

[image]

AlCalzone, 2020-12-20:
Thanks I'll look at the log tomorrow. Could you please create a new log of the startup with zwavejs2mqtt, now that it works? I'd like to see what has changed.

AlCalzone, 2020-12-21:
@praveen12bnitt Your OZW log contains almost no information :(

praveen12bnitt, 2020-12-21:
Do I have to enable specific logs ?

AlCalzone, 2020-12-21:
AFAIK OZW only has the option to log or not to log. @robertsLando any hint?

robertsLando, 2020-12-21:
@praveen12bnitt Go to settings page, zwave section, check the log level and set it to verbose. I would also enable logToFile too

robertsLando, 2020-12-21:
@praveen12bnitt When logging is enabled in zwave section it will log everything

praveen12bnitt, 2020-12-21:
Ok.. I will exclude 1 switch , enable the logs , increase the timeout and then add the device back.

praveen12bnitt, 2020-12-22:
[zwave-11.log](https://github.com/zwave-js/node-zwave-js/files/5730399/zwave-11.log)
[OZW_Log.txt](https://github.com/zwave-js/node-zwave-js/files/5730403/OZW_Log.txt)

I uploaded both the files. The node that i added was 009

AlCalzone, 2020-12-22:
Huh - now it responds almost immediately... I honestly have no idea what's going on or if there's even anything to do.

praveen12bnitt, 2020-12-22:
timeout is what helped me with zwave2mqtt.  Can you expose the timeout property ? i will try to increase the timeout and give it a shot.

AlCalzone, 2020-12-22:
That's something @robertsLando needs to implement. 

But your last zwave-js log looks perfectly fine if I didn't miss anything.

robertsLando, 2020-12-22:
Yep that's something I need to implement on my side. Could you open a feat request on zjs2m?

praveen12bnitt, 2020-12-22:
> That's something @robertsLando needs to implement. 
> 
> But your last zwave-js log looks perfectly fine if I didn't miss anything.

I will try to exclude and include the device back in zjs2m and see if anything changed..

praveen12bnitt, 2020-12-22:
> Yep that's something I need to implement on my side. Could you open a feat request on zjs2m? 

Will do.

robertsLando, 2020-12-23:
@praveen12bnitt I have added the possibility to add custom zwaveOptions. Just open the `settings.json` file located in `store` directory you will see a json like: 

```json

{
    "mqtt": { <mqtt options> }, 
    "gateway": { <gw options> }, 
    "zwave": {
        "port": "/dev/ttyACM0",
        "commandsTimeout": 30,
        "networkKey": "01234567891234567891325469874561",
        "logLevel": "debug",
        "logToFile": true,
        "options": { } <------- THIS WILL EXTEND ZWAVEOPTIONS (so you can specify timeouts, attempts etc...
    }
}
```

robertsLando, 2020-12-28:
I think this can be closed

closed 2020-12-30 by AlCalzone