zwave-js/zwave-js #1231: Yale Security YRD120 does not work

praveen12bnitt, 2020-12-20:
**Describe the bug**
Yale Security YRD120 door lock does not work

**To Reproduce**
Yale Security YRD120  is already added the to zwave network with zwave2mqtt. It show fine in zwave2mqtt. Then i moved to zwavejs2mqtt it shows up as a Node, but it does not identity it as a lock and the manufacturer and product also is empty.  Cannot do any operation with it in zwavejs2mqtt

**Device information**
Which device(s) is/are affected?  ale Security YRD120
What are the node IDs? 008

praveen12bnitt, 2020-12-20:
[zwave4.log](https://github.com/zwave-js/node-zwave-js/files/5721576/zwave4.log)

AlCalzone, 2020-12-21:
The interview fails when the driver requests the stored user codes, because an invalid message is received. Could you send me your network key (e.g. on slack) so I can try to figure out why the message is deemed invalid?

AlCalzone, 2020-12-21:
The problem was caused by a malformed UserCode CC report. The specs require that these commands contain a code with at least 4 digits, even if the requiested code is not in use. Your device left out the code field, thus the message format validation failed.

PR #1235 fixes that - I think it is fair to allow an empty code when it is not in use.

PR #1235 (2020-12-21): fix(cc): accept UserCodeCCReports without user code for status available
fixes: #1231
closed 2020-12-21 by AlCalzone