zwave-js/zwave-js #1232: Greenwave disable network LED setting

jghaanstra, 2020-12-21:
**Is your feature request related to a problem? Please describe.**
Greenwave sockets have an annoying feature where the LED on the socket starts blinking after a configurable time of inactivity. The highest value you can set this to is however 255 minutes (setting 112-0-1). For sockets that are not used that much or for instance at night this blinking occurs after this time. To avoid the LED from blinking communication between the controller is needed. On zwave2mqtt there was a user setting that you could set to "Disable" for instance every 4 hours to make sure the blinking does not occur. The setting didnt do anything else, it would automatically update to enable again so you could just send disable every 4 hours and that would avoid the LED for starting to blink. The setting in zwave2mqtt was 112-1-4.

I do see a setting under config within zwavejs2mqtt called LED for network error (122-0-4) which seems to be the same setting. Setting this from the UI however does not stop the light from blinking. It looks like it's not actually being saved and I'm not seeying anything being published to MQTT either. So i'd appreciate if this could be fixed somehow so my LEDs dont start blinking at night. Below are the ID's of the Greenwave products I'm talking about.

* Single socket: 153-2-2 (0x0099-0x0002-0x0002)
* multi socket brick: 153-4-3 (0x0099-0x0004-0x0003)

AlCalzone, 2020-12-21:
I have the single socket plug and setting the config value does disable the blinking altogether.
Maybe a problem in zwavejs2mqtt @robertsLando ?

robertsLando, 2020-12-21:
@AlCalzone I think this could be a config issue. Does those devices have a `DIsable` entry on available states?

AlCalzone, 2020-12-21:
[image]

AlCalzone, 2020-12-21:
yes

robertsLando, 2020-12-21:
@jghaanstra Could you go to the control panel, select the node and from the node tab under the table download the node json ? then attach it here

jghaanstra, 2020-12-21:
Sure thing. This is one of my single socket plugs.

[node_11.zip](https://github.com/zwave-js/node-zwave-js/files/5723812/node_11.zip)

robertsLando, 2020-12-21:
Seems you have it: 

```
{
    "id": "11-112-0-4",
    "nodeId": 11,
    "commandClass": 112,
    "commandClassName": "Configuration",
    "property": 4,
    "propertyName": "LED for network error",
    "type": "number",
    "readable": true,
    "writeable": true,
    "description": "If the LED should indicate a network error by flashing or not",
    "label": "LED for network error",
    "default": 0,
    "genre": "config",
    "min": 0,
    "max": 1,
    "list": true,
    "states": [{
      "text": "Disable the LED for network error",
      "value": 0
    }, {
      "text": "Enable the LED for network error",
      "value": 1
    }]
  }
```

That's really strange, let me look at the ui