zwave-js/zwave-js #1238: Add a way to load additional config files aside from the bundled ones

AlCalzone, 2020-12-21:
e.g. for testing new device configs or devices like Z-Uno (which are basically user-programmed)

robertsLando, 2020-12-22:
I think we could extend this to: allow using an external devices db. Also IMO would be fine to add the possibility to store devices db to a location choosen by user. When zwavejs is lunched it will check if there is a valid index file there, if not it will copy its default db in that location

AlCalzone, 2020-12-22:
I'm not too keen with copying. I'd rather check if there is a valid index file in the given location and just use it.

robertsLando, 2020-12-23:
@AlCalzone The problem is when you are using docker, you need to setup volumes

AlCalzone, 2020-12-23:
Why is that a problem?

robertsLando, 2020-12-23:
Because usually you create only one volume for each container, in this way we have to create one also for devices db

robertsLando, 2020-12-23:
From our discussion:

> When not in Docker, just updating the @zwave-js/config dependency would do the trick. Applications (or zwave-js in the future) can do that automatically. Or download the tarball with npm pack @zwave-js/config, then extract the device config dir to the target folder

blhoward2, 2021-01-10:
@AlCalzone @robertsLando Anyone reason not to just put the device files under the store and update them there? Qt-ozw does this so they are more accessible and changes persist across image updates.

AlCalzone, 2021-01-10:
Mostly because how it is done has to be compatible with non-Docker environments. By including the config files in `@zwave-js/config` allows us to version them semantically and use `npm` to update the definitions.

The idea behind this issue to allow users to store modified files in a "safe" location, just like you ask for.

closed 2021-04-20 by AlCalzone