zwave-js/zwave-js #1255: Reduce writes to JSON store

robertsLando, 2020-12-29:
> is it possible to disable storing [node].values.jsonl on disk? Constant disk writes add wear to storage for embedded devices...

Ref: https://github.com/zwave-js/zwavejs2mqtt/issues/115#issuecomment-751955662

AlCalzone, 2020-12-29:
Disabling the storage makes no sense because after a restart or crash all interviews would need to be repeated completely.

I can do you one better though... The value store is based on `jsonl-db`, which has options to throttle the FS access:
https://github.com/AlCalzone/jsonl-db#controlling-file-system-access

We could expose a part of those options (or create presets): https://github.com/zwave-js/node-zwave-js/blob/fa4faff39fd49b50702f0f859a2e4206ec67c334/packages/zwave-js/src/lib/driver/Driver.ts#L720-L733

robertsLando, 2020-12-29:
> We could expose a part of those options (or create presets):

I think this would be a good addon to driver options :)

AlCalzone, 2020-12-29:
Since we're still working on 6.x, lets merge the `fs`, `cacheDir` and these new options into a new object-type property. Now's the time for housekeeping.

AlCalzone, 2020-12-29:
And I prefer presets in order to not overload the options with stuff users can mess up.

robertsLando, 2020-12-29:
> And I prefer presets in order to not overload the options with stuff users can mess up

I'm +1 for this! 

> lets merge the fs, cacheDir and these new options into a new object-type property

Yep, we could wrap all them in a `storage` property of driver options?

PR #1260 (2020-12-30): feat: expose option to throttle FS access
With this PR, we expose a new option to configure throttling of the filesystem access.
The option `storage.throttle` has the following three options:
* `"fast"` - Every change is written to disk immediately, the DB is compressed each minute (if there were 5 changes)
* `"slow"` - Up to 500 changes are buffered for 5 minutes, the DB is compressed every hour (if there were at least 100 changes)
* `"normal"` (default) - Compromise between the two, corresponds to the legacy behavior

To clean up, the `fs` option was renamed to `storage.driver` and the `cach [... cut]
closed 2020-12-30 by AlCalzone