zwave-js/zwave-js #1257: Aeotec D/W Sensor 7 - Tilt binary sensor missing

eddKv, 2020-12-30:
**Describe the bug**
Der Aeotec Door/Window Sensor 7 hat einen Sendor zur Bestimmung der Kippstellung eines Fensters. Laut Zwave Definition wird dies nicht über eine Notification berichtet, sondern über eine binäre Sensormeldung. Also wenn Neigung/Tilt erkannt, dann sendet 0x0B. Leider finde ich unter den Objects keine Binary Sensoren für diese Node.

**To Reproduce**
Keine binären Sensordaten. Der Kippstatus wird nicht angezeigt.
Interessant ist, dass es in der notifications.json ein enum existiert für so einen Zustand, aber für diesen Sensor von Aeotec scheint dies so nicht zu funktionieren. Ausschnitt:
"name": "Door state",
				"idle": false,
				"states": {
					"0x16": {
						"label": "Window/door is open",
						"params": {
							"type": "enum",
							"values": {
								"0x00": "Open in regular position",
								"0x01": "Open in tilt position"
							}
						}
					},

**Device information**
Fenstersensor ZWA008
Node_012

**Logfile:** 
[zwave-464.log](https://github.com/zwave-js/node-zwave-js/files/5754179/zwave-464.log)
[image]

AlCalzone, 2020-12-30:
Was steht hinter `accessControl_doorState`? Die Werte sind im Screenshot leider abgeschnitten...
Laut Log kommt hier 22 für offen und 23 für geschlossen rein.

Binary_Sensor sollte laut Anleitung unterstützt sein, hier sehe ich aber keine Meldungen.
Die initiale Abfrage ist auch fehlgeschlagen, aber das scheint ein Bug zu sein.

marcus-j-davies, 2020-12-30:
Die Neigungsfunktion des Aeotec Türsensors 7 entspricht nicht dem, was sie behauptet.
Der Ausgangszustand muss offen sein (über die Nähe des Magneten - oder das Fehlen eines Magneten), und der Neigungssensor prüft danach, ob eine Neigung vorliegt. - Wie lange danach bin ich mir nicht sicher, aber es ist vielleicht ein kleines Zeitfenster

Dies ist möglicherweise der Grund, warum Sie das Ereignis nicht sehen.

eddKv, 2020-12-31:
Danke für die schnelle Antwort. Hinter dem doorState steht "Window/door is open (22)", was erstmal korrekt ist (wie auch von marcus-j-davies geschrieben), aber ich verstehe nicht wie die binäre Sensormeldung für die Neigung erfolgen soll, wenn es gar keinen Binary Sensor gibt. Oder ensteht dieser Wert erst wenn so ein Signal erstmalig gesendet wurde? Ich habe über 5 Minuten gewartet, aber leider kam nichts an. Ich habe gesehen, dass bereits ein Fix erstellt wurde, ist das die von Ihnen gemeinte Verknüpfung über eine Motion?

marcus-j-davies, 2020-12-31:
Ich bin derzeit nicht qualifiziert, die Details zu zwave-js zu beantworten. Ich erfahre immer noch, dass es sich um Interna handelt.

Ich habe jedoch auch den gleichen Sensor (2 davon), und die Neigungsfunktion hat bei mir nie funktioniert - evan mit verschiedenen zwave-Bibliotheken.

Anscheinend ist die Neigungsfunktion ... Flocken.

siehe Kommentar Nr. 10 zu diesem Thread.

https://forums.homeseer.com/forum/lighting-primary-technology-plug-ins/lighting-primary-technology-discussion/z-wave-homeseer/1329572-anybody-using-aeotec-door-window-sensor- 7

eddKv, 2021-01-04:
Danke für Eure Antwort und den Fix. Ich habe nun die Binären Sensoren auch in der Liste von Objekten und es funktioniert.

AlCalzone, 2021-01-04:
Kann also geschlossen werden?

eddKv, 2021-01-04:
Ja

closed 2021-01-04 by AlCalzone