zwave-js/zwave-js #1259: Support Aeotec Recessed Door Sensor 7

larstobi, 2020-12-30:
Support Aeotec Recessed Door Sensor 7 is currently unknown to zwave-js.

After inclusion (secure) it says:
Unknown manufacturer: 881
Unknown product: 187
Product: 2

User guide:
[Aeotec Recessed Door Sensor 7 Engineering Specification.pdf](https://github.com/zwave-js/node-zwave-js/files/5754360/Aeotec.Recessed.Door.Sensor.7.Engineering.Specification.pdf)

https://aeotec.freshdesk.com/support/solutions/articles/6000226848-recessed-door-sensor-7-user-guide-

Web UI Console Debug log during inclusion:
[recessed7.log](https://github.com/zwave-js/node-zwave-js/files/5754374/recessed7.log)

The log contains some  "DRIVER Message with invalid data received. Dropping it" messages.

AlCalzone, 2020-12-30:
> The log contains some "DRIVER Message with invalid data received. Dropping it" messages.

That seem to be invalid encrypted messages. I need the full zwave-js log and your network key to debug further.

larstobi, 2020-12-30:
I removed the device and re-added it, also securely, but wasn't able to reproduce the invalid data messages.

larstobi, 2020-12-30:
Exported json of node (also, re-added so new node ID is 84):
[node_84.json](https://github.com/zwave-js/node-zwave-js/files/5754598/node_84.json.log)

PR #1268 (2021-01-04): feat(config): add Aeotec ZW187 Recessed Door Sensor 7
Add Aeotec ZW187 Recessed Door Sensor 7

Closes #1259
closed 2021-01-04 by AlCalzone