zwave-js/zwave-js #1270: Implement message queued/dequeued events

marcus-j-davies, 2021-01-01:
Happy new year to you all.

I'm not sure if this has any weight to it?
But I think 2 new controller events could provide some insurance for those with battery operated devices.

Whenever I am sending to a battery operated node or something that doesn't need to be a wake at that moment - whilst there is no reason to believe that they won't be delivered on their next wake up, I think it could add some value to some how flag that the message has been queued i.e a "**message queued**" event that they can subscribe to and subsequently a  "**message dequeued**" event.

The arguments for these events could be the id of the target node, the command class, and the time of the original message.

I think it removes the "is that going to be sent when they wake up" kind of doubt.

AlCalzone, 2021-01-01:
Just to be sure I understand the use case correctly and clarify potential confusion:

> whilst there is no reason to believe that they won't be delivered on their next wake up

This is actually not guaranteed. Depending on the queue length and the responsiveness of the device/network, it could be that a queued message takes 2 or more wakeups to be delivered.

> I think it could add some value to some how flag that the message has been queued

Calling `node.setValue`, any CC API or `sendMessage/sendCommand` **will** queue at least one message. I don't see how we can add any value here.

> it could allow front end users of the lib to list pending messages in their apps.

**That** could be interesting however. I'm leaning towards adding this request to https://github.com/zwave-js/node-zwave-js/issues/985 (specifically the point "queue size")

marcus-j-davies, 2021-01-01:
> This is actually not guaranteed. Depending on the queue length and the responsiveness of the device/network, it could be that a queued message takes 2 or more wakeups to be delivered.

Got ya!

> Calling node.setValue, any CC API or sendMessage/sendCommand will queue at least one message. I don't see how we can add any value here.

Being advised that a message has been queued **due to a sleeping node**, feels as if it could add some closure, on the users request, certainly for those who are new to the ZWave protocol, and looking to get into home automation.

> That could be interesting however. I'm leaning towards adding this request to #985 (specifically the point "queue size")

if one could somehow get a sense of what the queue contains - that should provide all the benefits, and will allow for some UI's that present what is waiting to be done on the network.

AlCalzone, 2021-01-01:
Transparency is a good thing and I would really like to give users tools to understand what is happening. 

I've added your suggestion to the other issue, so I'm going to mark this one as a
Duplicate of #985

AlCalzone, 2021-01-01:
Duplicate of #985

closed 2021-01-01 by AlCalzone

LordMike, 2021-01-10:
Bonus to having events per message, rather than just counters, is also being able to see how long it actually took to deliver that "turn on" message. Would help immensely in tracking down where delays are occurring, when lights appear slow.

AlCalzone, 2021-01-10:
Further discussion in #985 please