zwave-js/zwave-js #1277: GE Switches and Dimmers do not reflect double-up or double-down because they use the BasicCC for those events

blhoward2, 2021-01-03:
**Describe the bug**

GE dimmer/switches are capable of double tap events. In zwave2mqtt these show as events under the node, with 255 indicating double tap up and 0 indicating double tap down. (My ozw device files were set to ignore the command mapping for class 32 from qt-openzwave and I carried this over when I switched so I don't know if it was necessary.

These are the steps to make it work under qt-openzwave: https://community.home-assistant.io/t/howto-add-double-tap-support-for-ge-switches-and-dimmers-in-ozw-beta/228871/16.

The original qt-openzwave issue which has more detailed info is at: OpenZWave/qt-openzwave#60.

This appears to result from this: #388 

Non-compliant or not, it is how they work... Is there a way to IgnoreMapping here like in ozw 1.6?

**To Reproduce**
Steps to reproduce the behavior:
1. Pair switches in zwavejs2mqtt. 
2. Double tap up or down

**Device information**
Which device(s) is/are affected?  Various GE switches and dimmers (confirmed on 14294, 14291)
What are the node IDs?

**Logfile:** 

Note: I have a script that resets the light to the default levels upon turning on. That accounts for the 50 --> 50 messages.

Double-tap up (from a prior double-tap up state, hence 255-->255):

[image]

Double-tap up from off, just in case being in a former double-tap state affects it (note this one is out of order on the console log):

[image]

Double-tap down:

[image]

For comparison, off to normal on:

[image]

For comparison, on to normal off:

[image]

Console log:

[image]

AlCalzone, 2021-01-03:
Thanks for the detailed logs but I don't quite follow what is the expected behavior here. Could you break down for each event **what you expect to happen** and **what actually happens**?

blhoward2, 2021-01-03:
When you double tap up the device should send a BasicCC with a value of 255, but the load isn’t altered. When you double tap down it sends a Basic CC with a value of 0, but the load isn’t altered. When you do that it sends commands via direct association to Group 3 or you can react through an automation to the basic set. This has worked this way with SmartThings and ozw 1.6. 1.4 worked similarly but it sent a zwave event instead of a Basic CC. 

Right now, the 255 and 0 are treated as reports as it probably wasn’t proper for GE to use the BasicCC this way. Because the actual load doesn’t change not only can you not use double tap with an automation, if you do it through direct association the current value goes out of sync with the actual device. It says 255 but the device is at 50, 0, or some other level. Likewise, if you double tap down the current value goes to 0 but the device is actually still on. 

In ozw 1.6 you can define to IgnoreMapping in the device file, which prevents it being treated like a report per the standard. This is discussed in the one link I posted and fishwaldo explains that these devices needed to be treated as not following the standard for this to work. 

In ozw you can also set auto=“true” to have the controller automatically added to group 3 so it receives the events. This isn’t strictly required but makes it easier to setup for new users.

AlCalzone, 2021-01-03:
Got it, thanks! If you want to chime in, you can edit the association configuration of these devices to make group 3 a lifeline:
https://zwave-js.github.io/node-zwave-js/#/development/config-files?id=associations
(or just figure out which device IDs are affected)

I'll add a compat flag to disable mapping.

blhoward2, 2021-01-03:
Thanks! Happy to test once you’re there.

AlCalzone, 2021-01-03:
#1278 adds a compat option for this. You can add (or edit) the `compat` property in those device files to contain:
```jsonc
"compat": {
  // other options (if any)
  disableBasicMapping: true,
}
```

blhoward2, 2021-01-03:
You're fast! You don't off the top of your head know the build arg to build a zwavejs2mqtt docker image incorporating this commit for testing do you? I've never done it with a pending commit to a dependency.

AlCalzone, 2021-01-03:
let me check if I find this somewhere

AlCalzone, 2021-01-03:
https://github.com/zwave-js/zwavejs2mqtt/tree/master/docker#build
Should be the ZWJ_BRANCH arg

blhoward2, 2021-01-03:
Which device file do I need to edit to add the compat flag?

./node_modules/@zwave-js/config/config/devices/0x0063/ge_14291_zw4005.json
./node_modules/zwave-js/node_modules/@zwave-js/config/config/devices/0x0063/ge_14291_zw4005.json

I added it to one and re-interveiwed everything but that didn't seem to do it. I'm trying the other one now. It should be at the same level as the other options?

```
// Jasco Products GE 14291 (ZW4005)
// In-Wall Paddle Switch
{
    "_approved": true,
    "manufacturer": "Jasco Products",
    "manufacturerId": "0x0063",
    "label": "GE 14291 (ZW4005)",
    "description": "In-Wall Paddle Switch",
    "devices": [
        {
            "productType": "0x4952",
            "productId": "0x3036"
        },
        {
            "productType": "0x4952",
            "productId": "0x3185"
        }
    ],
    "firmwareVersion": {
        "min": "0.0",
        "max": "255.255"
    },
    "supportsZWavePlus": true,
    "compat": {
        // other options (if any)
        "disableBasicMapping": true,
    },
    "paramInformation": {
 (Continued)
```

AlCalzone, 2021-01-03:
That looks correct. New log please

blhoward2, 2021-01-03:
Is there a file that would reflect the node cache and whether the compat was recognized while setting up the devices? I had a json file in the store listing the nodes and the command classes. I wiped it thinking it would be like the ozwcache but it hasn't been recreated even after all of the devices are interviewed. I'm unsure if there is in an incompatibility between the state of the change to logging as well, but the zwave log file is no longer being populated even though it it is set to log to file. Weird.

This is what I built this morning: DOCKER_BUILDKIT=1 docker build --build-arg SRC=git-clone-src --build-arg Z2M_BRANCH=master --build-arg ZWJ_BRANCH=compat-disable-basic-mapping --no-cache -f zwavejs2mqtt/docker/Dockerfile.contrib -t zwavejs2mqtt .

I'll send you the console log file shortly.

blhoward2, 2021-01-03:
I see that you've merged it so I'm rebuilding with: 

DOCKER_BUILDKIT=1 docker build --build-arg SRC=git-clone-src --build-arg Z2M_BRANCH=master --build-arg ZWJ_BRANCH=master --no-cache -f zwavejs2mqtt/docker/Dockerfile.contrib -t zwavejs2mqtt .

Once that is done I'll manually edit the device files and give it another try.

blhoward2, 2021-01-04:
Nope. I can only access the console log as it isn't logging to file for some reason:

```
00:30:25.836 DRIVER « [Node 003] [REQ] [ApplicationCommand]

                      └─[BasicCCSet]

                          target value: 255

00:30:25.842 CNTRLR   [Node 003] treating BasicCC Set as a report

00:30:25.844 CNTRLR   [Node 003] [~] [Multilevel Switch] currentValue: 50 => 255        [Endpoint 0]

2021-01-04 00:30:25.847 INFO ZWAVE: Node 3: value updated: 38-0-currentValue 50 => 255

00:30:25.991 SERIAL « 0x01080004006702820115                                              (10 bytes)

00:30:25.994 DRIVER   Dropping message because it could not be deserialized: The command class Hail 

                      is not implemented

00:30:25.994 SERIAL » [ACK]                                                                   (0x06)
```

AlCalzone, 2021-01-04:
There are cache files (store directory in zjs2mqtt) but the log seems like it didn't apply the option.
Could you send me the cache files before wiping them?

blhoward2, 2021-01-04:
That's what is weird...I'm not seeing a cache file at all when I run this image. Nor is the log file being updated despite the Log to File option under Zwave being turned on.

[image]

AlCalzone, 2021-01-04:
@robertsLando any idea?

blhoward2, 2021-01-04:
To be clear also, no changes to settings. And going back to the dev image brought back both cache files and the log.

robertsLando, 2021-01-04:
@AlCalzone When discovering a value I read the config to get the meter type sensor type etc... that error is thrown by zwavejs so the question is if it's ok that the config file isn't loaded at that point, or should i catch that error?

blhoward2, 2021-01-04:
If you read up further, there is a bigger issue perhaps. I’m not seeing any cache files and the logger stops working if I build an image of both masters.

robertsLando, 2021-01-04:
@blhoward2 The problem is that if you are using zwavejs master it has some breaking changes (introduced with version 6.0.0) that still needs to be sorted in zwavejs2mqtt

blhoward2, 2021-01-04:
Is there anyway to see how the node has been setup? I just need to verify if the compat flag added was acted upon so I can test this change.

robertsLando, 2021-01-04:
@blhoward2 Please use this: https://github.com/zwave-js/zwavejs2mqtt/pull/84

AlCalzone, 2021-01-04:
> Is there anyway to see how the node has been setup?

Not without debugging - it should be on the private `_deviceConfig` property of the node instance.

blhoward2, 2021-01-04:
I’m happy to try that later if you can point me in the general direction.

kpine, 2021-01-05:
I also own these switches and use this functionality, so I'm interested in seeing this resolved.

Just to confirm, for all devices BasicCC Set commands are automatically mapped to some other value? Coming from OZW, only Reports are mapped by default. If the mapping is disabled, how is the command recognized? Is it treated as a value or some kind of event?

In OZW a Basic Set triggers a "node event" instead of a value change, so a state is never maintained. That approach allows these switches to be used like a fake Central Scene. For example, when I do a double tap (Basic Set 0xff) it triggers my automation software to set the load to 100% brightness. If I double tap off (Basic Set 0x0) it goes to 20% brightness. When restarting the automation software or the zwave-js application (e.g. zwavejs2mqtt) it will incorrectly trigger the automation unless it there is something that can distinguish it from being a cached value. It's also a problem if the application doesn't transmit updates when there's no change in value, even though that has meaning to the automation (double-tap on -> set level to 100%, change level to 5%, double-tap on -> set level to 100%).

blhoward2, 2021-01-05:
> I also own these switches and use this functionality, so I'm interested in seeing this resolved.
>
> Just to confirm, for all devices BasicCC Set commands are automatically mapped to some other value? [...]

This is sort of how it works. In ozw 1.4 it sends a node event. In ozw 1.6 it does not, it updates a mapped sensor value using the basic set. The first step is to get it working with the value unmapped. Then we can figure out the best way to represent the event.

kpine, 2021-01-05:
> This is sort of how it works. In ozw 1.4 it sends a node event. In ozw 1.6 it does not, it updates a mapped sensor value using the basic set. 

Node Events are the default 1.6 behavior. To force a value requires setting compatibility flags in the device XML files. Not sure it's really relevant, except that the value behavior makes it difficult to handle double-tap events.

> The first step is to get it working with the value unmapped. Then we can figure out the best way to represent the event.

OK, well I think that's what my question was, I guess it is not decided yet. I just wanted to point out that there are applications of this that would benefit by having a way to distinguish between values that were triggered "now". vs. being cached, if an "event" is not the right API. Ozwdaemon is lacking that distinction which is a pain to deal with.

blhoward2, 2021-01-05:
> Node Events are the default 1.6 behavior. To force a value requires setting compatibility flags in the device XML files. Not sure it's really relevant, except that the value behavior makes it [...]

You're right. I looked back and was misremembering. In any event, irrelevant as you say. Updating the sensor didn't make it difficult..you got a sensor to automate off of. And an MQTT topic which is what I used in Appdaemon.

kpine, 2021-01-05:
> In any event, irrelevant as you say. Updating the sensor didn't make it difficult..you got a sensor to automate off of. And an MQTT topic which is what I used in Appdaemon.

For my automations it's difficult because there is no way to distinguish between a cached value vs. a live value. HA does not provide the capability via the sensors, and ozwd does not via the MQTT messages. I don't want the automations to trigger when HA or the z-wave gateway restart, only when the switch is actually pressed, the cached state is not useful. I've handled this by comparing the timestamps of the message vs "now", but it's not reliable.

blhoward2, 2021-01-05:
> @blhoward2 Please use this: [zwave-js/zwavejs2mqtt#84](https://github.com/zwave-js/zwavejs2mqtt/pull/84)
> 
> The branch name is `fix#23`

This branch would not start when built with the master branch of node-zwave-js. I left a comment in the PR for it with the error message.

robertsLando, 2021-01-05:
@kpine In version 6 there are `value notification`  and `value updated` events to distinguish them

robertsLando, 2021-01-05:
@blhoward2 I think that's a build error of zwave-js, are you using docker or are you building it your own?

AlCalzone, 2021-01-05:
> Just to confirm, for all devices BasicCC Set commands are automatically mapped to some other value? Coming from OZW, only Reports are mapped by default. If the mapping is disabled, how is the command recognized? Is it treated as a value or some kind of event?

With the aforementioned config flag, the Basic CC is not hidden and you will see persistent values for it. So far it made no sense to distinguish between Set and Report, because they mean the same most of the time.

What you write above sounds to me more like we need a different flag that treats Basic CC as events, not states.

blhoward2, 2021-01-05:
> @blhoward2 I think that's a build error of zwave-js, are you using docker or are you building it your own?

I used docker, using the command from the docker documentation you linked me to. I tried both the git-src one and downloading/checking out and the local-src one.

robertsLando, 2021-01-05:
Are you using zwavejs master with zwavejs2mqtt master?

blhoward2, 2021-01-05:
> Are you using zwavejs master with zwavejs2mqtt master?

For this I used the branch you indicated, plus the Zwavejs master. If you look at the screenshot where I said the build was uneventful the exact build command is at the top.

AlCalzone, 2021-01-05:
Dockerfile needs an update ==> https://github.com/zwave-js/zwavejs2mqtt/pull/154

robertsLando, 2021-01-05:
The problem is explained [here](https://github.com/zwave-js/zwavejs2mqtt/pull/154)

@larstobi The fix needed in DOcker contrib is explained there

robertsLando, 2021-01-05:
@blhoward2 try now, fixed by https://github.com/zwave-js/zwavejs2mqtt/pull/154

blhoward2, 2021-01-06:
Logging is broken so I can't see the debug-level, but I think this is working. Double tap is now updating command class 32 instead of 38.

```
2021-01-06 00:07:54.343 INFO ZWAVE: Node 3: value added: 32-0-currentValue => 255
2021-01-06 00:07:54.655 INFO ZWAVE: Node 3: value added 3-32-0-currentValue => 255
2021-01-06 00:08:07.259 INFO ZWAVE: Node 3: value updated: 38-0-currentValue 50 => 50
2021-01-06 00:08:20.287 INFO SOCKET: User disconnected 9lBSkwUPlCn3boyrAAAB
2021-01-06 00:08:27.776 INFO ZWAVE: Node 3: value updated: 32-0-currentValue 255 => 255
2021-01-06 00:08:32.627 INFO ZWAVE: Node 3: value updated: 32-0-currentValue 255 => 0
2021-01-06 00:08:38.582 INFO ZWAVE: Node 3: value updated: 32-0-currentValue 0 => 255
```

blhoward2, 2021-01-06:
> I also own these switches and use this functionality, so I'm interested in seeing this resolved.
>
> Just to confirm, for all devices BasicCC Set commands are automatically mapped to some other value? [...]

I haven't yet tested the reset part, but I did check and it does send the double-up or double-end to mqtt even when in that state already. 255-->255, etc.

blhoward2, 2021-01-06:
Thanks all! I think the node-js-zwave level of implementing this is complete. Further discussion under zwavejsmqtt: https://github.com/zwave-js/zwavejs2mqtt/issues/116

kpine, 2021-01-12:
> @kpine In version 6 there are `value notification` and `value updated` events to distinguish them

Could we get Basic Set to be treated as a notification? This change fixes the Basic Set issue for GE switches, but they are still values instead of notifications.

blhoward2, 2021-01-12:
> > @kpine In version 6 there are `value notification` and `value updated` events to distinguish them
> 
> Could we get Basic Set to be treated as a notification? This change fixes the Basic Set issue for GE switches, but they are still values instead of notifications.

This may make sense. I know @jcam was looking at this for Central Scene last night.

kpine, 2021-01-12:
> Does that break anything though where a switch actually needs basic set?

That's a valid question. My only perspective is coming from OZW, and pre-ozwd time, Basic Set was always an event, so in HA you'd use the `zwave.node_event` event. In OZW you have to jump through hoops to convert Set to Report (e.g. a value). However, I realize that's not necessarily the case here.

blhoward2, 2021-01-12:
> That's a valid question. My only perspective is coming from OZW, and pre-ozwd time, Basic Set was always an event, so in HA you'd use the `zwave.node_event` event. In OZW you have to jump through [...]

I too have only ever dealt with it in this context. I believe under the standard it is the failsafe for if anything to be able to control it, though. @AlCalzone thoughts as to if this ever gets used to actually control the switch?

kpine, 2021-01-12:
My follow question would be, if you are disabling the mapping it's for a good reason, would that be enough to assume notification instead of value. If not, could there be a setting as part of the flag that indicates which you prefer?

blhoward2, 2021-01-12:
Maybe it’s only used for basic control from other devices (e.g. direct association) and never the controller, so it would be ok? With respect to double taps it would probably be easier as it gets around HA not refreshing the value. I too recall it sending out zwave_events under ozw 1.4.

blhoward2, 2021-01-12:
Reopening so @AlCalzone can take a look

kpine, 2021-01-12:
> Maybe it’s only used for basic control from other devices (e.g. direct association) and never the controller, so it would be ok?

Typically Basic Set is for one device to control another, you can basically bypass the controller completely using associations. Almost all devices are required to respond to a basic set command in some way, for example a dimmer switch should allow for basic set to toggle or change the brightness. I doubt Jasco/GE has our use case in mind when they designed the switches this way. They probably just wanted to let you control other devices (with either single or double click) by direct association.

> With respect to double taps it would probably be easier as it gets around HA not refreshing the value. I too recall it sending out zwave_events under ozw 1.4.

Well, that's my motivation in digging into this. 😁 The ozw 1.4 event method worked perfectly with no complications. The incomplete method adopted by ozwd made this difficult because of the persistent state. I'm worried about how HA will handle the values with zwave-js as well, if they aren't notifications.

blhoward2, 2021-01-12:
@jcam posted this in slack last night, if you need an immediate workaround:
https://www.home-assistant.io/integrations/device_trigger.mqtt/

towerhand, 2021-01-12:
I’m using the mqtt trigger for the double taps with z2m and it works well, could never get it to work with qtozw due to the persistent values.

kpine, 2021-01-12:
By "z2m" you mean "zwavejs2mqtt" right? Is the topic retained? I assume it's not.

I'll be using the HA integration so I don't want the same problem to show up there.

blhoward2, 2021-01-12:
> I’m using the mqtt trigger for the double taps with z2m and it works well, could never get it to work with qtozw due to the persistent values.

Did you add this as a custom sensor or whatever yourself? Could you share the code?

AlCalzone, 2021-01-12:
Sorry, I don't follow here. Can somebody break it down for me?

blhoward2, 2021-01-12:
After this fix, a double-tap sends a value update, not a notification (event). The result is that HA makes this persistent and doesn’t see a subsequent double-up or double-down if it is already in the 0/255 state. Under ozw 1.4 doing this generate an event that came across as a zwave_event. Central Scene has a similar problem. If the device was last used to activate scene 1, it won’t see a second scene 1 to automate off o.

towerhand, 2021-01-12:
I meant Zwave2mqtt, I would be ok if zwavejs2mqtt behaves the same which reading the issue seems like it’s heading that way.

i can share the automation in a couple hours and test the new version of zjs2m this weekend.

kpine, 2021-01-12:
> Sorry, I don't follow here. Can somebody break it down for me?

Here's how OZW 1.4 (and 1.6 non-ozwd) dealt with Basic Set:

1. Device sends Basic Set to controller
2. In OZW this goes gets propagated through the NodeEvent API
3. In HA (which consumes the C++ API via Python) handles this as an event, and you can write automations to trigger on the event.

At no time in the above is the state of the Basic Set value stored anywhere. When I restart HA there is no risk of my automation ever being triggered. If I send duplicate events, such as two on double taps sequentially, the event is triggered because it doesn't care about previous state. You can compare this to the use case to a Central Scene notification if you want.

In ozwdaemon (and some future OZW version) the path forward is to get rid of the Node Event API entirely, and treat Basic Set just like a Basic Report, as a value (it also does this with Central Scene values). There are multiple problems with this:

1. By default, OZW was masking value refreshes, which means two double-tap on events (basic set 0xff) would not be reported. This was changed to emit a duplicate MQTT topic with a new timestamp. It looks like you've already solved this problem.
2. Sometimes applications also mask state values that are received, but don't change value. HA disables this for certain values. It's not clear yet if HA would detect duplicate value updates and emit a state trigger w/o change.
3. Depending on the design of the application, at startup when it processes all the existing values, it may or may not emit state changes when values are added. It does for Basic values, which means an automation based on this will get triggered, unless some precautions are taken. It remains to be determined how HA would handle this value for the zwave-js integration, but based on past implementation it would be state trigger.

In my ideal world, for these devices a Basic Set would be treated as a notification, like Central Scene, because that's what they are. I could then ask HA to add an event trigger to achieve the same behavior as OZW 1.4.

kpine, 2021-01-12:
> I meant Zwave2mqtt, I would be ok if zwavejs2mqtt behaves the same which reading the issue seems like it’s heading that way.

Ah, zwave2mqtt is using OZW 1.6 and uses the "deprecated" Node Event API. That explains the difference from ozwd. Although, if you picked up any of the recent XML changes it might not be working the same way.

blhoward2, 2021-01-12:
@robertsLando apparently made changes in zwavejs2mqtt that drops the retain flag. I think we have two different potential fixes going on here. Does that fix this concern? Though it wouldn’t apply to the HA integration.

[24 more replies]

PR #1379 (2021-01-14): feat(config): add compat option treatBasicSetAsEvent and use it for GE devices
fixes: #1277
closed 2021-01-14 by AlCalzone