zwave-js/zwave-js #1283: Add missing Neo Coolcam device (INFO ZWAVE: Node 6 ready: Unknown manufacturer 600 - Unknown product 4135 (512))

dkjsj, 2021-01-04:
**Describe the bug**
Tried to add a Neo Coolcam Plug device. It' results in the information from the logfile:

`INFO ZWAVE: Node 6 ready: Unknown manufacturer 600 - Unknown product 4135 (512)`

Besides that - the device in it self works. 

Any guidance on how to have the right manufactor and product listed?

**Device information**
Which device(s) is/are affected?  Shenzhen Neo Electronics Co Ltd NAS-WR01Z Power plug 12A.

Link to z-wavealliance.org manual: [link](https://products.z-wavealliance.org/ProductManual/File?folder=&filename=Manuals/1783/Smart-Plug%20User%20Guide%20V2.doc)

Link to Product page: [link](https://szneo.com/en/products/show.php?id=189)

What are the node IDs?
Screendump:
[image]

**Logfile:** 

```
2021-01-04 12:56:47.755 INFO ZWAVE: Controller status: Secure inclusion started
2021-01-04 12:56:47.755 INFO ZWAVE: Success zwave api call startInclusion true
2021-01-04 12:57:21.991 INFO ZWAVE: Controller status: Inclusion stopped
2021-01-04 12:57:21.996 INFO ZWAVE: Node 6: added
2021-01-04 12:57:21.997 INFO ZWAVE: Node added 6
2021-01-04 12:57:22.095 INFO ZWAVE: Node 6 is alive
2021-01-04 12:57:22.096 INFO ZWAVE: Controller status: Driver: Node 006 supports Security, but no network key was configured. Continuing interview non-securely.
2021-01-04 12:57:22.147 INFO ZWAVE: Node 6: value added: 114-0-manufacturerId => 600
2021-01-04 12:57:22.148 INFO ZWAVE: Node 6: value added: 114-0-productType => 512
2021-01-04 12:57:22.148 INFO ZWAVE: Node 6: value added: 114-0-productId => 4135
2021-01-04 12:57:22.208 INFO ZWAVE: Node 6: value added: 134-0-libraryType => 3
2021-01-04 12:57:22.210 INFO ZWAVE: Node 6: value added: 134-0-protocolVersion => 6.4
2021-01-04 12:57:22.211 INFO ZWAVE: Node 6: value added: 134-0-firmwareVersions => 2.23
2021-01-04 12:57:22.793 INFO ZWAVE: Node 6: value added: 134-0-sdkVersion => 6.81.3
2021-01-04 12:57:22.795 INFO ZWAVE: Node 6: value added: 134-0-applicationFrameworkAPIVersion => 4.1.2
2021-01-04 12:57:22.796 INFO ZWAVE: Node 6: value added: 134-0-applicationFrameworkBuildNumber => 52445
2021-01-04 12:57:22.798 INFO ZWAVE: Node 6: value added: 134-0-hostInterfaceVersion => unused
2021-01-04 12:57:22.799 INFO ZWAVE: Node 6: value added: 134-0-hostInterfaceBuildNumber => 0
2021-01-04 12:57:22.800 INFO ZWAVE: Node 6: value added: 134-0-zWaveProtocolVersion => 6.4.0
2021-01-04 12:57:22.801 INFO ZWAVE: Node 6: value added: 134-0-zWaveProtocolBuildNumber => 91
2021-01-04 12:57:22.802 INFO ZWAVE: Node 6: value added: 134-0-applicationVersion => 2.23.0
2021-01-04 12:57:22.802 INFO ZWAVE: Node 6: value added: 134-0-applicationBuildNumber => 52445
2021-01-04 12:57:22.908 INFO ZWAVE: Node 6: value added: 37-0-currentValue => false
2021-01-04 12:57:22.909 INFO ZWAVE: Node 6: value added: 37-0-targetValue => undefined
2021-01-04 12:57:22.970 INFO ZWAVE: Node 6: metadata updated: 50-0-reset
2021-01-04 12:57:23.026 INFO ZWAVE: Node 6: metadata updated: 50-0-value-65537
2021-01-04 12:57:23.027 INFO ZWAVE: Node 6: metadata updated: 50-0-previousValue-65537
2021-01-04 12:57:23.028 INFO ZWAVE: Node 6: metadata updated: 50-0-deltaTime-65537
2021-01-04 12:57:23.028 INFO ZWAVE: Node 6: value added: 50-0-value-65537 => -21474816.39
2021-01-04 12:57:23.029 INFO ZWAVE: Node 6: value added: 50-0-previousValue-65537 => -21474816.39
2021-01-04 12:57:23.030 INFO ZWAVE: Node 6: value added: 50-0-deltaTime-65537 => 12
2021-01-04 12:57:23.081 INFO ZWAVE: Node 6: metadata updated: 50-0-value-66049
2021-01-04 12:57:23.083 INFO ZWAVE: Node 6: metadata updated: 50-0-previousValue-66049
2021-01-04 12:57:23.088 INFO ZWAVE: Node 6: metadata updated: 50-0-deltaTime-66049
2021-01-04 12:57:23.089 INFO ZWAVE: Node 6: value added: 50-0-value-66049 => 0
2021-01-04 12:57:23.090 INFO ZWAVE: Node 6: value added: 50-0-previousValue-66049 => 0
2021-01-04 12:57:23.090 INFO ZWAVE: Node 6: value added: 50-0-deltaTime-66049 => 12
2021-01-04 12:57:23.139 INFO ZWAVE: Node 6: metadata updated: 50-0-value-66561
2021-01-04 12:57:23.141 INFO ZWAVE: Node 6: metadata updated: 50-0-previousValue-66561
2021-01-04 12:57:23.143 INFO ZWAVE: Node 6: metadata updated: 50-0-deltaTime-66561
2021-01-04 12:57:23.144 INFO ZWAVE: Node 6: value added: 50-0-value-66561 => 0
2021-01-04 12:57:23.145 INFO ZWAVE: Node 6: value added: 50-0-previousValue-66561 => 0
2021-01-04 12:57:23.146 INFO ZWAVE: Node 6: value added: 50-0-deltaTime-66561 => 12
2021-01-04 12:57:23.193 INFO ZWAVE: Node 6: metadata updated: 50-0-value-66817
2021-01-04 12:57:23.194 INFO ZWAVE: Node 6: metadata updated: 50-0-previousValue-66817
2021-01-04 12:57:23.195 INFO ZWAVE: Node 6: metadata updated: 50-0-deltaTime-66817
2021-01-04 12:57:23.195 INFO ZWAVE: Node 6: value added: 50-0-value-66817 => 0
2021-01-04 12:57:23.196 INFO ZWAVE: Node 6: value added: 50-0-previousValue-66817 => 0
2021-01-04 12:57:23.196 INFO ZWAVE: Node 6: value added: 50-0-deltaTime-66817 => 12
2021-01-04 12:57:23.904 INFO ZWAVE: Node 6: metadata updated: 113-0-Power Management-Over-load status
2021-01-04 12:57:23.941 INFO ZWAVE: Node 6: value added: 113-0-Power Management-Over-load status => 0
2021-01-04 12:57:23.995 INFO ZWAVE: Node 6: value added 6-37-0-currentValue => false
2021-01-04 12:57:23.996 INFO ZWAVE: Node 6: value added 6-37-0-targetValue => undefined
2021-01-04 12:57:23.996 INFO MQTT: Subscribing to z2m/+/+/+/+/set
2021-01-04 12:57:23.996 INFO ZWAVE: Node 6: value added 6-134-0-libraryType => 3
2021-01-04 12:57:23.997 INFO ZWAVE: Node 6: value added 6-134-0-protocolVersion => 6.4
2021-01-04 12:57:23.997 INFO ZWAVE: Node 6: value added 6-134-0-firmwareVersions => 2.23
2021-01-04 12:57:23.998 INFO ZWAVE: Node 6: value added 6-134-0-hardwareVersion => undefined
2021-01-04 12:57:23.998 INFO ZWAVE: Node 6: value added 6-134-0-sdkVersion => 6.81.3
2021-01-04 12:57:23.999 INFO ZWAVE: Node 6: value added 6-134-0-applicationFrameworkAPIVersion => 4.1.2
2021-01-04 12:57:23.999 INFO ZWAVE: Node 6: value added 6-134-0-applicationFrameworkBuildNumber => 52445
2021-01-04 12:57:24.000 INFO ZWAVE: Node 6: value added 6-134-0-hostInterfaceVersion => unused
2021-01-04 12:57:24.000 INFO ZWAVE: Node 6: value added 6-134-0-hostInterfaceBuildNumber => 0
2021-01-04 12:57:24.001 INFO ZWAVE: Node 6: value added 6-134-0-zWaveProtocolVersion => 6.4.0
2021-01-04 12:57:24.001 INFO ZWAVE: Node 6: value added 6-134-0-zWaveProtocolBuildNumber => 91
2021-01-04 12:57:24.002 INFO ZWAVE: Node 6: value added 6-134-0-applicationVersion => 2.23.0
2021-01-04 12:57:24.002 INFO ZWAVE: Node 6: value added 6-134-0-applicationBuildNumber => 52445
2021-01-04 12:57:24.003 INFO ZWAVE: Node 6: value added 6-114-0-manufacturerId => 600
2021-01-04 12:57:24.003 INFO ZWAVE: Node 6: value added 6-114-0-productType => 512
2021-01-04 12:57:24.004 INFO ZWAVE: Node 6: value added 6-114-0-productId => 4135
2021-01-04 12:57:24.004 INFO ZWAVE: Node 6: value added 6-50-0-value-65537 => -21474816.39
2021-01-04 12:57:24.004 INFO ZWAVE: Node 6: value added 6-50-0-previousValue-65537 => -21474816.39
2021-01-04 12:57:24.005 INFO ZWAVE: Node 6: value added 6-50-0-deltaTime-65537 => 12
2021-01-04 12:57:24.005 INFO ZWAVE: Node 6: value added 6-50-0-value-66049 => 0
2021-01-04 12:57:24.005 INFO ZWAVE: Node 6: value added 6-50-0-previousValue-66049 => 0
2021-01-04 12:57:24.006 INFO ZWAVE: Node 6: value added 6-50-0-deltaTime-66049 => 12
2021-01-04 12:57:24.006 INFO ZWAVE: Node 6: value added 6-50-0-value-66561 => 0
2021-01-04 12:57:24.006 INFO ZWAVE: Node 6: value added 6-50-0-previousValue-66561 => 0
2021-01-04 12:57:24.007 INFO ZWAVE: Node 6: value added 6-50-0-deltaTime-66561 => 12
2021-01-04 12:57:24.007 INFO ZWAVE: Node 6: value added 6-50-0-value-66817 => 0
2021-01-04 12:57:24.007 INFO ZWAVE: Node 6: value added 6-50-0-previousValue-66817 => 0
2021-01-04 12:57:24.008 INFO ZWAVE: Node 6: value added 6-50-0-deltaTime-66817 => 12
2021-01-04 12:57:24.008 INFO ZWAVE: Node 6: value added 6-50-0-reset => undefined
2021-01-04 12:57:24.008 INFO ZWAVE: Node 6: value added 6-113-0-Power Management-Over-load status => 0
2021-01-04 12:57:24.011 INFO ZWAVE: Node 6 ready: Unknown manufacturer 600 - Unknown product 4135 (512)
2021-01-04 12:57:24.012 INFO ZWAVE: Node 6: interview completed, all values are updated
```

AlCalzone, 2021-01-04:
Duplicate of #1221

AlCalzone, 2021-01-04:
> Any guidance on how to have the right manufactor and product listed?

We need a config file for that. See also the linked original issue. If you'd like to help out, here are some docs to get you started:
https://zwave-js.github.io/node-zwave-js/#/development/config-files

closed 2021-01-04 by AlCalzone