zwave-js/zwave-js #1298: [feat] Implement Hail Class (and collect examples for use in doing so)

blhoward2, 2021-01-05:
**Is your feature request related to a problem? Please describe.**

As discussed in https://github.com/zwave-js/node-zwave-js/issues/1287, the Hail class should be implemented to support older devices and to reduce the need for users to reconfigure their devices. This is a usability issue as devices relying on Hail will not see a state change upon manual control, control via direct association, or when the device intends to send a report.

For users, the Hail class effectively asks the controller to poll devices. **Please collect examples of what values are polled upon a Hail and post them to this issue to assist in implementation.**

**Describe the solution you'd like**
Implement new Class/controller actions upon receiving a Hail

AlCalzone, 2021-01-06:
I've taken a stab at this. Basically we just refresh all sensor and actuator values from the node when a Hail is received. Not sure if this is the correct way to do it, but we can try.

@blhoward2 mind giving it a test-run? The bot should post a test build in the linked PR soon.

blhoward2, 2021-01-06:
Will do

blhoward2, 2021-01-08:
I can now build a docker image including this but I cannot test it due to the broken logging. The dropping hail class message only appears in the zwave log...

AlCalzone, 2021-01-08:
Ok let's just wait until tomorrow, we should have sorted out the logging by then.
Otherwise it's just guessing blindly 😅

blhoward2, 2021-01-08:
Once logging is back up, is there a way to see the actual payload of the Hail message? I’d like to see if they’re all the same but at the debug level I just get a message (under the master branch) saying the message has been dropped.  Do I need to finally get around to figuring out the debugger?

AlCalzone, 2021-01-08:
It doesn't have any payload. That's basically just 2 bytes which identify the Hail CC.

blhoward2, 2021-01-08:
Ok, helpful thanks.

PR #1308 (2021-01-10): feat(cc): implement and handle Hail CC
This PR adds support for the Hail CC. Until we know better what should be done in this case, refresh all Actuator and Sensor values from the hailing node (like we do when a NIF is received without a Lifeline)

fixes: #1298
closed 2021-01-10 by AlCalzone