zwave-js/zwave-js #1304: [bug] Association group label is not respected

blhoward2, 2021-01-06:
**Describe the bug**
I'm unsure if this issue arises at the node-js-zwave or zwavejs2mqtt levels. I created new device files for GE switches and dimmers  #1277 / #1303. Despite configuring the label for the newly defined association groups per the documentation, the UI does not reflect that naming. (Group 3 should be named Double Tap for clarity.) The Lifeline designation for Group 3 is respected so I know that portion of the device file is being imported.

E.g.:

[image]

But I have renamed that group to: Basic - Double Tap

**To Reproduce**
Steps to reproduce the behavior:
1. Go to create label for node
2. Rebuild cache or refresh info (I tried both)

**Device information**
Which device(s) is/are affected?  Multiple GE switches and dimmers
What are the node IDs?

**Logfile:**

blhoward2, 2021-01-06:
@robertsLando Please let me know if this actually belongs under zwavejs2mqtt. Thanks!

robertsLando, 2021-01-06:
Could you send me the config file used by your device?

AlCalzone, 2021-01-06:
That might be because the device supports the `Association Group Information CC`, which also advertises association labels. I'll need to look at the code what has preference here.

blhoward2, 2021-01-06:
They’re the ones I uploaded to the PR.

AlCalzone, 2021-01-06:
Yep, that should be the labels reported by the `Association Group Information CC`. Seems a bit dumb IMO to have the device report different labels than in the config files.

We could add a property the the association definitions in config files to tell zwave-js to prefer them over what AGI reports.

PR #1570 (2021-02-03): fix(controller): prefer configured association labels over AGI
fixes: #1304

Seems like the `Effort: medium` label was not really accurate 😅
closed 2021-02-03 by AlCalzone