zwave-js/zwave-js #1309: [bug] Unable to properly include Thermofloor Z-TRM2FX thermostats

jkaberg, 2021-01-07:
**Version**

Build/Run method
- [X] Docker
- [ ] PKG
- [ ] Manually built (git clone - npm install - npm run build ) 

zwavejs2mqtt version: develop (1.0.0-alpha.2)

**Describe the bug**
Unable to properly include [Thermofloor Z-TRM2FX thermostats](https://github.com/zwave-js/node-zwave-js/blob/9fa9a67cd52321de801efdb4d324dfe023759e86/packages/config/config/devices/0x019b/z-trm2fx_3.0.json) using the normal inclusion method.

This happends to all my Z-TRM2FX thermostats (I have 6). All these included fine on OZW. Somehow one odd Thermostat actully got included after many heals/refresh info and container restarts. Couldnt reproduce this on the following 5.

Except for node 45 in the picture below, the other thermostats where added many hours ago and should have completed thire interviews. It seems they hang on interview stage **NodeInfo** or **ProtocolInfo**.

**To Reproduce**
Steps to reproduce the behavior:
1. Actions -> Start inclusion [RUN]
2. Vise versa on thermostat

**Expected behavior**
The thermostat properly included with zwavejs2mqtt. 

**Additional context**

Picture (taken after the logs below - notice node 45):
[image]

Logs (first excluding the thermostat, as to then include it again - alas Node 44 and 45 is that same unit):
```
zwavejs2mqtt_1     | 2021-01-06 18:05:04.406 INFO SOCKET: Event ZWAVE_API emitted to CqEDoxtGxPW0_I3lAAAD
zwavejs2mqtt_1     | 2021-01-06 18:05:04.410 INFO APP: Zwave api call: startExclusion [ [length]: 0 ]
zwavejs2mqtt_1     | 2021-01-06 18:05:09.927 INFO ZWAVE: Controller status: Exclusion started
zwavejs2mqtt_1     | 2021-01-06 18:05:09.928 INFO ZWAVE: Success zwave api call startExclusion true
zwavejs2mqtt_1     | 2021-01-06 18:05:20.899 INFO ZWAVE: ValueRemoved: no such node: 44 error
zwavejs2mqtt_1     | 2021-01-06 18:05:20.901 INFO ZWAVE: Node 44: value removed: 114-0-manufacturerId
zwavejs2mqtt_1     | 2021-01-06 18:05:20.902 INFO ZWAVE: ValueRemoved: no such node: 44 error
zwavejs2mqtt_1     | 2021-01-06 18:05:20.902 INFO ZWAVE: Node 44: value removed: 114-0-productType
zwavejs2mqtt_1     | 2021-01-06 18:05:20.903 INFO ZWAVE: ValueRemoved: no such node: 44 error
zwavejs2mqtt_1     | 2021-01-06 18:05:20.903 INFO ZWAVE: Node 44: value removed: 114-0-productId
zwavejs2mqtt_1     | 2021-01-06 18:05:20.904 INFO ZWAVE: ValueRemoved: no such node: 44 error
zwavejs2mqtt_1     | 2021-01-06 18:05:20.904 INFO ZWAVE: Node 44: value removed: 134-0-libraryType
zwavejs2mqtt_1     | 2021-01-06 18:05:20.906 INFO ZWAVE: ValueRemoved: no such node: 44 error
zwavejs2mqtt_1     | 2021-01-06 18:05:20.906 INFO ZWAVE: Node 44: value removed: 134-0-protocolVersion
zwavejs2mqtt_1     | 2021-01-06 18:05:20.907 INFO ZWAVE: ValueRemoved: no such node: 44 error
zwavejs2mqtt_1     | 2021-01-06 18:05:20.907 INFO ZWAVE: Node 44: value removed: 134-0-firmwareVersions
zwavejs2mqtt_1     | 2021-01-06 18:05:20.908 INFO ZWAVE: ValueRemoved: no such node: 44 error
zwavejs2mqtt_1     | 2021-01-06 18:05:20.908 INFO ZWAVE: Node 44: value removed: 134-0-sdkVersion
zwavejs2mqtt_1     | 2021-01-06 18:05:20.909 INFO ZWAVE: ValueRemoved: no such node: 44 error
zwavejs2mqtt_1     | 2021-01-06 18:05:20.909 INFO ZWAVE: Node 44: value removed: 134-0-applicationFrameworkAPIVersion
zwavejs2mqtt_1     | 2021-01-06 18:05:20.910 INFO ZWAVE: ValueRemoved: no such node: 44 error
zwavejs2mqtt_1     | 2021-01-06 18:05:20.910 INFO ZWAVE: Node 44: value removed: 134-0-applicationFrameworkBuildNumber
zwavejs2mqtt_1     | 2021-01-06 18:05:20.911 INFO ZWAVE: ValueRemoved: no such node: 44 error
zwavejs2mqtt_1     | 2021-01-06 18:05:20.911 INFO ZWAVE: Node 44: value removed: 134-0-hostInterfaceVersion
zwavejs2mqtt_1     | 2021-01-06 18:05:20.912 INFO ZWAVE: ValueRemoved: no such node: 44 error
zwavejs2mqtt_1     | 2021-01-06 18:05:20.912 INFO ZWAVE: Node 44: value removed: 134-0-hostInterfaceBuildNumber
zwavejs2mqtt_1     | 2021-01-06 18:05:20.913 INFO ZWAVE: ValueRemoved: no such node: 44 error
zwavejs2mqtt_1     | 2021-01-06 18:05:20.913 INFO ZWAVE: Node 44: value removed: 134-0-zWaveProtocolVersion
zwavejs2mqtt_1     | 2021-01-06 18:05:20.913 INFO ZWAVE: ValueRemoved: no such node: 44 error
zwavejs2mqtt_1     | 2021-01-06 18:05:20.914 INFO ZWAVE: Node 44: value removed: 134-0-zWaveProtocolBuildNumber
zwavejs2mqtt_1     | 2021-01-06 18:05:20.914 INFO ZWAVE: ValueRemoved: no such node: 44 error
zwavejs2mqtt_1     | 2021-01-06 18:05:20.914 INFO ZWAVE: Node 44: value removed: 134-0-applicationVersion
zwavejs2mqtt_1     | 2021-01-06 18:05:20.915 INFO ZWAVE: ValueRemoved: no such node: 44 error
zwavejs2mqtt_1     | 2021-01-06 18:05:20.915 INFO ZWAVE: Node 44: value removed: 134-0-applicationBuildNumber
zwavejs2mqtt_1     | 2021-01-06 18:05:20.921 INFO ZWAVE: ValueRemoved: no such node: 44 error
zwavejs2mqtt_1     | 2021-01-06 18:05:20.921 INFO ZWAVE: Node 44: value removed: 67-1-setpoint-1
zwavejs2mqtt_1     | 2021-01-06 18:05:20.922 INFO ZWAVE: ValueRemoved: no such node: 44 error
zwavejs2mqtt_1     | 2021-01-06 18:05:20.922 INFO ZWAVE: Node 44: value removed: 67-1-setpoint-2
zwavejs2mqtt_1     | 2021-01-06 18:05:20.923 INFO ZWAVE: ValueRemoved: no such node: 44 error
[... 263 lines cut ...]
zwavejs2mqtt_1     | 2021-01-06 18:09:39.107 INFO ZWAVE: Node 45: metadata updated: 112-0-1
zwavejs2mqtt_1     | 2021-01-06 18:09:39.108 INFO ZWAVE: Node 45: metadata updated: 112-0-2
zwavejs2mqtt_1     | 2021-01-06 18:09:39.109 INFO ZWAVE: Node 45: metadata updated: 112-0-3
zwavejs2mqtt_1     | 2021-01-06 18:09:39.110 INFO ZWAVE: Node 45: metadata updated: 112-0-4
zwavejs2mqtt_1     | 2021-01-06 18:09:39.111 INFO ZWAVE: Node 45: metadata updated: 112-0-5
zwavejs2mqtt_1     | 2021-01-06 18:09:39.112 INFO ZWAVE: Node 45: metadata updated: 112-0-6
zwavejs2mqtt_1     | 2021-01-06 18:09:39.113 INFO ZWAVE: Node 45: metadata updated: 112-0-7
zwavejs2mqtt_1     | 2021-01-06 18:09:39.114 INFO ZWAVE: Node 45: metadata updated: 112-0-8
zwavejs2mqtt_1     | 2021-01-06 18:09:39.115 INFO ZWAVE: Node 45: metadata updated: 112-0-9
zwavejs2mqtt_1     | 2021-01-06 18:09:39.115 INFO ZWAVE: Node 45: metadata updated: 112-0-10
zwavejs2mqtt_1     | 2021-01-06 18:09:39.116 INFO ZWAVE: Node 45: metadata updated: 112-0-11
zwavejs2mqtt_1     | 2021-01-06 18:09:39.117 INFO ZWAVE: Node 45: metadata updated: 112-0-12
zwavejs2mqtt_1     | 2021-01-06 18:09:39.118 INFO ZWAVE: Node 45: metadata updated: 112-0-13
zwavejs2mqtt_1     | 2021-01-06 18:09:39.118 INFO ZWAVE: Node 45: metadata updated: 112-0-14
zwavejs2mqtt_1     | 2021-01-06 18:09:39.119 INFO ZWAVE: Node 45: metadata updated: 112-0-15
zwavejs2mqtt_1     | 2021-01-06 18:09:39.120 INFO ZWAVE: Node 45: metadata updated: 112-0-16
zwavejs2mqtt_1     | 2021-01-06 18:09:39.120 INFO ZWAVE: Node 45: metadata updated: 112-0-17
zwavejs2mqtt_1     | 2021-01-06 18:09:39.121 INFO ZWAVE: Node 45: metadata updated: 112-0-18
zwavejs2mqtt_1     | 2021-01-06 18:09:39.122 INFO ZWAVE: Node 45: metadata updated: 112-0-19
zwavejs2mqtt_1     | 2021-01-06 18:09:39.122 INFO ZWAVE: Node 45: metadata updated: 112-0-20
zwavejs2mqtt_1     | 2021-01-06 18:09:39.123 INFO ZWAVE: Node 45: metadata updated: 112-0-21
zwavejs2mqtt_1     | 2021-01-06 18:09:39.123 INFO ZWAVE: Node 45: metadata updated: 112-0-22
```

I tried both debug and silly log levels (from the UI), but I only got info? ^

jkaberg, 2021-01-07:
@AlCalzone Is this enough? do you need anything else?

[debug.log](https://github.com/zwave-js/node-zwave-js/files/5782254/debug.log)

jkaberg, 2021-01-08:
@AlCalzone I could try to fix this myself, albeit I'm unsure how. I see from the other thread that you mention adding an combat entry to the device config. Is there any documentation  (I couldn't find any) that I can take a look at?

AlCalzone, 2021-01-08:
The docs won't tell you how to implement a compat option. If you want to dive in, this PR should help you figure out how it is done:
https://github.com/zwave-js/node-zwave-js/pull/1278

If the new option is `true`, the following query should not be executed:
https://github.com/zwave-js/node-zwave-js/blob/aaa0d823fa34df35140f85db33dcd5c7464cd9c0/packages/zwave-js/src/lib/commandclass/ConfigurationCC.ts#L581-L583

jkaberg, 2021-01-08:
@AlCalzone alright! But wouldnt whatever the interview is doing at this point (collecting information?) be dropped by this? Is the information not needed?

I'll look into it during the weekend

AlCalzone, 2021-01-08:
`getInfo` just returns the device-defined description of the parameter (if the device supports that), so it is not critical

jkaberg, 2021-01-09:
@AlCalzone Ok, so I gave it a shot. Seems I end up with the same failure after adding compat?

Did restart the container inbetween editing the config file for changes to take effect (I did confirm after restart that the changes were still there before inclusion)

Attaching logs and config file
[z-trm2fx_3.0.json.txt](https://github.com/zwave-js/node-zwave-js/files/5790260/z-trm2fx_3.0.json.txt)
[debug2.log](https://github.com/zwave-js/node-zwave-js/files/5790261/debug2.log)

Any suggestions?

jkaberg, 2021-01-09:
And a quick question while trying to understand the logs:

```
06:42:47.745 SERIAL » 0x011d0013301698813512ff7d3204fac64ebc3ef34b6793ae65550f91254f8b    (31 bytes)
06:42:47.745 DRIVER » [Node 048] [REQ] [SendData]
                      │ transmit options: 0x25
                      │ callback id:      79
                      └─[SecurityCCCommandEncapsulation]
                        │ nonce id: 243
                        └─[SecurityCCCommandsSupportedGet]
06:42:47.747 SERIAL « [ACK]                                                                   (0x06)
06:42:47.754 SERIAL « 0x0104011301e8                                                       (6 bytes)
06:42:47.754 SERIAL » [ACK]                                                                   (0x06)
06:42:47.754 DRIVER « [RES] [SendData]
                        was sent: true
06:42:48.015 SERIAL « 0x010700134f00001bbf                                                 (9 bytes)
06:42:48.015 SERIAL » [ACK]                                                                   (0x06)
06:42:48.015 DRIVER « [REQ] [SendData]
                        callback id:     79
                        transmit status: OK
06:42:49.617 CNTRLR   [Node 048] did not respond after 1/3 attempts. Scheduling next try in 500 ms.
```
So the `did not respond after 1/3...` message, is it refering to the last command, eg in this case to:
```06:42:47.745 DRIVER » [Node 048] [REQ] [SendData]
                      │ transmit options: 0x25
                      │ callback id:      79
                      └─[SecurityCCCommandEncapsulation]
                        │ nonce id: 243
                        └─[SecurityCCCommandsSupportedGet]
```
? If thats the case, how do we drop/compat `SecurityCCCommandEncapsulation` and/or `SecurityCCCommandsSupportedGet`?

AlCalzone, 2021-01-09:
Regarding the security stuff - that's just the driver figuring out that the device is not securely included.

It seems like your device does not have a problem with the getInfo yet, because the `api.getProperties(0);` to find the first parameter is also failing. Maybe the compat option should instead disable the V3 interview entirely and choose the branch `if (this.version < 3) {`

jkaberg, 2021-01-09:
@AlCalzone I think you might be right, looking over the config file only [V2 is mentioned](https://github.com/OpenZWave/open-zwave/blob/master/config/thermofloor/heatit056.xml#L172-L183)

I guess such feature/workaround isnt implemeneted yet, atleast I didn't see anything in the [docs](https://github.com/zwave-js/node-zwave-js/blob/bc0b69308021c6a037e41a08d041aae4b2cde453/docs/development/config-files.md) about it (?)

AlCalzone, 2021-01-10:
@jkaberg One thing I just found in your log:
```
06:42:58.254 CONFIG   Error loading device config /usr/src/app/node_modules/@zwave-js/config/config/
                      devices/0x019b/z-trm2fx_3.0.json
```
I'm not sure what causes this. Just keep an eye out for this when testing the PR

PR #1341 (2021-01-11): feat(config): add compat option skipConfigurationInfoQuery
This PR adds a new compat option to skip the `ConfigurationCC` parameter info query which causes devices like the `Z-TRM2fx` to flood the network with garbage reports.

In addition, it adds a workaround for a bug **also** present in this device from hell, which causes the V3 query for the first parameter number to **also** send an invalid response.

fixes: #1309
fixes: #1128
closed 2021-01-11 by AlCalzone