zwave-js/zwave-js #1311: @AlCalzone I think this may need another bit of work, there is now a 46202.json and a ge_46202.json that refer to the same device IDs.

AlCalzone, 2021-01-07:
@AlCalzone I think this may need another bit of work, there is now a 46202.json and a ge_46202.json that refer to the same device IDs.
Also, should there be a test added to the config checker to block having two configs for the same IDs?

__Originally posted by @jcam in https://github.com/zwave-js/node-zwave-js/pull/1303#issuecomment-756005678__

blhoward2, 2021-01-07:
46202 should have been deleted (I removed this from my commit) and ge_46202 amended with the changes. It looks like you amended 46202 instead. That’s the only one.

AlCalzone, 2021-01-07:
Is the 46202 file correct the way it is?

blhoward2, 2021-01-07:
Correct. I compared them and 46202 is correct and ahead of ge_46202.

blhoward2, 2021-01-07:
PR submitted to delete ge_46202

blhoward2, 2021-01-29:
@AlCalzone lint now checks for duplicate device files, so this should be closed, correct?

closed 2021-01-29 by AlCalzone