zwave-js/zwave-js #1314: Kwikset 914c lock not reporting manual lock change state

jaburges, 2021-01-07:
**Describe the bug**
BE469ZP lock, when turning the lock manually the state of `/boltStatus` changes.

914c Kwikset lock does NOT report manual lock changes

**To Reproduce**
manual lock turn
Device responds to zwave (and MQTT) commands

**Device information**
Which device(s) is/are affected?  914c
What are the node IDs? 104 & 107

Log is huge - i'll work on grabbing a smaller chunk

AlCalzone, 2021-01-07:
> Log is huge - i'll work on grabbing a smaller chunk

You can try to set the LOG_NODES env variable to 104,107

AlCalzone, 2021-01-31:
You should be able to configure polling, e.g. react to specific device notifications now in zwavejs2mqtt. Please try if that is enough for you - or reopen if necessary.

closed 2021-01-31 by AlCalzone

rccoleman, 2021-02-04:
I can see how to have zwjs2mqtt poll a value, but I don't see a way to poll a value based on a notification.  Or perhaps I'm misunderstanding what you're suggesting.

I have a couple of Kwikset 914 locks that send a notification for manual lock/unlock and I'm currently triggering on that event in HA and "fixing" the lock state, but it sounds like I may be able to do that in zwjs2mqtt - poll a value when a notification is received.  Is that possible?

AlCalzone, 2021-02-04:
The idea is that you can (hopefully soon) set up an automation in HA which asks zwave-js to poll the value. I'll make some investigations if that is something that makes sense to do inside zwave-js.

rccoleman, 2021-02-04:
I see.  I'm currently triggering on the notification event and changing the state manually, but I previously had it send the appropriate lock/unlock command to sync up the state.  I did see a mention of refreshing a single value, and that would also be an option.  Here's the automation that I'm currently using, one for each lock:

```
- alias: Fix front door lock
  initial_state: true
  trigger:
    - platform: event
      event_type: zwave_js_event
      event_data:
        node_id: 44
  condition:
    condition: template
    value_template: "{{ 'Manual' in trigger.event.data.label }}"
  action:
    - data:
        entity_id:
          - lock.front_door_lock
        state: "{{ 'unlocked' if 'unlock' in trigger.event.data.label else 'locked' }}"
      service: python_script.change_state
```

It uses my custom python script to change the internal HA state, but the action could just as easily be the following:

```
      entity_id: lock.front_door_lock
      service: lock.{{ 'unlock' if 'unlock' in trigger.event.data.label else 'lock' }}
```

It might be better to fix it at the source rather than delegating to the clients, but I can understand why you might want to avoid cluttering the codebase with special cases.  Maybe if it was a general "do this when you get this notification", it would be okay.

AlCalzone, 2021-02-04:
I've raised #1602 to investigate if this is a good idea to do automatically.