zwave-js/zwave-js #1316: [bug] Meter report errors

blhoward2, 2021-01-08:
**Describe the bug**
All of my devices that check-in with energy reports throw an error under the master branch with zwavejsmqtt/fix#23. I'm see this on multiple devices including Aeotec Switch 6s, Aeon Nano Dimmers, and an Aeon Microdimmer.

```
2021-01-08 03:42:44.407 ERROR GATEWAY: Error while discovering value 103-50-0-deltaTime-65537 of node 103: The config has not been loaded yet! The config has not been loaded yet!

ZWaveError: The config has not been loaded yet!

    at lookupMeter (/usr/src/app/node_modules/@zwave-js/config/src/Meters.ts:80:9)

    at Object.meterType (/usr/src/app/lib/Constants.js:21:19)

    at Gateway.discoverValue (/usr/src/app/lib/Gateway.js:1598:32)

    at Gateway.onValueChanged (/usr/src/app/lib/Gateway.js:234:10)

    at ZwaveClient.emit (events.js:314:20)

    at ZwaveClient.EventEmitter.emit (domain.js:483:12)

    at ZwaveClient.updateValue (/usr/src/app/lib/ZwaveClient.js:742:12)

    at ZwaveClient.onNodeValueUpdated (/usr/src/app/lib/ZwaveClient.js:390:15)

    at ZWaveNode.emit (events.js:314:20)

    at ZWaveNode.EventEmitter.emit (domain.js:483:12)

    at ZWaveNode.translateValueEvent (/usr/src/app/node_modules/zwave-js/src/lib/node/Node.ts:352:8)

    at ValueDB.emit (events.js:314:20)

    at ValueDB.EventEmitter.emit (domain.js:483:12)

    at ValueDB.setValue (/usr/src/app/node_modules/zwave-js/node_modules/@zwave-js/core/src/values/ValueDB.ts:224:10)

    at MeterCCReport.persistValues (/usr/src/app/node_modules/zwave-js/src/lib/commandclass/MeterCC.ts:602:12)

    at new MeterCCReport (/usr/src/app/node_modules/zwave-js/src/lib/commandclass/MeterCC.ts:534:8)
```

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Device information**
Which device(s) is/are affected?  
What are the node IDs?

**Logfile:** 

I can't generate better logfiles because logging does not work as described in the fix#23 PR.

AlCalzone, 2021-01-08:
What's the difference to the zwavejs2mqtt-dev image?

You could try and run zwavejs2mqtt without Docker, to make sure the build process of that isn't the problem.

blhoward2, 2021-01-08:
I can try that tonight.

jcam, 2021-01-09:
Hmm, I'm getting this on my bridge controller branch too

AlCalzone, 2021-01-09:
I'm like 99% certain issue due to the faulty docker image. This is **just** zwave-js:

[image]

blhoward2, 2021-01-09:
So what’s breaking the image then? It’s in every image I’ve built off fix#23/master.

AlCalzone, 2021-01-09:
My guess is this https://github.com/zwave-js/zwavejs2mqtt/pull/181 but I'm not sure.

blhoward2, 2021-01-09:
@AlCalzone I do not get this error in a container built with both https://github.com/zwave-js/zwavejs2mqtt/pull/181 and https://github.com/zwave-js/zwavejs2mqtt/pull/84. Since it exists when built with only fix#23, https://github.com/zwave-js/zwavejs2mqtt/pull/181 must fix it.

blhoward2, 2021-01-09:
I’ve also confirmed that this is fixed when https://github.com/zwave-js/zwavejs2mqtt/pull/181 is built with the config directory properly linked (per my comment on that PR). As such, this seems to be an artifact.

closed 2021-01-09 by blhoward2