zwave-js/zwave-js #1318: throttle misbehaving devices and drop excessive messages

AlCalzone, 2021-01-08:
> @AlCalzone Separate from this specific issue... Obviously the controller can’t do anything about a device flooding the mesh itself, but is it feasible to throttle misbehaving devices and drop excessive messages in some manner? Or to drop meter-type reports altogether if the system gets under too heavy of a load? The idea being that the mesh may go to hell but at least the broker stays running and accessible to reconfigure the device/debug.

> _Originally posted by @blhoward2 in https://github.com/zwave-js/zwavejs2mqtt/issues/174#issuecomment-756757547_

---

This should be done generically. Detect floods of identical unsolicited messages and ignore them. Maybe issue a warning so users know what their devices do.

jkaberg, 2021-01-08:
@AlCalzone might I also suggest uping all default reporting values? I highly doubt default values below 5 min is beneficial for 99% for the users but they have an high impact on the overall network situation

If anyone needs lower than 5 min then they still have an option to lower it

AlCalzone, 2021-01-08:
I don't think that is feasibly within the scope of this request. Not every device allows configuring the reporting, every device that does has different configuration parameters which are interpreted differently.

What one could maybe do is to curate sane default configurations that could be configured when including a device. But that will require community support like the config files already do.

blhoward2, 2021-01-08:
Some notes for consideration when looking at this:

qt-openzwave handles this by dropping all same-value refreshes by default (SuppressValueRefresh=True). If that approach is ultimately adopted then the user needs to be able to disable on a system-wide basis (like ozw) or device-by-device (ideally). That's because some devices deliberately refresh their values, such as double-tap switches that may be currentValue=0 but are double-tapped down again. Duplicate scene messages may come in. Or probes and the like that are used for trending and need the subsequent same-value data point.

An additional stability enhancement, in conjunction with this, may be to identify misbehaving devices in a manner that they could be disabled on restart until action is taken by the user. So perhaps an async thread keeps track of what is being worked on and if the broker goes down ungracefully it disables that device on restart. This is a common feature of consumer-facing software, including Windows/Word/Excel et al that disable misbehaving plugins on subsequent startups.

A case example: qt-openzwave not long ago had a bug where Leviton devices would flood the network with messages upon a power state change until the broker crashed. The docker would restart and all would be well until another state change, then it would happen all over again. Another time I had a hardware failure that caused a switch to stop switching the relay and to spam out continuous button-down events. That sort of scenario could be avoided by disabling the device on a restart after it causes a crash.

These are all important stability considerations in light of the proposed HA server, as a flood of messages passed on to HA could take all of HA down along with the broker. (I'm unsure of what, if any, safeguards they have in place to prevent this.)

blhoward2, 2021-01-08:
> I don't think that is feasibly within the scope of this request. Not every device allows configuring the reporting, every device that does has different configuration parameters which are [...]

This gets dicey though because then you are reconfiguring devices in a way you can't undo automatically. If someone spins up zwavejs2mqtt (or the HA addon when there is one) and then their devices no longer work the old way there will be a lot of pissed off users. I already have some power-monitoring devices configured the way I want them to detect my sump pump running, etc. as well. And others have power reporting disabled altogether because I don't need those to report.

AlCalzone, 2021-01-08:
> qt-openzwave handles this by dropping all same-value refreshes by default

I don't like that (also for the reasons you mentioned). Aside from spamming, sending the same value again may convey a meaning. For Central Scene, it is even described in the specs that the device may send a message (with the same value) every 200 (or 250?) milliseconds (although the sequence ID changes here, which makes it easier to detect dupes).

In the absense of a sequence ID, I think it is fair to throttle the handling of identical messages without being too strict. Maybe the Central Scene interval is the sweet spot.

AlCalzone, 2021-01-08:
> Leviton devices would flood the network with messages upon a power state change until the broker crashed. The docker would restart and all would be well until another state change

Do you have an idea why they would stop flooding once the controller goes down? Most of the low-level communication is out of the driver's reach.

blhoward2, 2021-01-08:
> Do you have an idea why they would stop flooding once the controller goes down? Most of the low-level communication is out of the driver's reach.

Looks like a bug-induced flood from the controller.

"Leviton DZ15S and maybe other switches that support Version 2, don't send a V2 report, but still send what appears to be a V1 report without the Target vale."

"In that issue logs were included that show that the switch is reporting V2 of the Binary Switch command class. V2 introduced Target and Duration values. The logs also show that the switch is responding to Get commands with a V1 reply, that does not include the target value. That is the where the problem resides with this device. OZW expects a target value to be returned so it can decide whether or not it needs to refresh the value. I think it's getting confused since the target value is not returned. So if a device says it's V2 and reports V1 messages, it's unexpected. But..."

"This Leviton device reports that it supports V2 of the Switch Binary command class

When turning a switch ON, OZW sends a V2 Set command, which includes the target value (ON) and duration. It looks like the switch handles the Set message just fine.

After the Set, OZW immediately issues a Get to verify the value. Since the device supports V2, OZW expects a response with the current value and target value.

The Leviton sends a Report in response to the Get, but it is in V1 format so it includes only the *current value, not the target value. OZW doesn't check the length of the data and reads the target value from a buffer, which is likely some default initialized memory. In C++, this would be zero of initialized properly. So this appears as if the target value is OFF, but the current value is ON. Since the device is not at the (incorrect) target value, OZW will issue another Get again and again until current value == target (OFF) (does it stop?)."

Here is the issue: https://github.com/OpenZWave/qt-openzwave/issues/146

Here is the ultimate upstream fix: https://github.com/OpenZWave/open-zwave/pull/2370

AlCalzone, 2021-01-08:
> I think the lesson is that it is dangerous to assume that any device actually complies with the standard and everything needs to be verified.

Yep that quickly becomes very clear when working with any devices.

AlCalzone, 2023-10-19:
Recommended parameter settings are tracked in https://github.com/zwave-js/node-zwave-js/issues/6239
Making the user aware in https://github.com/zwave-js/node-zwave-js/issues/2909

closed 2023-10-19 by AlCalzone