zwave-js/zwave-js #1320: [feat] Add previously working questions to the bug template

blhoward2, 2021-01-08:
**Is your feature request related to a problem? Please describe.**
For debugging/troubleshooting purposes, it would be helpful to know whether a recent code change caused the problem or whether this is a brand new device/problem. It would also be helpful to know whether it was previously working (with no config changes) with a different platform so we could go look at how that platform handles the issue if need be.

**Describe the solution you'd like**
Amend the github bug template to collect this information.

**Describe alternatives you've considered**
Asking many posters the same questions over and over.

**Additional context**
N/A

AlCalzone, 2021-01-08:
Good point, wanna propose a change in a PR?

blhoward2, 2021-01-08:
Yes, I can do that tonight.

closed 2021-01-09 by AlCalzone