zwave-js/zwave-js #1322: Need config for Honeywell 39348/ZW4008 Switch

robertpeteuil, 2021-01-08:
**Is your feature request related to a problem? Please describe.**
No.  Just need a config file added.
 
**Describe the solution you'd like**
I'd like a file for the device added to the config directory, so `zwavejs2mqtt` can use it to properly identify my device.

Information on the switch:
- Manufacturer: 0x39, Product ID: 0x3135,Type ID: 0x4952 
- [ZWave Alliance Page on Switch](https://products.z-wavealliance.org/products/3599/)
- [Open-Zwave XML config](https://github.com/OpenZWave/open-zwave/blob/master/config/honeywell/39348-zw4008.xml)

**Explaining your use-case will go a long way in helping us understand.**
Would love for the device to be identified properly by `zwavejs2mqtt`.

**Describe alternatives you've considered**
Considered submitting a PR, but since I've no experience in creating z wave device JSON configs; and there's a lot of commits and cleanup going on in that folder - I thought better for everyone to just report it.

**Additional context**
- [Amazon listing for switch](https://www.amazon.com/gp/product/B07P8W5GH6)

AlCalzone, 2021-01-09:
Yeah, probably better to wait with your first PR. Currently trying to make it as easy as possible.

closed 2021-01-10 by AlCalzone

robertpeteuil, 2021-01-10:
This is amazing turnaround, thanks!

What’s the best way to report/submit missing devices in the future?

- I’m happy to just open an issue like here, but I’m also willing to attempt to help if it’s time-efficient for you.
  - I’m aware that sometimes user PR’s add more work & are not necessarily time effective. 
- Based on the some of the PRs I saw - it appears there’s a tool for parsing the json definition file from openZwave

AlCalzone, 2021-01-10:
PRs are the easiest way. We've plenty checks in place to make sure the files are okay.

You can import files from other databases. They should still be verified manually, but the import gives you a good starting point.

1. clone the repo
2. `yarn install`
3. One of the following:

To import devices 1234 and 5678 (imaginary) from https://opensmarthouse.org/zwavedatabase - when you click on a device, you'll see the ID in the URL
```
yarn run config import -s oh -Dd --ids 1234 5678
```

To import a device from OZW Github, run
```
yarn run config import -s ozw -Dd --ids 0x1234-0x2345-0x3456
```
Where the ID consists of the manufacturer id, product type and product ID. It is basically the key you see in this line
https://github.com/OpenZWave/open-zwave/blob/master/config/2gig/ct100.xml#L3, but formatted slightly differently.

robertpeteuil, 2021-01-10:
Thanks for the write-up.  Looking forward to helping as I move forward.

PR #1325 (2021-01-10): feat(config): add Honeywell 39348/ZW4008
...and fix lint error source.

We should probably wait for #1324 and clean up afterwards.

fixes: #1322