zwave-js/zwave-js #1339: [question]  How to correct behaviour of Philio PAN in-wall relay?

Swiftnesses, 2021-01-10:
Hi,

It seems the current behaviour of zwavejs2mqtt is the same as the 1st post here (3 switches, wrong behaviour):

https://github.com/OpenZWave/open-zwave/issues/2426

How can I correct this (with OZW I would modify the xml files), looking for some insight as to how to solve this kind of 'issue'.

Thank you in advance.

AlCalzone, 2021-01-10:
I am not entirely sure, but if I understand correctly, you'll need to add the compat flag [`preserveRootApplicationCCValueIDs`](https://zwave-js.github.io/node-zwave-js/#/development/config-files?id=preserverootapplicationccvalueids) to keep the root device (endpoint 0) from being hidden.

Part of the confusion might be that OZW starts counting at 1 and zwave-js at 0.

Swiftnesses, 2021-01-11:
Hi @AlCalzone, sounds a good place to start.

Can you point me in the right direction of how I can achieve this running zwavejs2mqtt in Docker?

Many thanks for the help - appreciate it.

AlCalzone, 2021-01-11:
@robertsLando Is an easy way to test device config modifications with Docker?

robertsLando, 2021-01-11:
@AlCalzone Actually nope. I could create an api that exec the `npm update @zwave-js/config` command?

robertsLando, 2021-01-11:
The problem is that this in docker needs a new volume in order to make it persistent

AlCalzone, 2021-01-11:
Okay then (until #1238 is done) we need a branch that changes this config file and @Swiftnesses can build the Dockerfile.contrib with that.

Swiftnesses, 2021-01-11:
Not sure it helps, but I always use persistent volumes, I wasn't sure if it was possible to implement such changes via one of the files (as little like OZW allows mapping of a custom config directory, with associated xmls.

Swiftnesses, 2021-01-11:
This is what I currently have in my docker volume....

[image]

AlCalzone, 2021-01-11:
The included configs are not in the mapped volume. You'll find them inside the container in `node_modules/@zwave-js/config/config/devices`.
Once the mentioned PR is implemented, you'll be able to just add additional configs to the volume and zwave-js will load them over the included ones.

Swiftnesses, 2021-01-14:
@robertsLando did I see a PR for this, if so, please let me know what I can do to test...

AlCalzone, 2021-01-14:
@Swiftnesses please give https://github.com/zwave-js/node-zwave-js/pull/1399 a try.

Swiftnesses, 2021-01-14:
@AlCalzone I use the mqtt docker (dev) by @robertsLando, what's the best way to deploy the fix?

AlCalzone, 2021-01-14:
Using the Dockerfile.contrib:
https://github.com/zwave-js/zwavejs2mqtt/tree/master/docker#building-a-container-using-dockerfilecontrib

Swiftnesses, 2021-01-14:
To make it easier I've mapped "/usr/src/app/node_modules/@zwave-js/config" in my container to a volume and modified the file as per your pull request. Is a restart / node refresh enough (it's a b**** to include this node again!).

Thanks :)

AlCalzone, 2021-01-14:
yep, just refresh the node info

Swiftnesses, 2021-01-14:
I've been waiting for the zwave network to come back up - I have run a refresh and then rediscovered the node. A new switch did appear! Sadly the node stopped responding (likely unrelated) - I'll try again in the morning :)

AlCalzone, 2021-01-16:
were you able to test yet?

closed 2021-01-17 by AlCalzone

Swiftnesses, 2021-01-17:
Howdy @AlCalzone I did, but it wasn't working as expected - the extra switch appeared and seemed to control the light, but the status wouldn't stay 'on', after turning it on, the light does in fact come on, but the switch then turns 'off' straight away, the the status is all wrong :(

Swiftnesses, 2021-01-17:
Apologies for the slow response - the sensor is in my workshop and it's horrible outside, plus my kids are at home (COVID!).

AlCalzone, 2021-01-17:
I think what you're seeing is just another case of all the issues concerning wrong status reports.

Swiftnesses, 2021-01-17:
Perhaps - I don't know, but my initial thought was without the switch showing the right status in Home Assistant, it's not going to be useful. Is there a fix coming, I'll retest if there is..

Thanks :)

Swiftnesses, 2021-01-19:
@AlCalzone just wondering if the status report fix is coming, until then I'll try and replace the node...

Swiftnesses, 2021-01-19:
BTW, I confirm it seems to be a switch status issue in Home Assistant. If I switch it on, the light comes on (the switch then goes off, the light remains on). If I quickly turn the switch on and off, the light goes off (obviously the switch goes off, as it always does 😄).

AlCalzone, 2021-01-19:
The problem is that zwave was not designed in a way that really supports mirroring the true status of devices. Almost all the integrations work around that by polling (which can quickly saturate the network) if the device does not send status updates by itself (which only newer devices do).

I can "fix" the part of a wrong status after changing it through zwave-js, but if the device does not report manual changes, it must be polled regularly. That also means your mirrored status **will** be delayed.

Swiftnesses, 2021-01-19:
The question is, is the status wrong because of something in zwave-js (with this device, or with status updates in general), or is it because we need to poll the device? I remember in OZW that I only needed to poll when I switched things at the wall (to ensure status matched), I didn't need to poll to get what we're fighting with here working - so my feeling is it's status related (a bug?).

AlCalzone, 2021-01-19:
OZW does verify the value change after setting it differently and I'm going to try and change that in zwave-js too.

Swiftnesses, 2021-01-19:
I trust in your knowledge - for now I'll try and find another relay. Give me a shout if I can help test anything!

PR #1399 (2021-01-17): fix(config): preserve root values for Philip PAN06
fixes: #1339