zwave-js/zwave-js #1344: [bug] Configuration seems to be empty / not show current values?

LordMike, 2021-01-11:
**Version**

Build/Run method
- [x] Docker
- [ ] PKG
- [ ] Manually built (git clone - npm install - npm run build ) 

zwavejs2mqtt version: v1.0.0-alpha.2

**Describe the bug**
When looking into a device that is completely discovered / interviewed, the configuration tab has all empty input boxes for various configs. In Z2M, this would show the current values.

Is this a missing feature, a bug, or did I do something wrong?

I've refreshed the info of nodes multiple times to no avail.

**To Reproduce**
N/A

**Expected behavior**
Configuration shows current configuration.

**Additional context**
Screenshot:

[image]

AlCalzone, 2021-01-11:
@LordMike Please choose a higher loglevel and try to send me the zwave-js log only (log to file).
I merged another PR earlier which includes a workaround for when the device doesn't respond with anything useful to the `getProperties(0)` - but your problem might be something different.

LordMike, 2021-01-11:
Will see if I can get that done. :)

I don't think I can run anything not docker though.. Will you guys (@robertsLando) make a develop image that has the latest/bleeding edge code from both repos?

AlCalzone, 2021-01-11:
You can use the Dockerfile.contrib over at zwavejs2mqtt to test out branches.

LordMike, 2021-01-11:
I made a "silly" log while interviewing the ZDB5100 Matrix. It made the following log snippet (I tailed the log file when it became silent after network start).

[log.txt](https://github.com/zwave-js/node-zwave-js/files/5798028/log.txt)

It has the following bit:

```
19:46:32.675 CNTRLR   [Node 009] ConfigurationCC: doing a complete interview...
19:46:32.675 CNTRLR » [Node 009] finding first configuration parameter...
19:46:32.677 SERIAL » 0x010b00130904700e00002568d9                                        (13 bytes)
19:46:32.677 DRIVER » [Node 009] [REQ] [SendData]
                      │ transmit options: 0x25
                      │ callback id:      104
                      └─[ConfigurationCCPropertiesGet]
                          parameter #: 0
19:46:32.679 SERIAL « [ACK]                                                                   (0x06)
19:46:32.685 SERIAL « 0x0104011301e8                                                       (6 bytes)
19:46:32.685 SERIAL » [ACK]                                                                   (0x06)
19:46:32.685 DRIVER « [RES] [SendData]
                        was sent: true
19:46:32.702 SERIAL « 0x010700136800000380                                                 (9 bytes)
19:46:32.702 SERIAL » [ACK]                                                                   (0x06)
19:46:32.702 DRIVER « [REQ] [SendData]
                        callback id:     104
                        transmit status: OK
19:46:32.718 SERIAL « 0x0110000400090a700f000000000100000096                              (18 bytes)
19:46:32.718 SERIAL » [ACK]                                                                   (0x06)
19:46:32.719 DRIVER « [Node 009] [REQ] [ApplicationCommand]
                      └─[ConfigurationCCPropertiesReport]
                          parameter #:  0
                          next param #: 0
                          value size:   0
                          value format: SignedInteger
19:46:32.720 CNTRLR   [Node 009] [+] [Configuration] interviewComplete: true [Endpoint 0] [internal]
```

LordMike, 2021-01-11:
I've attempted to build my own image with the master branches of the two repositories. I'm not sure which git hashes I'm running at, but the log output seems identical to the previous comment.

[log.txt](https://github.com/zwave-js/node-zwave-js/files/5798173/log.txt)

```
20:23:41.347 CNTRLR » [Node 009] finding first configuration parameter...
20:23:41.349 SERIAL » 0x010b00130904700e0000258332                                        (13 bytes)
20:23:41.349 DRIVER » [Node 009] [REQ] [SendData]
                      │ transmit options: 0x25
                      │ callback id:      131
                      └─[ConfigurationCCPropertiesGet]
                          parameter #: 0
20:23:41.364 SERIAL « [ACK]                                                                   (0x06)
20:23:41.365 SERIAL « 0x0104011301e8                                                       (6 bytes)
20:23:41.366 SERIAL » [ACK]                                                                   (0x06)
20:23:41.366 DRIVER « [RES] [SendData]
                        was sent: true
20:23:41.373 SERIAL « 0x01070013830000026a                                                 (9 bytes)
20:23:41.374 SERIAL » [ACK]                                                                   (0x06)
20:23:41.374 DRIVER « [REQ] [SendData]
                        callback id:     131
                        transmit status: OK
20:23:41.388 SERIAL « 0x0110000400090a700f000000000100000096                              (18 bytes)
20:23:41.389 SERIAL » [ACK]                                                                   (0x06)
20:23:41.390 DRIVER « [Node 009] [REQ] [ApplicationCommand]
                      └─[ConfigurationCCPropertiesReport]
                          parameter #:  0
                          next param #: 0
                          value size:   0
                          value format: SignedInteger
20:23:41.391 CNTRLR   [Node 009] [+] [Configuration] interviewComplete: true [Endpoint 0] [internal]
20:23:41.392 CNTRLR   [Node 009] BinarySwitchCC: doing a complete interview...
```

AlCalzone, 2021-01-11:
Looks like the workaround doesn't do anything in your case because the message is actually well-formatted. Please try the `config-get-properties-zero` branch.

closed 2021-01-13 by AlCalzone

LordMike, 2021-01-17:
Sorry I couldn't get back sooner. The current `latest` build of zwavejs2mqtt now has the configuration data available!

[image]

PR #1352 (2021-01-13): fix(cc): add another workaround for getProperties returning no first param
fixes: #1344