zwave-js/zwave-js #1348: Use `type: 'duration'` for Duration valueIds

robertsLando, 2021-01-11:
Ref. https://github.com/zwave-js/zwavejs2mqtt/issues/185

AlCalzone, 2021-01-11:
In addition we should consider https://github.com/zwave-js/zwavejs2mqtt/issues/185#issuecomment-758256815

robertsLando, 2021-01-12:
> Keep Duration objects, but give library users a way to work with them (zwave-js/node-zwave-js#1348 as a first step)
Add a method to serialize them as either a timespan string like you proposed or the current object format
Add a method to deserialize them from a timespan string or the current object format. Trying to use an invalid combination will throw.
Devs can then decide whether they need to work with timespan strings, value+unit objects or the original Duration object.

PR #1361 (2021-01-12): feat: add and use "duration" metadata type
fixes: #1348
closed 2021-01-12 by AlCalzone