zwave-js/zwave-js #1349: [feat] Import all certified devices from the zWave Alliance

blhoward2, 2021-01-11:
**Is your feature request related to a problem? Please describe.**
The zWave Alliance has an xml file on every certified device, which includes the device id, firmware versions, configuration parameters, etc. The plan would be to import all of those provided device xmls into the project to ensure the largest number of potential working devices, amending existing device files where necessary and adding new ones. There are 3,466 device xmls available as of 1/10/2021, though there is duplication among devices due to firmware revisions so there are in practice fewer "devices" as most users would consider them. This covers all frequency spectrums/countries.

**Note: I've already downloaded all xmls over a period of time so as to not abuse their server.** Let's please not all go pull down all 3,466 xmls from the zwave alliance as I don't want them to decide the database isn't worth their server being pounded.

Here is an example file: https://products.z-wavealliance.org/Products/4024/XML

For curiosity sake, here is the year-by-year breakdown of devices included, by certification date:

2014: 71
2015: 234
2016: 425
2017: 513
2018: 398
2019: 486
2020: 280
2021: 10

These xmls also include inclusion/exclusion instructions, reset directions, etc. This information can be used to auto-generate some of the supported devices documentation: https://github.com/zwave-js/zwavejs2mqtt/issues/148

**Describe the solution you'd like**
Amend the current import configuration routines to process these xmls and to generate/amend project device files.

AlCalzone, 2021-01-11:
> These xmls also include inclusion/exclusion instructions, reset directions, etc.

I'm open to adding that (which we can always do later though). Question is how much use the inclusion instructions are, as we can only show them **after** the device was included.

blhoward2, 2021-01-11:
> I'm open to adding that (which we can always do later though). Question is how much use the inclusion instructions are, as we can only show them **after** the device was included.

They'd be in the documentation, so if the user knows what device they have they could look.

cgarwood, 2021-01-13:
Having the wakeup/exclusion/etc metadata would be useful for the Home Assistant integration as well

AlCalzone, 2021-05-24:
@blhoward2 anything left to do here?

closed 2021-12-29 by blhoward2