zwave-js/zwave-js #1354: Support firmware downloading (reading fw from device)

LordMike, 2021-01-11:
**Is your feature request related to a problem? Please describe.**
I have an idea that's mostly a curiosity right now: Can we build a firmware database, with easily-usable binaries for different device types? (We probably can't... licensing and whatnot).

Part of this idea involves gathering firmwares that can be tricky to get. Downloading a firmware _from_ a device, would help a lot. Devices with newer firmwares could help efforts to flash older devices, even when the original manufacturer refuses or can't provide new firmware images.

**Describe the solution you'd like**
[In the docs](https://www.silabs.com/documents/login/miscellaneous/SDS13782-Z-Wave-Management-Command-Class-Specification.pdf#%5B%7B%22num%22%3A366%2C%22gen%22%3A0%7D%2C%7B%22name%22%3A%22XYZ%22%7D%2C89%2C735%2C0%5D) for the z-wave `COMMAND_CLASS_FIRMWARE_UPDATE_MD` CC, it says:

> The Firmware Update Meta Data Command Class may be used to transfer a firmware image to or from a Z-Wave node.

_This is for version 4+, and most of the devices I've seen are version 3 - but here's to hoping._

If it was possible, an API to download a firmware image from the device would be beneficial. The downloaded images would be as raw as they could be, without any additional packing or whatnot from manufacturers.. :)

**Describe alternatives you've considered**
N/A

**Additional context**
Entirely theoretical, it should be possible to flash one device via. another.. Hmm.. 

I originally [wrote about this on ozw](https://github.com/OpenZWave/open-zwave/issues/1563#issuecomment-647816537).

AlCalzone, 2021-01-11:
For reference:
[image]

Seems I missed that entirely 😬

LordMike, 2021-01-11:
I'm envisioning the pirate database of firmwares now. Keyed by manufacterer/product ids, and firmware versions.. :D

AlCalzone, 2021-01-11:
Yeah, I don't think many devices will offer their firmwares for download if they aren't already available freely.

LordMike, 2021-01-11:
Spoilsport. :P

blhoward2, 2021-01-12:
I think this is a bad idea as it exposes the project to potential copyright violations for little benefit. It’s one thing to let the user retrieve the firmware and another to distribute that binary without permission.

roflmao, 2021-01-12:
While making it possible doesnt mean that this will be a problem before anyone makes FW public. I know Fibaro doesnt make FW available, forcing people to buy their hub to get updates.

LordMike, 2021-01-12:
> 
> 
> I think this is a bad idea as it exposes the project to potential legal wrangling for little benefit. It’s one thing to let the user retrieve the firmware and another to distribute that binary without permission.

I wouldn't say zwavejs should make the store.. Merely implement the API to fetch the fw.. :)

jkaberg, 2021-01-12:
@LordMike I agreed, even more so if this is apart of the (official?) Z-Wave specification. Haveing feature parity would be great

blhoward2, 2021-01-12:
> I wouldn't say zwavejs should make the store.. Merely implement the API to fetch the fw.. :)

Ok, to be clear I don’t think the function is an issue, but there reason for doing so is to implement the full specification.

Hedda, 2021-02-09:
> Can we build a firmware database, with easily-usable binaries for different device types? (We probably can't... licensing and whatnot).

You would probably be OK legally if you provide firmware images as a public service as it should be allowed according to "fair use" doctrine as long as you only provide the binary files unmodified "as is" so that they can really only be flashed on the same product.

FYI, the Zigbee2MQTT community does almost exactly this for Zigbee devices OTA firmware update images that are not publicly available to be downloaded directly via an URL from the manufacturers, so that can perform OTA upgrades via zigbee-herdsman:

https://github.com/Koenkk/zigbee-OTA

Zigbee2MQTT, however, does not have a feature to download an FW image from a device, (that part is done manually today).

Repo has a JSON file with an index that gets updated each time a new OTA firmware update image is added to that GitHub repo:

https://github.com/Koenkk/zigbee-OTA/blob/master/index.json

Their collection is limited because they have decided to not collect/provide OTA firmware update images for Zigbee devices where the manufacturer already publicly provides OTA firmware via an URL. 

https://github.com/Koenkk/zigbee2mqtt/issues/1696

https://github.com/Koenkk/zigbee-herdsman/issues/60

There is also a similar discussion about creating an OTA community repository for zigpy where they talk about PROS and CONS here:

https://github.com/zigpy/zigpy/issues/154

blhoward2, 2021-02-09:
> > Can we build a firmware database, with easily-usable binaries for different device types? (We probably can't... licensing and whatnot).
> 
> 
> 
> > Part of this idea involves gathering firmwares that can be tricky to get.
> 
> 
> 
> > I'm envisioning the pirate database of firmwares now. Keyed by manufacterer/product ids, and firmware versions.. :D
> 
> 
> 
> FYI, the Zigbee2MQTT community does almost exactly this for Zigbee devices OTA firmware update images that are not publicly available to be downloaded directly via an URL from the manufacturers, so that can perform OTA upgrades via zigbee-herdsman:
> 
> 
> 
> https://github.com/Koenkk/zigbee-OTA
> 
> 
> 
> Zigbee2MQTT, however, does not have a feature to download an FW image from a device, (that part is done manually today).
> 
> 
> 
> Repo has a JSON file with an index that gets updated each time a new OTA firmware update image is added to that GitHub repo:
> 
> 
> 
> https://github.com/Koenkk/zigbee-OTA/blob/master/index.json
> 
> 
> 
> Their collection is limited because they have decided to not collect/provide OTA firmware update images for Zigbee devices where the manufacturer already publicly provides OTA firmware via an URL. 
> 
> 
> 
> https://github.com/Koenkk/zigbee2mqtt/discussions/2921
> 
> 
> 
> https://github.com/Koenkk/zigbee2mqtt/issues/1696
> 
> 
> 
> https://github.com/Koenkk/zigbee-herdsman/issues/60
> 
> 
> 
> There is also a similar discussion about creating an OTA community repository for zigpy where they talk about PROS and CONS here:
> 
> 
> 
> https://github.com/zigpy/zigpy/issues/154
> 
> 
> 
> > I think this is a bad idea as it exposes the project to potential legal wrangling for little benefit. It’s one thing to let the user retrieve the firmware and another to distribute that binary without permission.
> 
> 
> 
> > While making it possible doesnt mean that this will be a problem before anyone makes FW public. I know Fibaro doesnt make FW available, forcing people to buy their hub to get updates.
> 
> 
> 
>  It is probably OK legally if you provide firmware images in a repo as a public service as it should be allowed according to "fair use" doctrine as long as you only provide the binary files unmodified "as is" so that they can really only be flashed on the same product.
> 
> 
> 
> https://en.wikipedia.org/wiki/Fair_use

This is just wrong. And I AM a lawyer.

Hedda, 2021-02-09:
> > It is probably OK legally if you provide firmware images in a repo as a public service as it should be allowed according to "fair use" doctrine as long as you only provide the binary files unmodified "as is" so that they can really only be flashed on the same product.
> > https://en.wikipedia.org/wiki/Fair_use
> 
> This is just wrong. And I AM a lawyer. Fair use doesn't cover this.

Fair use might be the wrong term, I am not a lawyer, but do you really think that some the Zigbee2MQTT project is running a huge risk of being sued for providing Zigbee device OTA images which are not publicly available via URL from the original manufacturer?  

Again, it is not like the Zigbee2MQTT project is hacking the firmware, they are providing those OTA firmware image files unmodified which means that "as is" they are only useful to users that own exact same type and model of product from that manufacturer.

If it then more likely that any manufacturer who would not be happy with their firmware images being provided in a GitHub repository by a third-party like this would only submit a cease and desist letter? 

In that case that cease and desist letter would probably only come from one of those manufacturers, and if so it would be easy to comply by simply removing the image files from that manufacturer.

PS: Please do not blow this all out of propositions as I was just stating it as a fact that Zigbee2MQTT has such OTA collection which from that software can only be used to update the firmware of exact same products which the manufacturer intended it for.

blhoward2, 2021-02-09:
GitHub isn't the proper forum for a legal debate. I think a copyright letter would go to GitHub and GitHub can and does remove or limit accounts for hosting copyrighted material. You're also confusing the likelihood of enforcement with legality. Many manufacturers do in fact go after websites in such situations.

AlCalzone, 2021-02-09:
Side note: The entire issue depends on the devices actually responding to the download commands.

Hedda, 2021-02-09:
I think a copyright letter would go to GitHub and GitHub can and does remove or limit accounts for hosting copyrighted material. You're also confusing the likelihood of enforcement with legality. Many manufacturers do in fact go after websites in such situations.

Regardless, I would hope and assume that it is a non-issue in practice for this use-case and thus a moot point here. That is, the fear of potential legal consequences should hopefully not stop anyone here from considering the idea of implementing the same type of concept of hosting a repository with a collection of OTA image for Z-Wave devices that Zigbee2MQTT community has here today:

https://github.com/Koenkk/zigbee-OTA 

> GitHub isn't the proper forum for a legal debate. 

I agree.

blhoward2, 2021-02-09:
This forum isn't meant for this type of debate. We're going to stop spamming everyone now.

blhoward2, 2021-02-09:
Now the the timeout is over and thread cleaned up, I'm unlocking this again. In speaking with @AlCalzone we have no interest in hosting firmware images. Let's please limit discussion to the requested function to download firmware from the device, if doing so is even possible.

tkralphs, 2023-01-24:
I landed on this issue because I was hoping this functionality existed. I have several otherwise identical devices with different firmware versions that were obtained at different times. The oldest firmware has an acknowledged bug but the company has not released an upgrade while newer devices come with a patched version. All I want to be able to do is copy the firmware from one device to another, which it seems I should have a right to do as an owner of the device. This seems like a pretty clear cut legitimate use case.

tkralphs, 2023-01-24:
And by the way, thanks to the developers of zwave-js for the great work. I'm in the process of moving from SmartThings to Home Assistant and so far, the process has been very smooth. Working with the combination of zwave-js and Home Assistant has been a vastly better experience so far than working with SmartThings.

LordMike, 2023-01-24:
> I landed on this issue because I was hoping this functionality existed. I have several otherwise identical devices with different firmware versions that were obtained at different times. The oldest [...]

Iirc, the ability to download a firmware was removed in the FIRMWARE_MD v3 or something like that. At least, that's what I remember from the docs, that older versions had this idea that you could do stuff with the firmware, while later versions went away from it. I can only suspect that the original intent was like yours, while the later intent was influenced by "proprietary" software - but I'm guessing.. :)

LordMike, 2023-01-26:
Oh Wow. I must have completely misunderstood the docs for that command. I was certain they removed it :O

My bad.

Mike.

tkralphs, 2023-01-27:
If I'm looking at the right thing in the diagnostics file, the Firmware Update Meta Data version for the device whose firmware I wanted to download is v4. So yeah, guess it's not going to happen. The version for the devices with the older firmware is v2! Thanks for your help, I really appreciate your work on this.