zwave-js/zwave-js #1357: Support for Silicon Labs SLUSB001A - Z-Wave 700 UZB-7 USB Stick evaluation hardware and reference design

Hedda, 2021-01-12:
https://github.com/zwave-js/node-zwave-js/issues/1100 already mention Series 700 stick support however would just like to understand if Silicon Labs SLUSB001A - Silabs Z-Wave 700 UZB-7 USB Stick evaluation hardware is supported before I order one?

Surprisingly the SLUSB001A looks like one of the most affordable Z-Wave adapters available today. 

https://www.silabs.com/documents/public/data-sheets/DSH14483.pdf

https://www.silabs.com/documents/public/schematic-files/PHD14501.zip

Silabs community says that this UZB7 stick is backwards compatible with old  Z-Wave Serial API command set for Z-Wave Plus via USB for the older the UZB3 stick but I would like to use it in Z-Wave 700 mode to test the few new Z-Wave Gen7 devices available.

https://www.silabs.com/community/wireless/z-wave/forum.topic.html/uzb_7_backwards_compatiblewithuzb3-mFu8

This is same dongle that official Z-Wave 700 reference design from Silicon Labs Z-Wave 700 Wireless Starter Kit (SLWSTK6050A)

https://www.silabs.com/development-tools/wireless/z-wave/z-wave-700-starter-kit

https://www.silabs.com/development-tools/wireless/z-wave/z-wave-long-range-700-starter-kit

Follow-up question; this single UZB-7 USB Stick (SLUSB001A) hardware should be all regions but the host must change the actual RF region to be used, so the next question is if Z-Wave JS supports changing RF region for UZB-7 USB Stick (SLUSB001A)?

https://www.silabs.com/community/wireless/z-wave/knowledge-base.entry.html/2019/03/01/z-wave_700_how_toc-iQk2

https://www.silabs.com/wireless/z-wave/global-regions

PS: Region information will be stored permanently on the stick after being set by the host controller.

marcus-j-davies, 2021-01-12:
Disclaimer : Someone with more experience in zwave-js may want to back this up, but....

> #1100 already mention Series 700 stick support however would just like to understand if Silicon Labs SLUSB001A - Silabs Z-Wave 700 SLUSB7000A UZB-7 USB Stick evaluation hardware is supported before I order one?

Adding support should be quite easy - see [this](https://github.com/zwave-js/node-zwave-js/tree/master/packages/config/config)

Folder structure is : **manufacturerId** / **productId** / **productType_firmwareVersion.json**

>PS: Region information will be stored permanently on the stick after being set by the host controller.

We have talked about adding proprietary function support, depending on the protocol of the stick, you may be able to do this via **sendMessage**

example of turning off the LED on a Zwave stick

```
const turnLEDOff = new Message(driver, {
  type: MessageType.Request,
  functionType: 0xF2,
  payload: Buffer.from("5101000501", "hex"),
});
await driver.sendMessage(turnLEDOff, {
  priority: MessagePriority.Controller,
  supportCheck: false,
});
```

the same method may be able to set the region.

again - I would get a 2nd opinion on this.

Hedda, 2021-01-12:
> the same method may be able to set the region.

Another follow-up question is setting/adjusting TX power for SLUSB7000A UZB-7 dongle?

https://www.silabs.com/community/wireless/z-wave/forum.topic.html/how_to_set_tx_powertoslusb7000auzb-7dongle-MbOF

https://www.silabs.com/community/wireless/z-wave/knowledge-base.entry.html/2019/01/04/z-wave_700_how_toa-JSjE.html

AlCalzone, 2021-01-12:
Without having read all the references you posted:  
Aeotec's Gen7 stick seems to be supported with the new API since https://github.com/zwave-js/node-zwave-js/pull/1276, so I'd assume that you can at least control your network with that.

The additional commands you mention are either not implemented yet or proprietary and thus depend on having a good API concept for https://github.com/zwave-js/node-zwave-js/issues/1245

Hedda, 2021-01-12:
> The additional commands you mention are either not implemented yet or proprietary and thus depend on having a good API concept for #1245

OK, I understand. When you do decide to look at Z-Wave 700 I suggest you use a SLUSB001A / SLUSB7000A stick as reference.

AlCalzone, 2021-01-12:
I'm getting one of those soon, so I'll can test it first hand.

jcam, 2021-01-13:
Do you have this stick? If you can give a log of the startup, we can at least take care of the config descriptors file.

kpine, 2021-01-16:
If it helps, here are some logs and cache files, as well as screenshots from the PC-Controller software. I don't have any nodes added besides the controller.

[cache.zip](https://github.com/zwave-js/node-zwave-js/files/5823996/cache.zip) 
[zwave2.log](https://github.com/zwave-js/node-zwave-js/files/5824001/zwave2.log)

The log file actually has two runs. For some reason my stick had 4-virtual nodes. I'm not sure if it came that way from the factory,  or if they were added by the zipgateway when I was trying that previously. I factory reset the node after running that since it was causing some failures, and the virtual nodes were removed. I left that in log in case it was interesting for some reason.

Unrelated to 700-series, do you plan to update this to show 100k speed (or a separate field)?
```
            "maxBaudRate": 40000,
```

[image]

[image]

kpine, 2021-01-16:
Appears to have the same IDs as the Zooz. 😆 

[dump.txt](https://github.com/zwave-js/node-zwave-js/files/5824014/dump.txt)

AlCalzone, 2021-01-16:
The 100kbps speed is not documented in the public specs yet. If anyone has more details, let me know.

jcam, 2021-01-16:
Guessing the Zooz firmware guys forgot to update the IDs, and we should update the config to report SiLabs...

kpine, 2021-01-16:
> 
> 
> The 100kbps speed is not documented in the public specs yet. If anyone has more details, let me know.

Ah, you're right, I forget what's public and not. I thought it was public now because of this [comment](https://github.com/zwave-js/node-zwave-js/discussions/1329#discussioncomment-272205). Perhaps it will be soon.

syepes, 2021-02-08:
1+ For this integration

jcam, 2021-02-08:
@kpine @billiaz should we keep this issue open? From my understanding, this already is working.

kpine, 2021-02-08:
My SLUSB001A is working fine, so I think we can say it's "supported". Adding support for serial API functions (tx power) or displaying the 100kbps speed perhaps belong in their own issues?

AlCalzone, 2021-02-08:
Agreed. If the stick is working per se, any missing functionality should be it's own issue.

Hedda, 2021-02-09:
> > My SLUSB001A is working fine, so I think we can say it's "supported".
> 
> Agreed. If the stick is working per se, any missing functionality should be it's own issue.

Nice! Do you also feel that that is can be recommended to new Z-Wave JS users who haven't yet bought a Z-Wave USB stick?

Again, SLUSB001A (SLUSB7000A UZB-7) is surprisingly among the most affordable Z-Wave USB adapters readily available today:

https://www.google.com/search?q=SLUSB001A

SLUSB001A is sold for about less than $20/€20 from many large global distributors of electronic components around the world.

I'm thinking that installing a Z-Wave 700 series compatible stick from the start hopefully makes the setup more future-proof?

Hedda, 2021-02-09:
> Appears to have the same IDs as the Zooz. 😆

My guess is probably both [Zooz ZST10 700 Series S2 Stick](https://cdn.shopify.com/s/files/1/0218/7704/files/zooz-700-series-z-wave-plus-usb-stick-zst10-700-manual.pdf) and [Aeotec Z-Stick 7 Gen7](https://aeotec.freshdesk.com/support/solutions/articles/6000230552-z-stick-7-user-guide-) USB dongles are more or less just clones of the SLUSB001A/SLUSB7000A reference hardware so can use firmware images for it compiled by Silicon Labs for [first-mover advantage](https://en.wikipedia.org/wiki/First-mover_advantage)(?)

bushvin, 2021-04-16:
Is there currently any documentation on how to get series 7 (SLUSB7000A) adapters working?

zwavejs detects the adapter correctly, but I cannot join devices, the adapter doesn't seem to detect any devices to join.

AlCalzone, 2021-04-16:
It works for me, so I will need a log to look at.

bushvin, 2021-04-16:
Coincidentally, I found issue #1764, which resolved my issue.
It appears my firmware version (7.11) wasn't liked. So after updating it to the latest provided through Simplicity Studio, it started to work! Yay!

I tried to use the *Firmware Update* feature from the control panel, but no joy, unfortunately:
```
2021-04-16 23:16:11.764 INFO ZWAVE: Calling api beginFirmwareUpdate with args: [
  1,
  'ZW_SerialAPI_Controller_7.15.4_256_EFR32ZG14_REGION_EU.gbl',
  <Buffer eb 17 a6 03 08 00 00 00 00 00 00 03 01 01 00 00 fa 06 06 fa 10 00 00 00 d2 66 02 00 46 ee 4f 4c 1f 8a c0 48 8c 31 63 c6 f9 07 07 f9 24 00 00 00 72 3c ... 157486 more bytes>,
  0,
  [length]: 4
]
2021-04-16 23:16:11.774 INFO ZWAVE: Node 1 does not support the Command Class Firmware Update Meta Data! beginFirmwareUpdate {
  success: false,
  message: 'Node 1 does not support the Command Class Firmware Update Meta Data!'
}
```
Anyway, it works now!

AlCalzone, 2021-04-16:
> I tried to use the Firmware Update feature from the control panel

That's for OTA firmware update of the other nodes. OTW firmware update for the stick works quite differently and isn't implemented yet.

bushvin, 2021-04-16:
Thanks for clarifying!

closed 2021-05-04 by AlCalzone

Hedda, 2021-05-05:
> I tried to use the _Firmware Update_ feature from the control panel, but no joy, unfortunately

FYI, I posted a follow-up question about FW in the discussions section here -> https://github.com/zwave-js/node-zwave-js/discussions/2544

blamaz, 2021-09-07:
Somes online stores part no for the USB7 is SLUSB001A while others it's SLUSB7000A.
Is there any difference between the 2 ?

AlCalzone, 2021-09-07:
According to https://www.mouser.de/ProductDetail/Silicon-Labs/SLUSB001A?qs=u16ybLDytRbJsMfgk3EnSA==, the latter is an alias for the former, but don't take my word for it 😅

Hedda, 2021-09-08:
Yes those are the same hardware. FYI, tips is also that I started a discussion thread about SLUSB001A / SLUSB7000A here -> https://github.com/zwave-js/node-zwave-js/discussions/2544

PR #2500 (2021-05-04): feat: implement get/setPowerlevel, get/setRFRegion, auto-enable TX status reports
With this PR, we automatically configure supporting sticks to give extended TX status reports we'll be using in a later step.
Furthermore, this adds the methods get/setPowerlevel, get/setRFRegion to configure the stick's radio.

fixes: #1673
fixes: #1674
fixes: #1357